zephyrproject-rtos/zephyr #56070: ADC DMA fails on FRDM K64F

teburd, 2023-03-21:
**Describe the bug**
The adc_dma test, buildable using

```sh
west build -p always -b frdm_k64f -T tests/drivers/adc/adc_dma/drivers.adc-dma
```

fails since commit
6e21ebf2e03259db8ae216314ced9ed84eb3a0cd

It looks like perhaps the NXP specific changes made in this commit broke things.

teburd, 2023-03-21:
@heinwessels please take a look, commit points to a change you made to the test

heinwessels, 2023-03-22:
The failure shown in the description is caused by me changing the `interval_us` from `HW_TRIGGER_INTERVAL`  to `0`. I know this is fine the STM32 so I incorrectly assumed it should work for the NXP as well. I asked teburd to see if that fixes it, and it's definitely a problem. However, the test fails at a new location afterwards.

```
*** Booting Zephyr OS build zephyr-v3.3.0-1492-ga57810dc3b1e ***
Running TESTSUITE adc_dma
===================================================================
START - test_adc_asynchronous_call
Samples read: 0x0372 0x0000 0x00f4 0x0000 0x037a 0x0000 
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/adc/adc_dma/src/test_adc.c:176: check_samples: (INVALID_ADC_VALUE not equal to sample_value)
[5] should be empty
 FAIL - test_adc_asynchronous_call in 0.021 seconds
===================================================================
START - test_adc_invalid_request
E: Invalid resolution
E: Invalid resolution
Samples read: 0x0034 0x0000 
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/adc/adc_dma/src/test_adc.c:176: check_samples: (INVALID_ADC_VALUE not equal to sample_value)
[1] should be empty
 FAIL - test_adc_invalid_request in 0.022 seconds
===================================================================
START - test_adc_repeated_samplings
repeated_samplings_callback: done 1
Samples read: 0x0201 0x0000 
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/adc/adc_dma/src/test_adc.c:176: check_samples: (INVALID_ADC_VALUE not equal to sample_value)
[1] should be empty
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/kernel/sched.c:1750
        aborting essential thread 0x200007e0
E: r0/a1:  0x00000004  r1/a2:  0x000006d6  r2/a3:  0x00000080
E: r3/a4:  0x00002de5 r12/ip:  0x00000000 r14/lr:  0x0000663f
E:  xpsr:  0x41000011
E: Faulting instruction address (r15/pc): 0x00007c20
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x200007e0 (main)
E: Halting system
```

It's likely the crash is called by a test failing, but the cause of the test failing concrens me. `INVALID_ADC_VALUE not equal to sample_value` is a check I added, copied directly from `adc_api`, where it verfies the buffer is filled correctly. I'm wondering if this could point to a bug in the driver?

hakehuang, 2023-03-22:
> It's likely the crash is called by a test failing, but the cause of the test failing concrens me. `INVALID_ADC_VALUE not equal to sample_value` is a check I added, copied directly from `adc_api`, [...]

Thanks @heinwessels  for qiuck analysis, I am checking the driver.

heinwessels, 2023-03-22:
Having a look at `adc_mcux_adc16.c` it seems like the `dma_block.block_size` is set bigger than it should be, which would explain the `(INVALID_ADC_VALUE not equal to sample_value)` error. 

The potential problematic code is here:
https://github.com/zephyrproject-rtos/zephyr/blob/1dce3c3ee2e9db41a0111709c337bafce1a45834/drivers/adc/adc_mcux_adc16.c#L275

When starting the DMA the bytes-to-transfer (called `block_size`) is set to `sequence->buffer_size`. And for the test, the `buffer_size` is set to the actual size of the buffer, which is much larger than it needs to be for this specific test, in this case `24`. 

IMO `buffer_size` is the size of the buffer, and should not be used. (only to check that it's indeed big enough). Instead the `block_size` should be set to something like `sizeof(sample)`.

In the STM32 driver we set it explictly to the number of channels samples multiplied by the size of each sample, and ignore the size of the buffer.

@hakehuang do you think this might be it?

hakehuang, 2023-03-22:
@heinwessels , I will craete a PR for this fix

heinwessels, 2023-03-22:
> @heinwessels , I will craete a PR for this fix, btw will you fix the interval_us issue, or you want to fix in as well?

I'll create a fix for this one. It doesn't seem to be blocking the CI, so I think it's fine if it's two seperate PRs.

Thanks!

heinwessels, 2023-03-22:
@hakehuang I'm not sure how to fix the tests in a generic way, because I'm not sure of what the NXP drivers need. Could you possibly tell me why? For example, before my changes, the `interval_us` was set to 3 different values in three different tests. 

For `test_task_asynchronous_call` it's set to  `HW_TRIGGER_INTERVAL = 2` .
https://github.com/zephyrproject-rtos/zephyr/blob/aac03280ecf9e1a2376ebe006e2589bda480675c/tests/drivers/adc/adc_dma/src/test_adc.c#L247-L248
For `test_task_with_interval` it's set to `500`.
https://github.com/zephyrproject-rtos/zephyr/blob/aac03280ecf9e1a2376ebe006e2589bda480675c/tests/drivers/adc/adc_dma/src/test_adc.c#L312
For `test_task_repeated_samplings` it's set to `10000`.
https://github.com/zephyrproject-rtos/zephyr/blob/aac03280ecf9e1a2376ebe006e2589bda480675c/tests/drivers/adc/adc_dma/src/test_adc.c#L401-L403

For me this all seems like it should all be replaced with a single value, because that's the minimum value that seems to work, and most tests can use that. So I can replace the `test_task_asynchronous_call` and `test_task_repeated_samplings` with this value.
```C
#define MINIMUM_SAMPLING_TIME = 2
```

The only test that might need a different timing is `test_task_with_interval`, but the actual timing isn't tested at all, so technically the `interval_us` could be set to `2` as well. However, it seems to me that this test is a special case and needs an additional
```C
#define INTERVAL_SAMPLING_TIME = 500

// in the test_task_with_interval test

options.interval_us = INTERVAL_SAMPLING_TIME;
zassert_true(milliseconds_spent > ( options.extra_samplings * INTERVAL_SAMPLING_TIME ), "...");
```

What do you think? Can I fix the tests this way?

hakehuang, 2023-03-22:
> What do you think? Can I fix the tests this way?

the sample interval depends on  the hardware mode, the minial interval is 2us for nxp paltforms(consider the longsample mode). as to the 500US and 1000US, it is because I want to make sure the sample interval check is smaller than the sample time, such as I can internval at least twise in repeat sampling mode.

hakehuang, 2023-03-22:
> > @heinwessels , I will craete a PR for this fix, btw will you fix the interval_us issue, or you want to fix in as well?
> 
> I'll create a fix for this one. It doesn't seem to be blocking the CI, so I think it's fine if it's two seperate PRs.
> 
> Thanks!

For the changes to
```
blk_cfg->block_size = channel_count * sizeof(int16_t); 
```
How can the driver know the expect result to return? in your test it is 1 plus 4 extra, but buffer given 24. and I did not see an option pass to driver on the exact number of results expected. say if someone wants 2 records in adc read, how could they do this?

heinwessels, 2023-03-22:
@hakehuang 

>  the minial interval is 2us for nxp paltforms(consider the longsample mode),
Of course. This would be a `#define` that's set to `2` for NXP, but defaults to zero.

> As to the 500US and 1000US, it is because I want to make sure the sample interval is double of the sample interval time, such as I can get exact a new data back.

As far as I can see, it doesn't matter. If this `interval_us` is only used as a delay between the end of the previous sample, and the start of the next sample, as forced by the `adc_context.h`. It's not up to the driver implementation. Basically once a sample is completed, the driver calls `adc_context_on_sampling_done`. Then `adc_context` decides if it will delay or not, before triggering the next sample.

This means it _should_ still wait for your hardware timer.

The important part is this function:
https://github.com/zephyrproject-rtos/zephyr/blob/a7c011b0bee4ca3bd72f192641bee56526a07e7a/drivers/adc/adc_context.h#L203-L222

I will implement it this way, and then maybe you can check if the NXP tests still pass?

> How can the driver know the expect result to return? in your test it is 1 plus 4 extra, but buffer given 24. and I did not see an option pass to driver on the exact number of results expected. say if someone wants 2 records in adc read, how could they do this?

The `extra_samples` is handled by `adc_context` where it repeats a read for every `extra_sample`. This means your driver doesn't have to account for it. `adc_context` _should_ automatically increment the buffer pointer for example. And since your driver doesn't support sampling multiple _channels_ at once then `channel_count` will always be `1`, and thus your DMA will only have to transfer the data for one single read.

hakehuang, 2023-03-22:
> The `extra_samples` is handled by `adc_context` where it repeats a read for every `extra_sample`. This means your driver doesn't have to account for it. `adc_context` _should_ automatically [...]

@heinwessels , the NXP DMA can support multiple samples and increate the buffer by HW, so it will not need the extra software involving. I think maybe you can just skip the extra check so only do below
```
		if (i < expected_count) {
			zassert_not_equal(INVALID_ADC_VALUE, sample_value,
				"[%u] should be filled", i);
		} 
```
and remove the
```
		if (i < expected_count) {
			zassert_not_equal(INVALID_ADC_VALUE, sample_value,
				"[%u] should be filled", i);
		} else {
			zassert_equal(INVALID_ADC_VALUE, sample_value,
				"[%u] should be empty", i);
		}
```

heinwessels, 2023-03-22:
> he NXP DMA can support multiple samples and increate the buffer by HW

In that case it should still set the `block_size` correctly, meaning `sizeof(int16_t) * (1 + options.extra_samples)`. 

The test is correct. If the user wants a read with 4 extra samples and supplies a buffer, then only `2 * 5 = 10` bytes should be written by the ADC driver. Even if the buffer is bigger than that. Currently your driver is writing 24 bytes.

hakehuang, 2023-03-22:
> I will implement it this way, and then maybe you can check if the NXP tests still pass?

sure, please add me to your PR review list. Thanks

hakehuang, 2023-03-22:
@heinwessels , I find another issue here, see below

```
 ( (*** Booting Zephyr OS build zephyr-v3.3.0-1516-g1dce3c3ee2e9 ***
Running TESTSUITE adc_dma
===================================================================
START - test_adc_asynchronous_call
Samples read: 0x0371 0x003c 0x0371 0x003c 0x0371 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000
0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000
0xffff8000 0xffff8000 0xffff8000
 PASS - test_adc_asynchronous_call in 0.024 seconds
===================================================================
START - test_adc_invalid_request
E: Invalid resolution
E: Invalid resolution
Samples read: 0x032d 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000
0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000 0xffff8000
0xffff8000 0xffff8000 0xffff8000
 PASS - test_adc_invalid_request in 0.029 seconds
===================================================================
START - test_adc_repeated_samplings
repeated_samplings_callback: done 1
Samples read: 0x0371 0x003c
    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/adc/adc_dma/src/test_adc.c:176: check_samples: (INVALID_ADC_VALUE not equal to sample_value)
[1] should be empty
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/kernel/sched.c:1750
        aborting essential thread 0x20000740
E: r0/a1:  0x00000004  r1/a2:  0x000006d6  r2/a3:  0x000000a2
E: r3/a4:  0x00002de5 r12/ip:  0x00000000 r14/lr:  0x0000663f
E:  xpsr:  0x41000011
E: Faulting instruction address (r15/pc): 0x00007c20
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x20000740 (idle)
E: Halting system

```

the issue is that is I correct the adc_mcux_adc16.c with 2 transfer and set the block_size to sizeof(int16_t) * (1 + options.extra_samples), and the test_adc_repeated_samplings still fails as it expect to get only one result, but as our driver do 1 + extra_samples in batches, the adc_context_on_sampling_done does not work here, how to handle this?

heinwessels, 2023-03-22:
@hakehuang I still think your NXP driver should have `block_size = sizeof(int16_t)`. Can you try it with that? There's no mention of `extra_samples` in your driver, so how would the DMA know to repeat the sampling?

How the NXP driver should work:
1. User calls `read` with `extra_samples = 2`. There are three channels selected.
2. Driver configures DMA to read the three channels _once_. This means DMA length should be set to `3 * sizeof(int16_t)` or `POPCOUNT(data->channels) * sizeof(int16_t)`.
3. All three channels are read, which calls `adc_dma_callback`, which in turn calls `adc_context`'s `adc_context_on_sampling_done`.
4. Now `adc_context` decides what to do next. If there are `extra_samples` specified it will trigger the read again, meaning start again at step `2`. `adc_context` will also call the callback, which might return `ADC_ACTION_REPEAT`, which will force it to sample again anyway.

heinwessels, 2023-03-22:
> and I find in st driver when dma is done below code is moved by the channel count, so st driver looks ok in this case.

Ah yeah, it might be that you have to add that. It's that exact line that caused me to improve the unit tests to check the buffer which is now failing for NXP, because I also forgot it. You can see it here in https://github.com/zephyrproject-rtos/zephyr/pull/52965#discussion_r1110034653

heinwessels, 2023-03-22:
@hakehuang great news! So that means my PR works, and I will remove the draft label.

> they want the dma to do multiply samples, not only one sample a time it is critial for their user case, any idea to provide such support as well

And my customer also maybe want's such a feature. I've opened a RFC for the ADC API to support this. If you could comment there that you need it it might get more interaction and decided sooner. You can find it here: https://github.com/zephyrproject-rtos/zephyr/issues/55783

hakehuang, 2023-03-22:
> * User calls `read` with `extra_samples = 2`. There are three channels selected.
> * Driver configures DMA to read the three channels _once_. This means DMA length should be set to `3 * sizeof(int16_t)` or `POPCOUNT(data->channels) * sizeof(int16_t)`.
> * All three channels are read, which calls `adc_dma_callback`, which in turn calls `adc_context`'s `adc_context_on_sampling_done`.
> * Now `adc_context` decides what to do next. If there are `extra_samples` specified it will trigger the read again, meaning start again at step `2`. `adc_context` will also call the callback, which might return `ADC_ACTION_REPEAT`, which will force it to sample again anyway.

nxp adc channel dma mode does not handle the multi-channel well,  I will add the missing parts as well, Thanks to point this out.

hakehuang, 2023-06-01:
> @hakehuang is this issue resolved?

yes it is resolved, and we see another failure [#58467](https://github.com/zephyrproject-rtos/zephyr/issues/58467)

closed 2023-06-01 by hakehuang