zephyrproject-rtos/zephyr #56072: Invalid KConfig arguments causes spurious errors for certain boardfiles

glarsennordic, 2023-03-21:
**Describe the bug**
For some (but not all) boardfiles, introducing an erroneous KConfig argument causes bad warnings to be printed.

For instance, let's start with an example of normal behavior:

1. Navigate to `zephyr/samples/net/sockets/echo`.
2. execute `west build -p -b cc3235sf_launchxl -- -DCONFIG_NET_CONFIG_LOG_LEVEL=y -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y`

As expected, this will produce the following KConfig warning output:

> warning: the value 'y' is invalid for NET_CONFIG_LOG_LEVEL (defined at subsys/net/Kconfig.template.log_config.net:48), which has type int -- assignment ignored

However, if we instead use the `nrf9160dk_nrf9160_ns` board: `west build -p -b nrf9160dk_nrf9160_ns -- -DCONFIG_NET_CONFIG_LOG_LEVEL=y -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y`

We get the following incorrect warning in addition to the one seen before:

> warning: The choice symbol NET_SOCKETS_LOG_LEVEL_DBG (defined at
subsys/net/Kconfig.template.log_config.net:36) was selected (set =y), but no symbol ended up as the
choice selection. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_NET_SOCKETS_LOG_LEVEL_DBG and/or look up
NET_SOCKETS_LOG_LEVEL_DBG in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

This warning is completely incorrect, `NET_SOCKETS_LOG_LEVEL_DBG` should be able to be set with no issues.

Strangely, once this warning occurs, it will persist on subsequent builds so that even:

`west build -p -b nrf9160dk_nrf9160_ns -- -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y`

triggers the warning, although the build will otherwise succeed.

Seemingly, the only way to clear the warning is to abort the above build just after it prints the warning, then run it again.

So far, the only two boards in Zephyr that I have tried are `nrf9160dk_nrf9160_ns` and `cc3235sf_launchxl`, but this also affects the out-of-tree board `thingy91_nrf9160_ns` board.

It isn't clear to me whether this is specific to the nrf9160, or if it might be a broader issue, but I've been encouraged to raise it here os that the right people can take a look.

It's worth noting that this seems to apply to all log-level settings, not just `NET_SOCKETS_LOG_LEVEL` and `NET_CONFIG_LOG_LEVEL`. Those two are merely the first two I could find that apply to the `echo` sample.

**Impact**
This is an annoyance; It causes confusion if one mistakenly forgets to append `_DBG` to a log output level. Once you are aware of the problem, it is easy enough to circumvent it by just restarting builds and avoiding incorrect KConfig arguments. Still, it signals that there may be some kind of bug in the KConfig processor.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: zephyr-sdk-0.15.0
 - Commit SHA or Version used: 483126f492eaefd1bbd9daf7543a5dfd6a429795

glarsennordic, 2023-03-21:
@jackrosenthal Pinging you since you are now maintaining KConfig

nordicjm, 2023-03-22:
> However, if we instead use the nrf9160dk_nrf9160_ns board: west build -p -b nrf9160dk_nrf9160_ns -- -DCONFIG_NET_CONFIG_LOG_LEVEL=y -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y

This is incorrect, you cannot select a choice, `CONFIG_NET_CONFIG_LOG_LEVEL` needs to be dropped

nordicjm, 2023-03-22:
The reason you get the warning is you are trying to select a symbol without enabling its dependencies which is not valid, this includes the dependencies and works: `west build -p -b nrf9160dk_nrf9160_ns -- -DCONFIG_LOG=y -DCONFIG_NET_LOG=y -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y`

Closing, not a bug.

closed 2023-03-22 by nordicjm

glarsennordic, 2023-03-22:
@nordicjm 

> This is incorrect, you cannot select a choice, CONFIG_NET_CONFIG_LOG_LEVEL needs to be dropped

Yes, that I am aware of, but the issue I am pointing out is that this (seems to) cause an additional unrelated warning to be printed beyond the expected `warning: the value 'y' is invalid for NET_CONFIG_LOG_LEVEL`

> The reason you get the warning is you are trying to select a symbol without enabling its dependencies which is not valid, this includes the dependencies and works `west build -p -b nrf9160dk_nrf9160_ns -- -DCONFIG_LOG=y -DCONFIG_NET_LOG=y -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y`

What about the fact that I have had `west build -p -b nrf9160dk_nrf9160_ns -- -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y` ***succeed with no warnings***? If `-DCONFIG_LOG=y -DCONFIG_NET_LOG=y` is a necessity, then is this build succeeding a bug?

It feels very strange to me that repeating the same build twice will yield different results.

glarsennordic, 2023-03-23:
@nordicjm 

`west build -p -b nrf9160dk_nrf9160_ns -- -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y` does not contain any invalid arguments.

It either should either always fail, or should always succeed (depending on whether `-DCONFIG_LOG=y -DCONFIG_NET_LOG=y` is required)  and I have seen ***both*** behaviors for that _exact_ command, simply by performing it repeatedly.

If you really think that is alright, then I'll drop it, but I feel that I am failing to communicate clearly what has me concerned so I am restating it one last time

nordicjm, 2023-03-24:
In your first message, you specifically say you use `-DCONFIG_NET_CONFIG_LOG_LEVEL=y` which is invalid. I've just ran `west build -p -b nrf9160dk_nrf9160_ns -- -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y` 9 times and each time the warning was emitted.

glarsennordic, 2023-03-24:
> In your first message, you specifically say you use -DCONFIG_NET_CONFIG_LOG_LEVEL=y which is invalid.

@nordicjm I was apparently confused in my first message, I thought the warning was ***not*** expected for `west build -p -b nrf9160dk_nrf9160_ns -- -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y`, **especially since it doesn't print (ever) for** `west build -p -b cc3235sf_launchxl -- -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y`. I was always aware that `-DCONFIG_NET_CONFIG_LOG_LEVEL=y ` is invalid, but from my confused perspective it ***appeared*** to be affecting other commands.

But you corrected me on that, so that confusion is gone. I think we can drop the issue of `-DCONFIG_NET_CONFIG_LOG_LEVEL=y` for the time being.

But there does remain the fact that **I can get inconsistent behavior with**  `nrf9160dk_nrf9160_ns`. See the following terminal output:

```
PS C:\ncs\zephyr_kconfig_bug\zephyr\samples\net\sockets\echo> west build -p -b nrf9160dk_nrf9160_ns -- -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y
-- west build: making build dir C:\ncs\zephyr_kconfig_bug\zephyr\samples\net\sockets\echo\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo
-- CMake version: 3.22.1
-- Found Python3: C:/Python39/python.exe (found suitable exact version "3.9.5") found components: Interpreter
-- Cache files will be written to: C:/ncs/zephyr_kconfig_bug/zephyr/.cache
-- Zephyr version: 3.3.99 (C:/ncs/zephyr_kconfig_bug/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: nrf9160dk_nrf9160_ns, Revision: 0.7.0
-- Found host-tools: zephyr 0.15.0 (C:/Users/Georges/zephyr-sdk-0.15.0)
-- Found toolchain: zephyr 0.15.0 (C:/Users/Georges/zephyr-sdk-0.15.0)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/ncs/zephyr_kconfig_bug/zephyr/boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160_ns.dts
-- Generated zephyr.dts: C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo/build/zephyr/dts.cmake

warning: The choice symbol NET_SOCKETS_LOG_LEVEL_DBG (defined at
subsys/net/Kconfig.template.log_config.net:36) was selected (set =y), but no symbol ended up as the
choice selection. See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_NET_SOCKETS_LOG_LEVEL_DBG and/or look up
NET_SOCKETS_LOG_LEVEL_DBG in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing C:/ncs/zephyr_kconfig_bug/zephyr/Kconfig
Loaded configuration 'C:/ncs/zephyr_kconfig_bug/zephyr/boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160_ns_defconfig'
Merged configuration 'C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo/prj.conf'
Merged configuration 'C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo/build/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to 'C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo/build/zephyr/.config'
Kconfig header saved to 'C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo/build/zephyr/include/generated/autoconf.h'
PS C:\ncs\zephyr_kconfig_bug\zephyr\samples\net\sockets\echo> ^C
PS C:\ncs\zephyr_kconfig_bug\zephyr\samples\net\sockets\echo> west build -p -b nrf9160dk_nrf9160_ns -- -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo
-- CMake version: 3.22.1
-- Found Python3: C:/Python39/python.exe (found suitable exact version "3.9.5") found components: Interpreter
-- Cache files will be written to: C:/ncs/zephyr_kconfig_bug/zephyr/.cache
-- Zephyr version: 3.3.99 (C:/ncs/zephyr_kconfig_bug/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: nrf9160dk_nrf9160_ns, Revision: 0.7.0
-- Found host-tools: zephyr 0.15.0 (C:/Users/Georges/zephyr-sdk-0.15.0)
-- Found toolchain: zephyr 0.15.0 (C:/Users/Georges/zephyr-sdk-0.15.0)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/ncs/zephyr_kconfig_bug/zephyr/boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160_ns.dts
-- Generated zephyr.dts: C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo/build/zephyr/dts.cmake
Parsing C:/ncs/zephyr_kconfig_bug/zephyr/Kconfig
Loaded configuration 'C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo/build/zephyr/.config'
No change to configuration in 'C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo/build/zephyr/.config'
No change to Kconfig header in 'C:/ncs/zephyr_kconfig_bug/zephyr/samples/net/sockets/echo/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
[... cut ...]
The highlights:

First, I attempt `west build -p -b nrf9160dk_nrf9160_ns -- -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y`

I get the warning.

Right after I get the warning, I abort the build (ctrl-C). I attempt `west build -p -b nrf9160dk_nrf9160_ns -- -DCONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y` again.

This time, the build goes through with no warning.

So maybe the bug here, actually, is "aborting build early suppresses CMake warnings". I apologize for my confusion early on. Hopefully things are clear now?

nordicjm, 2023-03-27:
What you are seeing is because you started configuring a build, files were written, but the overall cmake files which are actually used to build the project and what the python scripts use to determine if it is a valid cmake project are not there. @mbolivar-nordic do you consider this a bug in west if when doing west build the pristine option only works if cmake completely configured the project before rather than partially before being aborted with ctrl + c?

mbolivar-nordic, 2023-03-27:
> @mbolivar-nordic do you consider this a bug in west if when doing west build the pristine option only works if cmake completely configured the project before rather than partially before being aborted with ctrl + c?

If by "the pristine option" you mean `--pristine always`, then no; I would consider this to be a bug in the build system, since west just calls out to the build system to do that.

If you mean `--pristine auto`, then an incompletely configured build system is likely to cause difficult to diagnose problems. The `auto` mode usually works but it is by necessity based on heuristics and doesn't always get things right.

mbolivar-nordic, 2023-03-27:
I wonder if we need some sort of sentinel files in the build system directory that contain the cmake start and stop times, so we can tell if configuration completed successfully or not. Something like that' is the only way I can think of to start making builds transactional and more robust. That's more a question for you and @tejlmand though, wouldn't you agree @nordicjm ?