zephyrproject-rtos/zephyr #56133: Treewide: clock control API minor casting enhancement

pdgendt, 2023-03-23:
**Is your enhancement proposal related to a problem? Please describe.**
The clock control driver is currently used treewide like:
```c
clock_control_on(clock_dev, (clock_control_subsys_t *)&cfg->clock_cfg);
clock_control_get_rate(clock_dev, (clock_control_subsys_t *)&cfg->clock_cfg, &rate);
```
Where the `clock_control_subsys_t` is an opaque type which the API expects
```c
typedef void *clock_control_subsys_t;
// ...
typedef int (*clock_control)(const struct device *dev,
			     clock_control_subsys_t sys);

typedef int (*clock_control_get)(const struct device *dev,
				 clock_control_subsys_t sys,
				 uint32_t *rate);
```

**Describe the solution you'd like**
`s/(clock_control_subsys_t *)/(clock_control_subsys_t)/` for all uses of the clock control API.

carlescufi, 2023-03-28:
Architecture WG:

- @pdgendt states that is already a mix of usage with and without asterisk in-tree
- @henrikbrixandersen states that this is different from #49333 because the API is left untouched
- Does *not* require any changes to out-of-tree because using with or without asterisk does not trigger warnings
- @galak agrees this is a worthwhile cleanup
- Agree to squash the commits
- No objections in the call

pdgendt, 2023-03-29:
> Note that typedef could be left as is, deprecated, but all its usages/function signatures updated without creating a breaking change. I think it's worth considering this option now that we touch dozens of files for a typedef that doesn't make any sense.

I didn't change the `typedef` in the PR.

gmarull, 2023-03-29:
> > Note that typedef could be left as is, deprecated, but all its usages/function signatures updated without creating a breaking change. I think it's worth considering this option now that we touch dozens of files for a typedef that doesn't make any sense.
> 
> I didn't change the `typedef` in the PR.

I know, just suggesting that cleanup could remove typedef usage entirely, use void * in functions, but keep the typedef for OOT users, so change doesn't break their code.

pdgendt, 2023-03-29:
> I know, just suggesting that cleanup could remove typedef usage entirely, use void * in functions, but keep the typedef for OOT users, so change doesn't break their code.

That would touch a lot more files, and casting to `void *` would still need to be done with all the API calls to avoid compiler messages like:
```sh
warning: passing argument 2 of 'clock_control_on' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
# or
note: expected 'void *' but argument is of type 'const struct atmel_sam_pmc_config *'
```

PR #56270 (2023-04-05): Treewide: Update clock control API usage
Replace all `(clock_control_subsys_t *)` casts with `(clock_control_subsys_t)` as it is supposed to be used in the clock control API:

```c
typedef void *clock_control_subsys_t;
// ...
typedef void (*clock_control_cb_t)(const struct device *dev,
				   clock_control_subsys_t subsys,
				   void *user_data);
typedef int (*clock_control)(const struct device *dev,
			     clock_control_subsys_t sys);
typedef int (*clock_control_get)(const struct device *dev,
				 clock_control_subsys_t sys,
				 uint32_t *rate);
```

Fixes #56133
closed 2023-04-05 by carlescufi