zephyrproject-rtos/zephyr #56136: tests: lib: spsc_pbuf: libraries.spsc_pbuf_utilization fails

nordic-piks, 2023-03-23:
**Describe the bug**
The tests/lib/spsc_pbuf/libraries.spsc_pbuf_utilization fails

Tests:
- tests/lib/spsc_pbuf/libraries.spsc_pbuf_utilization

Observed for
- nrf52dk_nrf52832

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52dk_nrf52832 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/lib/spsc_pbuf -p nrf52dk_nrf52832 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
[2023-03-23T01:22:13.785Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.3.0-1516-g1dce3c3ee2e9 ***[0m
[2023-03-23T01:22:13.785Z] [0mRunning TESTSUITE test_spsc_pbuf[0m
[2023-03-23T01:22:13.785Z] [0m===================================================================[0m
[2023-03-23T01:22:13.785Z] [0mSTART - test_0cpy[0m
[2023-03-23T01:22:13.785Z] [0mPASS - test_0cpy in 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m===================================================================[0m
[2023-03-23T01:22:13.785Z] [0mSTART - test_0cpy_corner1[0m
[2023-03-23T01:22:13.785Z] [0mPASS - test_0cpy_corner1 in 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m===================================================================[0m
[2023-03-23T01:22:13.785Z] [0mSTART - test_0cpy_corner2[0m
[2023-03-23T01:22:13.785Z] [0mPASS - test_0cpy_corner2 in 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m===================================================================[0m
[2023-03-23T01:22:13.785Z] [0mSTART - test_0cpy_discard[0m
[2023-03-23T01:22:13.785Z] [0mPASS - test_0cpy_discard in 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m===================================================================[0m
[2023-03-23T01:22:13.785Z] [0mSTART - test_0cpy_smaller[0m
[2023-03-23T01:22:13.785Z] [0mPASS - test_0cpy_smaller in 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m===================================================================[0m
[2023-03-23T01:22:13.785Z] [0mSTART - test_largest_alloc[0m
[2023-03-23T01:22:13.785Z] [0mPASS - test_largest_alloc in 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m===================================================================[0m
[2023-03-23T01:22:13.785Z] [0mSTART - test_spsc_pbuf_ut[0m
[2023-03-23T01:22:13.785Z] [0mPASS - test_spsc_pbuf_ut in 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m===================================================================[0m
[2023-03-23T01:22:13.785Z] [0mSTART - test_spsc_pbuf_ut_cache[0m
[2023-03-23T01:22:13.785Z] [0mPASS - test_spsc_pbuf_ut_cache in 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m===================================================================[0m
[2023-03-23T01:22:13.785Z] [0mSTART - test_stress[0m
[2023-03-23T01:22:13.785Z] [0mReading from an interrupt, writing from a thread[0m
[2023-03-23T01:22:13.785Z] [0m0% remaining:14000 ms[0m
[2023-03-23T01:22:13.785Z] [0m0% remaining:13000 ms[0m
[2023-03-23T01:22:13.785Z] [0m0% remaining:12000 ms[0m
[2023-03-23T01:22:13.785Z] [0mZtress execution report:[0m
[2023-03-23T01:22:13.785Z] [0mcontext 0:[0m
[2023-03-23T01:22:13.785Z] [0m- executed:25383, preempted:0[0m
[2023-03-23T01:22:13.785Z] [0m- ticks initial:4, optimized:4[0m
[2023-03-23T01:22:13.785Z] [0mcontext 1:[0m
[2023-03-23T01:22:13.785Z] [0m- executed:7774, preempted:2001[0m
[2023-03-23T01:22:13.785Z] [0m- ticks initial:4, optimized:4[0m
[2023-03-23T01:22:13.785Z] [0mAverage CPU load:28%, measurements:32[0m
[2023-03-23T01:22:13.785Z] [0mWrites:11749 failures: 0[0m
[2023-03-23T01:22:13.785Z] [0mWriting from an interrupt, reading from a thread[0m
[2023-03-23T01:22:13.785Z] [0m100% remaining:14000 ms[0m
[2023-03-23T01:22:13.785Z] [0mZtress execution report:[0m
[2023-03-23T01:22:13.785Z] [0mcontext 0:[0m
[2023-03-23T01:22:13.785Z] [0m- executed:11230, preempted:0[0m
[2023-03-23T01:22:13.785Z] [0m- ticks initial:4, optimized:4[0m
[2023-03-23T01:22:13.785Z] [0mcontext 1:[0m
[2023-03-23T01:22:13.785Z] [0m- executed:3117, preempted:1000[0m
[2023-03-23T01:22:13.785Z] [0m- ticks initial:4, optimized:4[0m
[2023-03-23T01:22:13.785Z] [0mAverage CPU load:27%, measurements:14[0m
[2023-03-23T01:22:13.785Z] [0mWrites:19491 failures: 9118[0m
[2023-03-23T01:22:13.785Z] [0mPASS - test_stress in 4.686 seconds[0m
[2023-03-23T01:22:13.785Z] [0m===================================================================[0m
[2023-03-23T01:22:13.785Z] [0mSTART - test_stress_0cpy[0m
[2023-03-23T01:22:13.785Z] [0m7% remaining:14000 ms[0m
[2023-03-23T01:22:13.785Z] [0m16% remaining:13000 ms[0m
[2023-03-23T01:22:13.785Z] [0m24% remaining:12000 ms[0m
[2023-03-23T01:22:13.785Z] [0m33% remaining:11000 ms[0m
[2023-03-23T01:22:13.785Z] [0m42% remaining:10000 ms[0m
[2023-03-23T01:22:13.785Z] [0m48% remaining:9000 ms[0m
[2023-03-23T01:22:13.785Z] [0m56% remaining:8000 ms[0m
[2023-03-23T01:22:13.785Z] [0m64% remaining:7000 ms[0m
[2023-03-23T01:22:13.785Z] [0m73% remaining:6000 ms[0m
[2023-03-23T01:22:13.785Z] [0m83% remaining:5000 ms[0m
[2023-03-23T01:22:13.785Z] [0m93% remaining:4000 ms[0m
[2023-03-23T01:22:13.785Z] [0mZtress execution report:[0m
[2023-03-23T01:22:13.785Z] [0mcontext 0:[0m
[2023-03-23T01:22:13.785Z] [0m- executed:45456, preempted:0[0m
[2023-03-23T01:22:13.785Z] [0m- ticks initial:4, optimized:4[0m
[2023-03-23T01:22:13.785Z] [0mcontext 1:[0m
[2023-03-23T01:22:13.785Z] [0m- executed:42837, preempted:1000[0m
[2023-03-23T01:22:13.785Z] [0m- ticks initial:4, optimized:4[0m
[2023-03-23T01:22:13.785Z] [0mAverage CPU load:56%, measurements:116[0m
[2023-03-23T01:22:13.785Z] [0m9% remaining:14000 ms[0m
[2023-03-23T01:22:13.785Z] [0m16% remaining:13000 ms[0m
[2023-03-23T01:22:13.785Z] [0m23% remaining:12000 ms[0m
[2023-03-23T01:22:13.785Z] [0m31% remaining:11000 ms[0m
[2023-03-23T01:22:13.785Z] [0m39% remaining:10000 ms[0m
[2023-03-23T01:22:13.785Z] [0mAssertion failed at ../../../../../../../../zephyr/zephyr/tests/lib/spsc_pbuf/src/main.c:502: stress_claim_free: check_buffer(buf, len, ctx->read_cnt) is non-zero[0m
[2023-03-23T01:22:13.785Z] [0mFAIL - test_stress_0cpy in 17.521 seconds[0m
[2023-03-23T01:22:13.785Z] [0m===================================================================[0m
[2023-03-23T01:22:13.785Z] [0mSTART - test_utilization[0m
[2023-03-23T01:22:13.785Z] [0mPASS - test_utilization in 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m===================================================================[0m
[2023-03-23T01:22:13.785Z] [0mTESTSUITE test_spsc_pbuf failed.[0m
[2023-03-23T01:22:13.785Z] [0m------ TESTSUITE SUMMARY START ------[0m
[2023-03-23T01:22:13.785Z] [0mSUITE FAIL -  90.91% [test_spsc_pbuf]: pass = 10, fail = 1, skip = 0, total = 11 duration = 22.216 seconds[0m
[2023-03-23T01:22:13.785Z] [0m- PASS - [test_spsc_pbuf.test_0cpy] duration = 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m- PASS - [test_spsc_pbuf.test_0cpy_corner1] duration = 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m- PASS - [test_spsc_pbuf.test_0cpy_corner2] duration = 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m- PASS - [test_spsc_pbuf.test_0cpy_discard] duration = 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m- PASS - [test_spsc_pbuf.test_0cpy_smaller] duration = 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m- PASS - [test_spsc_pbuf.test_largest_alloc] duration = 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m- PASS - [test_spsc_pbuf.test_spsc_pbuf_ut] duration = 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m- PASS - [test_spsc_pbuf.test_spsc_pbuf_ut_cache] duration = 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m- PASS - [test_spsc_pbuf.test_stress] duration = 4.686 seconds[0m
[2023-03-23T01:22:13.785Z] [0m- FAIL - [test_spsc_pbuf.test_stress_0cpy] duration = 17.521 seconds[0m
[2023-03-23T01:22:13.785Z] [0m- PASS - [test_spsc_pbuf.test_utilization] duration = 0.001 seconds[0m
[2023-03-23T01:22:13.785Z] [0m------ TESTSUITE SUMMARY END ------[0m
[2023-03-23T01:22:13.785Z] [0m===================================================================[0m
[2023-03-23T01:22:13.785Z] [0mRunID: ab40a64ac3bc2b6d4ae74c1f54f7c396[0m
[2023-03-23T01:22:13.785Z] [0mPROJECT EXECUTION FAILED[0m
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.15.2
 - Commit SHA or Version used: zephyr-v3.3.0-1516-g1dce3c3ee2e9

nordic-krch, 2023-04-04:
Run it few times but could not reproduce.

nordic-piks, 2023-04-04:
> Run it few times but could not reproduce.

Checking at CI, current reproduction rate is 1/11. Can you do more testing?

katgiadla, 2023-04-27:
The same issue occurred on the board `nrf5340dk_nrf5340_cpuapp_ns` for scenario `tests/lib/spsc_pbuf/libraries.spsc_pbuf`.

```
[2023-04-26T22:03:33.265Z] [0mERROR   - *** Booting Zephyr OS build zephyr-v3.3.0-3164-ga0ad7b7752e6 ***[0m
[2023-04-26T22:03:33.265Z] [0mRunning TESTSUITE test_spsc_pbuf[0m
[2023-04-26T22:03:33.265Z] [0m===================================================================[0m
[2023-04-26T22:03:33.265Z] [0mSTART - test_0cpy[0m
[2023-04-26T22:03:33.265Z] [0mPASS - test_0cpy in 0.001 seconds[0m
[2023-04-26T22:03:33.265Z] [0m===================================================================[0m
[2023-04-26T22:03:33.265Z] [0mSTART - test_0cpy_corner1[0m
[2023-04-26T22:03:33.265Z] [0mPASS - test_0cpy_corner1 in 0.001 seconds[0m
[2023-04-26T22:03:33.265Z] [0m===================================================================[0m
[2023-04-26T22:03:33.265Z] [0mSTART - test_0cpy_corner2[0m
[2023-04-26T22:03:33.265Z] [0mPASS - test_0cpy_corner2 in 0.001 seconds[0m
[2023-04-26T22:03:33.265Z] [0m===================================================================[0m
[2023-04-26T22:03:33.265Z] [0mSTART - test_0cpy_discard[0m
[2023-04-26T22:03:33.265Z] [0mPASS - test_0cpy_discard in 0.001 seconds[0m
[2023-04-26T22:03:33.265Z] [0m===================================================================[0m
[2023-04-26T22:03:33.265Z] [0mSTART - test_0cpy_smaller[0m
[2023-04-26T22:03:33.265Z] [0mPASS - test_0cpy_smaller in 0.001 seconds[0m
[2023-04-26T22:03:33.265Z] [0m===================================================================[0m
[2023-04-26T22:03:33.265Z] [0mSTART - test_largest_alloc[0m
[2023-04-26T22:03:33.265Z] [0mPASS - test_largest_alloc in 0.001 seconds[0m
[2023-04-26T22:03:33.265Z] [0m===================================================================[0m
[2023-04-26T22:03:33.265Z] [0mSTART - test_spsc_pbuf_ut[0m
[2023-04-26T22:03:33.265Z] [0mPASS - test_spsc_pbuf_ut in 0.001 seconds[0m
[2023-04-26T22:03:33.265Z] [0m===================================================================[0m
[2023-04-26T22:03:33.265Z] [0mSTART - test_spsc_pbuf_ut_cache[0m
[2023-04-26T22:03:33.265Z] [0mPASS - test_spsc_pbuf_ut_cache in 0.001 seconds[0m
[2023-04-26T22:03:33.265Z] [0m===================================================================[0m
[2023-04-26T22:03:33.265Z] [0mSTART - test_stress[0m
[2023-04-26T22:03:33.265Z] [0mReading from an interrupt, writing from a thread[0m
[2023-04-26T22:03:33.265Z] [0m0% remaining:14000 ms[0m
[2023-04-26T22:03:33.265Z] [0m0% remaining:13000 ms[0m
[2023-04-26T22:03:33.265Z] [0mZtress execution report:[0m
[2023-04-26T22:03:33.265Z] [0mcontext 0:[0m
[2023-04-26T22:03:33.265Z] [0m- executed:23683, pre[0m
[2023-04-26T22:03:33.265Z] [0m0% remaining:12000 msempted:0[0m
[2023-04-26T22:03:33.265Z] [0m- ticks initial:4, optimized:4[0m
[2023-04-26T22:03:33.265Z] [0mcontext 1:[0m
[2023-04-26T22:03:33.265Z] [0m- executed:6801, preempted:2001[0m
[2023-04-26T22:03:33.265Z] [0m- ticks initial:4, optimized:4[0m
[2023-04-26T22:03:33.265Z] [0mAverage CPU load:32%, measurements:30[0m
[2023-04-26T22:03:33.265Z] [0mWrites:10174 failures: 0[0m
[2023-04-26T22:03:33.265Z] [0mWriting from an interrupt, reading from a thread[0m
[2023-04-26T22:03:33.265Z] [0m100% remaining:14000 ms[0m
[2023-04-26T22:03:33.265Z] [0mZtress execution report:[0m
[2023-04-26T22:03:33.265Z] [0mcontext 0:[0m
[2023-04-26T22:03:33.265Z] [0m- executed:8436, preempted:0[0m
[2023-04-26T22:03:33.265Z] [0m- ticks initial:4, optimized:4[0m
[2023-04-26T22:03:33.265Z] [0mcontext 1:[0m
[2023-04-26T22:03:33.265Z] [0m- executed:2283, preempted:1000[0m
[2023-04-26T22:03:33.265Z] [0m- ticks initial:4, optimized:4[0m
[2023-04-26T22:03:33.265Z] [0mAverage CPU load:42%, measurements:10[0m
[2023-04-26T22:03:33.265Z] [0mWrites:15842 failures: 7008[0m
[2023-04-26T22:03:33.265Z] [0mPASS - test_stress in 4.151 seconds[0m
[2023-04-26T22:03:33.265Z] [0m===================================================================[0m
[2023-04-26T22:03:33.265Z] [0mSTART - test_stress_0cpy[0m
[2023-04-26T22:03:33.265Z] [0m8% remaining:14000 ms[0m
[2023-04-26T22:03:33.265Z] [0m20% remaining:13000 ms[0m
[2023-04-26T22:03:33.265Z] [0m33% remaining:12000 ms[0m
[2023-04-26T22:03:33.265Z] [0m43% remaining:11000 ms[0m
[2023-04-26T22:03:33.265Z] [0m52% remaining:10000 ms[0m
[2023-04-26T22:03:33.265Z] [0m62% remaining:9000 ms[0m
[2023-04-26T22:03:33.265Z] [0m74% remaining:8000 ms[0m
[2023-04-26T22:03:33.265Z] [0m85% remaining:7000 ms[0m
[2023-04-26T22:03:33.265Z] [0m96% remaining:6000 ms[0m
[2023-04-26T22:03:33.265Z] [0mZtress execution report:[0m
[2023-04-26T22:03:33.265Z] [0mcontext 0:[0m
[2023-04-26T22:03:33.265Z] [0m- executed:36028, preempted:0[0m
[... 15 lines cut ...]
[2023-04-26T22:03:33.265Z] [0mFAIL - test_stress_0cpy in 19.016 seconds[0m
[2023-04-26T22:03:33.265Z] [0m===================================================================[0m
[2023-04-26T22:03:33.265Z] [0mSTART - test_utilization[0m
[2023-04-26T22:03:33.265Z] [0mPASS - test_utilization in 0.001 seconds[0m
[2023-04-26T22:03:33.265Z] [0m===================================================================[0m
[2023-04-26T22:03:33.265Z] [0mTESTSUITE test_spsc_pbuf failed.[0m
[2023-04-26T22:03:33.265Z] [0m------ TESTSUITE SUMMARY START ------[0m
[... cut ...]
[2023-04-26T22:03:33.265Z] [0m- FAIL - [test_spsc_pbuf.test_stress_0cpy] duration = 19.016 seconds[0m
[2023-04-26T22:03:33.265Z] [0m- PASS - [test_spsc_pbuf.test_utilization] duration = 0.001 seconds[0m
[2023-04-26T22:03:33.265Z] [0m------ TESTSUITE SUMMARY END ------[0m
[2023-04-26T22:03:33.265Z] [0m===================================================================[0m
[2023-04-26T22:03:33.265Z] [0mRunID: 6458b3e1649a38a66c5f941f1ae35424[0m
[2023-04-26T22:03:33.265Z] [0mPROJECT EXECUTION FAILED[0m
[2023-04-26T22:03:33.265Z] [0m[0m
```