zephyrproject-rtos/zephyr #56153: shell is unusable if used with UART that has flow control (whereby RTS is not asserted at boot time)

nordicjm, 2023-03-23:
**Describe the bug**
If a board has a UART whereby hardware flow control is enabled and the shell is enabled, if the board boots and sees RTS is de-asserted (i.e. cannot send), the shell is unresponsive, even if RTS then becomes asserted later on. It will not respond to any input.

Seemingly, once it receives enough input to fill an rx buffer and log an error, it then starts responding again.

**To Reproduce**
Build smp_svr with shell enabled for a board with flow control on UART enabled, boot board without serial port open, then open serial port, send commands or garbage data and press enter.

**Expected behavior**
Shell to work

**Impact**
Annoyance

**Logs and console output**
Output when sending the same thing over and over before it starts working again:
```
test command
test command
test command
test command
test command
test command
[\1b[0
0
\1b:00:06.105,560] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
[00:00:06.105,621] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
[00:00:06.374,420] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
[00:00:06.374,511] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
[00:00:06.374,603] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
[00:00:06.374,694] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
[00:00:06.374,786] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
[00:00:06.374,877] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
[00:00:06.374,938] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
[00:00:06.375,030] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
[00:00:06.375,122] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
[00:00:06.375,213] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
[1;32muart:~$ \1b[mtest command
\1b[1;31mtest: command not found
\1b[m\1b[1;32muart:~$ \1b[mtest command
\1b[1;31mtest: command not found
\1b[m\1b[1;32muart:~$ \1b[mtest comma[nd
\1b[00:00:06.375,274] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
;31mtest: command not found
\1b[m\1b[1;32muart:~$ \1b[mtest command
\1b[1;31mtest: command not found
\1b[m\1b[1;32muart:~$ \1b[mtest command\1b[20D\1b[J[00:00:06.105,560] \1b[1;33m[<w0>0:00:06.375,366] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
[00:00:06.375,457] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
 shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[mtest command\1b[20D\1b[J[00:00:06.105,621] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[mtest command\1b[20D\1b[J[00:00:06.374,420] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[mtest command\1b[20D\1b[J[00:00:06.374,511] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[mtest commandtest command
\1b[20D\1b[J[00:00:06.374,603] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[mtest command\1b[20D\1b[J[00:00:06.374,694] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[mtest commandtest command
\1b[1;31mtest: command not found
\1b[m\1b[1;32muart:~$ \1b[m\1b[8D\1b[J[00:00:06.374,786] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[m\1b[8D\1b[J[00:00:06.374,877] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[m\1b[8D\1b[J[00:00:06.374,938] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[m\1b[8D\1b[J[00:00:06.375,030] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[m\1b[8D\1b[J[00:00:06.375,122] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[m\1b[8D\1b[J[00:00:06.375,213] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[m\1b[8D\1b[J[00:00:06.375,274] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[m\1b[8D\1b[J[00:00:06.375,366] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[m\1b[8D\1b[J[00:00:06.375,457] \1b[1;33m<wrn> shell_uart: RX ring buffer full.\1b[0m
\1b[1;32muart:~$ \1b[m\1b[8D\1b[J\1b[1;32muart:~$ \1b[m
```

**Environment (please complete the following information):**
 - OS: Linux
 - Commit SHA or Version used: 67d851fb648acb5bea1f0a26a7c369437684d66f

nashif, 2023-05-30:
@jakub-uC  Can you please take a look and see if this is still bug?

jakub-uC, 2023-05-30:
@nordicjm  which board did you use?

closed 2023-10-21 by github-actions[bot]