zephyrproject-rtos/zephyr #56162: codeowners: bitbucket syntax error complain

MichaelZaidman, 2023-03-23:
The BitBucket pull request checker complains about:
    
    Code Owners failed. Syntax error CODEOWNERS#566 : Invalid Code Owner rule Caused by: Missing or invalid identifier: Use a valid user or group identifier. Caused by: Invalid identifier: Bitbucket user name Caused by: '@' expected but ',' found

It is caused by a comma in the file name introduced by commit 4be438254a51e230e929055504fc4871a96f478d last month.

        codeowners: Add code owner for INTEL FPGA NIOSV

Removing this line in our clone, as shown below, solved the problem, but the better fix would be removing the comma from the file names.

```
diff --git a/CODEOWNERS b/CODEOWNERS
index 5481798968..8f0055eb65 100644
--- a/CODEOWNERS
+++ b/CODEOWNERS
@@ -563,7 +563,6 @@
 /dts/bindings/sensor/*ina23*              @bbilas
 /dts/bindings/sensor/st*                  @avisconti
 /dts/bindings/smbus/                      @finikorg
-/dts/bindings/cpu/intel,niosv.yaml        @sweeaun
 /dts/common/                              @galak
 /include/                                 @nashif @carlescufi @galak @MaureenHelm
 /include/zephyr/drivers/*/*litex*                @mateusz-holenko @kgugala @pgielda
```

stephanosio, 2023-03-23:
Related to https://github.com/zephyrproject-rtos/zephyr/issues/16000

stephanosio, 2023-03-23:
We used to have a CI check to prevent this exactly issue (comma in CODEOWNERS file); but, that was removed in https://github.com/zephyrproject-rtos/zephyr/pull/53219.

@nashif @carlescufi Should we bring back the CODEOWNERS check in the `check_compliance.py` until we decide to get rid of it for good?

carlescufi, 2023-05-19:
> @nashif @carlescufi Should we bring back the CODEOWNERS check in the `check_compliance.py` until we decide to get rid of it for good?

I don't think so, even GitHub tells us that our syntax is invalid. First we'd need to clean it up, in my opinion.

closed 2023-08-03 by github-actions[bot]