zephyrproject-rtos/zephyr #56163: posix_common derived test fails on some qemu platforms when the CONFIG_TICKLESS_KERNEL=n is added

hakehuang, 2023-03-23:
**Describe the bug**
posix_common derived test fails on some qemu platforms when the CONFIG_TICKLESS_KERNEL=n is added

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 qemu_cortex_r5 and qemu_cortex_a53
 - What have you tried to diagnose or workaround this issue?
 see discussion https://github.com/zephyrproject-rtos/zephyr/pull/55208 
1. add a delay in the test_pthread_descriptor_leak loop can help
2. use tickless mode can work as well
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
No.

**To Reproduce**
Steps to reproduce the behavior:
1. add CONFIG_TICKLESS_KERNEL=n to prj.conf
2.  do below
```
west build -b qemu_cortex_r5 -- -DCONFIG_NEWLIB_LIBC=y
west build -t run
```

**Expected behavior**
PASS

**Impact**
some potential race condition in pthread handling

**Logs and console output**

```
*** Booting Zephyr OS build zephyr-v3.3.0-1516-g1dce3c3ee2e9 ***
Running TESTSUITE posix_apis
===================================================================
START - fnmatch
 PASS - fnmatch in 0.003 seconds
===================================================================
START - test_clock_gettime_rollover
CONFIG_SYS_CLOCK_TICKS_PER_SEC: 100
rollover_s: 184467440737095516
t-1: {184467440737095516, 140000000}
t+0: {184467440737095516, 150000000}
t+1: {0, 0}
 PASS - test_clock_gettime_rollover in 0.054 seconds
===================================================================
START - test_nanosleep_0_1
 PASS - test_nanosleep_0_1 in 0.018 seconds
===================================================================
START - test_nanosleep_0_1000000000
 PASS - test_nanosleep_0_1000000000 in 0.001 seconds
===================================================================
START - test_nanosleep_0_1001
 PASS - test_nanosleep_0_1001 in 0.017 seconds
===================================================================
START - test_nanosleep_0_500000000
 PASS - test_nanosleep_0_500000000 in 0.509 seconds
===================================================================
START - test_nanosleep_0_n1
 PASS - test_nanosleep_0_n1 in 0.001 seconds
===================================================================
START - test_nanosleep_1_0
 PASS - test_nanosleep_1_0 in 1.008 seconds
===================================================================
START - test_nanosleep_1_1
 PASS - test_nanosleep_1_1 in 1.020 seconds
===================================================================
START - test_nanosleep_1_1001
 PASS - test_nanosleep_1_1001 in 1.020 seconds
[... 152 lines cut ...]
Handler Signal value :20 for 12 times
 PASS - test_posix_timer in 2.121 seconds
===================================================================
START - test_pthread_descriptor_leak

    Assertion failed at WEST_TOPDIR/zephyr/tests/posix/common/src/pthread.c:598: posix_apis_test_pthread_descriptor_leak: (pthread_join(pthread1, &unused) is non-zero)
unable to join thread 31
 FAIL - test_pthread_descriptor_leak in 0.032 seconds
===================================================================
START - test_sleep
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/kernel/sched.c:1797
        aborted _current back from dead
E: r0/a1:  0x00000004  r1/a2:  0x00000705  r2/a3:  0xff00002c
E: r3/a4:  0x000089b9 r12/ip:  0x000242e8 r14/lr:  0x0000b2a1
E:  xpsr:  0x0000013f
E: Faulting instruction address (r15/pc): 0x0000c934
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x20038 (unknown)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, )
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v3.3.0-1516-g1dce3c3ee2e9

cfriedt, 2023-04-07:
On commit 336803e96f547747b9cec4511d1767c4c30a7b52, I'm unable to reproduce this on `main` with the command below (also tried `CONFIG_TICKLESS_KERNEL=n` in `prj.conf`).
```shell
west build -p always -b qemu_cortex_a53 -t run tests/posix/common -- -DCONFIG_TICKLESS_KERNEL=n -DCONFIG_NEWLIB_LIBC=y
```

Oddly, what I have found with `qemu_cortex_r5` is that if disable other tests within the same testsuite, the test passes. E.g.
```shell
for i in tests/posix/common/src/*.c; do j=$(basename $i); if [ "$j" = "main.c" -o $j = "pthread.c" ]; then continue; fi; k=$i.ignore; mv $i $k; done
west build -p always -b qemu_cortex_r5 -t run tests/posix/common -- -DCONFIG_TICKLESS_KERNEL=n -DCONFIG_NEWLIB_LIBC=y
```

I am able to reproduce this on `qemu_cortex_r5` without disabling other parts of the testsuite though.

That leads me to believe we are looking at possible stack corruption (stack overflow). Another alternative is that there may be some uninitialized variable.

cfriedt, 2023-04-07:
I tried the following, which sets stack sizes to a minimum size, but the problem persisted.

```shell
git checkout .
west build -p always -b qemu_cortex_r5 tests/posix/common -- \
    -DCONFIG_TICKLESS_KERNEL=n -DCONFIG_NEWLIB_LIBC=y
  grep STACK build/zephyr/.config | \
  grep -v "^#" | \
  grep -v "=\(y\|=n\)$" | \
  while read -r LINE; do \
    VAL="$(echo "$LINE" | awk '{split($0,a,"="); print a[2];}')" \
    if [ $((2*VAL)) -lt 4096 ]; then \
      NEWVAL=8192 \
    else \
      NEWVAL=$((2*VAL)) \
    fi \
    CONFIG="$(echo "$LINE" | awk '{split($0,a,"="); print a[1];}')" \
    echo "$CONFIG=$NEWVAL" \
  done | \
tee -a tests/posix/common/prj.conf \
west build -p always -b qemu_cortex_r5 -t run tests/posix/common -- \
    -DCONFIG_TICKLESS_KERNEL=n -DCONFIG_NEWLIB_LIBC=y
```

hakehuang, 2023-04-11:
> That's very odd that this issue only presents itself with `qemu_cortex_r5` as well. @hakehuang - this happens with NXP hardware too, right?

no, only in QEMU or renode platform

hakehuang, 2023-04-11:
> Oddly, what I have found with `qemu_cortex_r5` is that if I disable other tests within the same testsuite, the test passes. E.g.

if the pthread test is removed, all other will not fail, and per my testing it is only the last test case in the pthread which will cause problem for itself or for following test, a simple way to check is add a sleep right after the test. which will trigger the problem.

cfriedt, 2023-04-29:
@hakehuang @dleach02 - please take a look at #57403 - it seems the solution was a lot easier than we anticipated.

cfriedt, 2023-05-02:
Re-opening because it still is not fixed.
I was able to exacerbate the problem by greatly increasing the number of `pthread_create()` / `pthread_join()` requests up to 100 times the maximum number of pthreads.

It should not be necessary to re-initialize the same attribute over and over again as well.

Just to be clear, we are not actually leaking descriptors, but we are seeing the effects of some kind of race condition between `pthread_create()`, and `pthread_join()`.

```cpp
ZTEST(posix_apis, test_pthread_descriptor_leak)
{
	pthread_t pthread1;
	pthread_attr_t attr;

	zassert_ok(pthread_attr_init(&attr));
	zassert_ok(pthread_attr_setstack(&attr, &stack_e[0][0], STACKS));

	/* If we are leaking descriptors, then this loop will never complete */
	for (size_t i = 0; i < CONFIG_MAX_PTHREAD_COUNT * 100; ++i) {
		zassert_ok(pthread_create(&pthread1, &attr, create_thread1, NULL),
			   "unable to create thread %zu", i);
		zassert_ok(pthread_join(pthread1, NULL), "unable to join thread %zu", i);
	}
}
```

The same thing definitely does not occur with `k_thread_create()` and `k_thread_join()`, so there is definitely a bug here, and it exists in `main` as well as `v2.7-branch`. 

It will likely help to create a reliable testcase where we're not executing the entire `tests/posix/common` suite.  Something like the above code snippet

cfriedt, 2023-05-02:
Setting to priority medium as it affects multiple platforms.

cfriedt, 2023-05-07:
#57637 can be used to more reliably reproduce the race condition. Still working on a fix.

andyross, 2023-05-09:
See commit message in linked PR.  The root cause was a hand-rolled attempt at pthread_join() when it needed to be using the underlying k_thread_join() instead.

cfriedt, 2023-05-21:
As it turns out, there is also a race between `k_thread_create()` and `k_thread_join()`

andyross, 2023-05-21:
Can you file that as a bug?  There really shouldn't be as I'm reading the code.

cfriedt, 2023-05-22:
> Can you file that as a bug? There really shouldn't be as I'm reading the code.

@andyross - created https://github.com/zephyrproject-rtos/zephyr/issues/58116 for you

cfriedt, 2023-06-05:
Just wanted to note here that although this issue has highlighted an underlying issue with `k_thread_create()` and `k_thread_join()` which has been mitigated (at least made so that the probabilities are quite negligible in parts per billion), the race still very much exists in pthreads.

Working on a fix for this which I hope will be accepted into v3.4.0

cfriedt, 2023-06-06:
Probably should not have been closed

PR #57403 (2023-04-29): tests: posix: pthread: init pthread_attr_t on each iteration
The `test_pthread_descriptor_leak` test was causing a kernel panic on some platforms. Initially, it was not clear why.

The usual cases were examined - race conditions, stack sizes, etc. Still no luck.

As it turns out, recycling a thread stack (or at least the `pthread_attr_t`) in-place does not work on some platforms, and we need to reinitialize the `pthread_attr_t` and set set the stack property again prior to calling `pthread_create()`.

Fixes #56163

Tested with:
```shell
west build -p always -b qemu_cortex_r5 -t run tests/posix [... cut]
PR #58924 (2023-06-09): posix: pthread: leverage zephyr's synchronization and threading api
**Note: USE_NEGATIVE_ERRNO compliance failures are false positives (pthread functions return +ve errno values)**

7 commits. Please see each commit message for details.

* posix: pthread: reimplement pthread_barrier using zephyr objects
* posix: mutex: use k_mutex instead of posix_mutex
* posix: cond: use k_condvar instead of posix_cond
* posix: pthread: rework pthreads to use zephyr api
* posix: pthread: add option for pthread_create() barrier
* tests: posix: improvements to pthread_pressure test
* posix: pthread: mitigate pt [... cut]
closed 2023-06-09 by nashif