zephyrproject-rtos/zephyr #56171: In some condition writing to CAN->ECR register crashes kernel on NXP platform

awojak, 2023-03-24:
**Describe the bug**
Found problem in FLEXCAN on RT1062xxxxB. When CAN Rx line is pulled to GND (some of transceivers + isolation is doing this in low power) writing to ECR register in mcux_flexcan_start() crash kernel. The problem first was observed on our custom hardware, but can be easily reproduced on MIMXRT1060-EVKB board.

I found exact place of the problem. Writing to CAN->ECR register in function mcux_flexcan_start() line 210 cause kernel crashes.
> config->base->ECR &= ~(CAN_ECR_TXERRCNT_MASK | CAN_ECR_RXERRCNT_MASK);

When I commented out this line of code, problem is not observe anymore.

**To Reproduce**

Problem could be reproduce on MIMXRT1060-EVKB. Modified version of the `blinky` sample was used.

boards/mimxrt1060_evkb.overlay
```c
&pinctrl {  
   pinmux_flexcan2: pinmux_flexcan2 {  
      group0 {  
         pinmux = <&iomuxc_gpio_ad_b0_05_gpio1_io05>;  
         bias-disable;  
         drive-strength = "r0";  
         slew-rate = "slow";  
         nxp,speed = "50-mhz";  
      };  
      group1 {  
         pinmux = <&iomuxc_gpio_ad_b0_14_flexcan2_tx>,  
            <&iomuxc_gpio_ad_b0_15_flexcan2_rx>;  
         drive-strength = "r0-3";  
         slew-rate = "slow";  
         nxp,speed = "50-mhz";  
      };  
   };  
};  
  
&flexcan2 {  
   pinctrl-0 = <&pinmux_flexcan2>;  
   pinctrl-names = "default";  
   status = "okay";  
   bus-speed = <250000>;  
   phase-seg1 = <7>;  
   phase-seg2 = <2>;  
   prop-seg = <6>;  
};
```
prj.conf
```
CONFIG_GPIO=y  
CONFIG_SHELL=y  
CONFIG_CAN=y
```
1. Open your favorite terminal and connect to the Zephyr serial console.
2. `can start can@401d0000`
3. In this step needed is to pulled RX line to GND. Can be simulated by connect CAN transceiver U23 Pin4 to GND. 
4. `can stop can@401d0000`
5. `can start can@401d0000`
6. see error (kernel crash)

**Logs and console output**
```
uart:~$ *** Booting Zephyr OS build zephyr-v3.2.0  ***
uart:~$ can start can@401d4000 
starting can@401d4000
uart:~$ can stop can@401d4000
stopping can@401d4000
uart:~$ can start can@401d4000
starting can@40
```

**Environment (please complete the following information):**
 - OS: Debian 11
 - Toolchain Zephyr SDK 0.15.0
 - Zephyr 3.2.0

henrikbrixandersen, 2023-03-24:
@awojak Thank you for this detailed bug report. I will look into a solution which still allows us to clear the error counters on start.

henrikbrixandersen, 2023-06-13:
I have encountered a similar issue on the `mimxrt1060_evkb` when running the `samples/subsys/canbus/isotp/` sample (all other in-tree CAN samples/tests pass on this board):

```
*** Booting Zephyr OS build v3.4.0-rc3-21-gbfa0a87277c3 ***
Start sending data
[00:00:01.001,000] <err> os: ***** BUS FAULT *****
[00:00:01.001,000] <err> os:   Imprecise data bus error
[00:00:01.001,000] <err> os: r0/a1:  0x401d8000  r1/a2:  0x00000224  r2/a3:  0x51801f80
[00:00:01.001,000] <err> os: r3/a4:  0x54a3003f r12/ip:  0x80000b54 r14/lr:  0x600068c9
[00:00:01.001,000] <err> os:  xpsr:  0x61000000
[00:00:01.001,000] <err> os: Faulting instruction address (r15/pc): 0x6000a900
[00:00:01.001,000] <err> os: >>> ZEPHYR FATAL ERROR 26: Unknown error on CPU 0
[00:00:01.001,000] <err> os: Current thread: 0x80000940 (unknown)
[00:000,000] <err> os: Halting system
```

```
$ addr2line -e zephyr/zephyr.elf 0x6000a900
/home/brix/Projects/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/flexcan/fsl_flexcan.c:528
```

Line 528 in `/home/brix/Projects/zephyrproject/modules/hal/nxp/mcux/mcux-sdk/drivers/flexcan/fsl_flexcan.c` is in `FLEXCAN_ExitFreezeMode()` where the HALT bit is cleared in the MCR register (`base->MCR &= ~CAN_MCR_HALT_MASK;`):

```
void FLEXCAN_ExitFreezeMode(CAN_Type *base)
{
#if (defined(FSL_FEATURE_FLEXCAN_HAS_MEMORY_ERROR_CONTROL) && FSL_FEATURE_FLEXCAN_HAS_MEMORY_ERROR_CONTROL)
    /* Clean FlexCAN Access With Non-Correctable Error Interrupt Flag to avoid be put in freeze mode. */
    FLEXCAN_ClearStatusFlags(base, (uint64_t)kFLEXCAN_FlexCanAccessNonCorrectableErrorIntFlag |
                                       (uint64_t)kFLEXCAN_FlexCanAccessNonCorrectableErrorOverrunFlag);
#endif

    /* Clear Freeze, Halt bits. */
    base->MCR &= ~CAN_MCR_HALT_MASK;
    base->MCR &= ~CAN_MCR_FRZ_MASK;

    /* Wait until the FlexCAN Module exit freeze mode. */
    while (0U != (base->MCR & CAN_MCR_FRZACK_MASK))
    {
    }
}

```

henrikbrixandersen, 2023-06-13:
@dleach02 I'm afraid I am out of ideas on this one. Can you get someone internally in NXP to take a look, please?

awojak, 2023-06-15:
@henrikbrixandersen I found workaround for this problem and for now works fine. So far, no kernel crashes have been observed.
Solution is to not use errata 9595 fix. In file `mcux/mcux-sdk/devices/MIMXRT1062/MIMXRT1062_features.h`, just set `#define FSL_FEATURE_FLEXCAN_HAS_ERRATA_9595 (0)`

henrikbrixandersen, 2023-06-16:
> @henrikbrixandersen I found workaround for this problem and for now works fine. So far, no kernel crashes have been observed.
> Solution is to not use errata 9595 fix. In file `mcux/mcux-sdk/devices/MIMXRT1062/MIMXRT1062_features.h`, just set `#define FSL_FEATURE_FLEXCAN_HAS_ERRATA_9595 (0)`

Good catch! I can confirm that defining `FSL_FEATURE_FLEXCAN_HAS_ERRATA_9595` to `(0)` also gets rid of the crash when writing to `base->MCR`. I suspect some code for addressing errata 9595 puts the FlexCAN controller into low-power mode, but fails to bring it out of low-power mode again, causing these register writes to fail.

henrikbrixandersen, 2023-06-16:
@dleach02 I've played a bit with `FLEXCAN_EnterFreezeMode()` as defined for Errata 9595 in https://github.com/zephyrproject-rtos/hal_nxp/blob/904830e8f684a9fd573751a1cdecde877ec49242/mcux/mcux-sdk/drivers/flexcan/fsl_flexcan.c#L371

One thing that strikes me, is that the `u32TimeoutCount` variable is never checked for being `0` after the loops - so a timeout has no effect on the remaining code path (and debugging shows that they do indeed timeout). The problem seems to depend on the timing of that function. Adding debugging statements changes the timing and makes the problem go away.

henrikbrixandersen, 2023-06-16:
Simply increasing `FLEXCAN_WAIT_TIMEOUT` from `1000` to `10000` also makes the immediate issue disappear.

henrikbrixandersen, 2023-07-26:
> Simply increasing `FLEXCAN_WAIT_TIMEOUT` from `1000` to `10000` also makes the immediate issue disappear.

@dleach02 Any input on this?

dleach02, 2023-08-09:
Reaching out to the internal SDK team to understand why this errata flag is set true for this SOC. I'm referencing them to this issue as well.

PR #63863 (2023-10-13): drivers: can: mcux: flexcan: override maximum HAL wait loop iterations
Add option for overriding the maximum number of wait loop iterations for entering/leaving freeze mode. Set the default to 10000 (as opposed to a default of 1000 used in the HAL).

Fixes: #56171
commit b9675b05 (2023-10-13): drivers: can: mcux: flexcan: override maximum HAL wait loop iterations

Add option for overriding the maximum number of wait loop iterations for
entering/leaving freeze mode. Set the default to 10000 (as opposed to a
default of 1000 used in the HAL).

Fixes: #56171