zephyrproject-rtos/zephyr #56197: BLE peripheral connection lost when configured with long

fabiobaltieri, 2023-03-24:
**Describe the bug**

Hi, bumped into an issue trying to setup a BLE peripheral with long connection intervals on `main` using the Zephyr BLE controller. It seems like the connection gets dropped by device shortly after the connection parameters updated event that brings up the intervals. The connection fails with the error `Disconnected (reason 0x08)` (`BT_HCI_ERR_CONN_TIMEOUT`).

I'm testing with a nRF52840DK v1.1.0 and an nRF52DK v1.1.1, as well as with a custom board. On the central side I used nRF Connect on a Pixel 6, as well as bluez on a Linux laptop (bluetoothctl: 5.66).

This has been discussed on discord and it seems like the issue was not there with the legacy LLCP stack in v3.3. These are the cases I've tested:

```
$ git describe
zephyr-v3.3.0-1669-g8bd0e13fc7
$ west build -p -b nrf52840dk_nrf52840 samples/bluetooth/peripheral_dis
```

Default settings, connection goes up to 50ms interval, works fine.

```
$ git describe
zephyr-v3.3.0-1669-g8bd0e13fc7
$ west build -p -b nrf52840dk_nrf52840 samples/bluetooth/peripheral_dis -DCONFIG_BT_PERIPHERAL_PREF_MIN_INT=700 -DCONFIG_BT_PERIPHERAL_PREF_MAX_INT=800 -DCONFIG_BT_PERIPHERAL_PREF_TIMEOUT=1000
```

Updates to 1s intervals, drops with 0x08.

```
$ git checkout v3.3.0
$ west update
$ west build -p -b nrf52840dk_nrf52840 samples/bluetooth/peripheral_dis -DCONFIG_BT_PERIPHERAL_PREF_MIN_INT=700 -DCONFIG_BT_PERIPHERAL_PREF_MAX_INT=800 -DCONFIG_BT_PERIPHERAL_PREF_TIMEOUT=1000
```

Same, 1s interval, connection drops.

```
$ git describe
zephyr-v3.3.0
$ west build -p -b nrf52840dk_nrf52840 samples/bluetooth/peripheral_dis -DCONFIG_BT_PERIPHERAL_PREF_MIN_INT=700 -DCONFIG_BT_PERIPHERAL_PREF_MAX_INT=800 -DCONFIG_BT_PERIPHERAL_PREF_TIMEOUT=1000 -DCONFIG_BT_LL_SW_LLCP_LEGACY=y
```

Stable.

The same board also works reliably using the NCS softdevice controller.

Can you folks reproduce the problem as well? Let me know if you need more info, happy to take a trace with the ble sniffer, need to put a bit of time into setting it up though.

**To Reproduce**
See the steps above.

**Expected behavior**
Stable connection.

**Impact**
Probably showstopper if the issue is confirmed, I imagine this may impact stack compliancy, and it's obviously a blocker for super low power sensors.

**Logs and console output**
Console output:

```
*** Booting Zephyr OS build zephyr-v3.3.0 ***                                   
Bluetooth initialized                                                           
Advertising successfully started                                                
Connected                                                                       
Disconnected (reason 0x08)                                                      
Connected                                                                       
Disconnected (reason 0x08)                                                      
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.15.2
 - Commit SHA or Version used: 8bd0e13fc7 and v3.3.0

fabiobaltieri, 2023-03-24:
cc @carlescufi @cvinayak

fabiobaltieri, 2023-03-24:
BLE sniffer trace, looks like once the central switches to 1s interval the peripheral never responds:

[image]

[nrf-long-adv.pcapng.gz](https://github.com/zephyrproject-rtos/zephyr/files/11065080/nrf-long-adv.pcapng.gz)

jakkra, 2023-03-24:
Hi, just wanted to add that I have had this exact same issue for a while, I assumed it was an issue in my config or something and didn't do any debugging (not critical for my app yet).

So if you are not able to reproduce I can also support with radio trace and provide more information (tested with my two different One Plus phones).

fabiobaltieri, 2023-03-28:
Hey ran few more tests with this over the weekend: the fail against bluez seems flaky, seems like sometimes bluez does not update the connection parameters  at all and then it's stable, not sure what's going on there. Also tried against the nrfconnect desktop tool and that works just fine, so maybe it's a bug in the phone stack? Though it also fails when connected to my Garmin watch (that's the target application I was looking into when I stumbled into this), not too sure what's running in there but AFAIK it's some Cypress device (It's a Forerunner 955).

Anyway, let me know if you have troubles reproducing it, I have a working setup for taking traces now so happy to use it.

cvinayak, 2023-03-29:
@jakkra and @fabiobaltieri This is most likely a regression in the new LLCP implementation for nRF SoCs. Timing in Controller for nRF SoC use a coarse tick and a remainder fraction, the new LLCP is missing implementation to use the remainder fraction for achieve microsecond resolution in timing calculation when performing Connection Parameter Updates, this will be fixed soon.

Thank you both for reporting and help confirm the regression.

cvinayak, 2023-04-13:
it is unfortunate that the regression exists since in the switch of default to refactored LLCP in e1c2c36f56

cvinayak, 2023-04-14:
@fabiobaltieri @jakkra regret the delay. I have verified a fix for a simple error, by using a peripheral_ht sample that prefers 1 second connection interval with iphone 13 pro (uses CI 0x320) , old iphone 6s (uses CI 0x32a!) and nexus 5 phone (uses CI 0x006, 0x0027 and refuses to switch to 1s on 3 more retries). Could you verify that #56839 fixes the issue with your phones?

fabiobaltieri, 2023-04-14:
> Could you verify that #56839 fixes the issue with your phones?

Yes! Tested this with the `peripheral_dis` example against a Pixel 6, and also with my custom application against both a Pixel 6 and a Garmin Forerunner 955. Connection switches to 1s intervals and seems stable in both cases. Thanks!!

cvinayak, 2023-04-14:
@fabiobaltieri  Thank you for reporting and verifying the fix.

jakkra, 2023-04-14:
@cvinayak can confirm it solved the issue in my application too! Thanks!

cvinayak, 2023-04-14:
@jakkra Thank you too!

PR #56839 (2023-04-14): Bluetooth: Controller: Fix connection update window offset data type
Fix regression in refactored LLCP using uint16_t instead of uint32_t in storing the win_offset_us value. This caused connection update to fail with incorrect window offset being used to schedule the connection radio events.

Regression since commit e1c2c36f569a ("Bluetooth: controller:
llcp: set refactored as default").

Fixes #56197.
closed 2023-04-14 by carlescufi