zephyrproject-rtos/zephyr #56200: Always respect ZephyrBuildConfiguration_ROOT

SkryptX, 2023-03-24:
The default mechanism for finding the `ZephyrBuildConfig.cmake` is this:
```
find_package(ZephyrBuildConfiguration
  QUIET NO_POLICY_SCOPE
  NAMES ZephyrBuild
  PATHS ${ZEPHYR_BASE}/../*
  NO_CMAKE_PATH
  NO_CMAKE_ENVIRONMENT_PATH
  NO_SYSTEM_ENVIRONMENT_PATH
  NO_CMAKE_PACKAGE_REGISTRY
  NO_CMAKE_SYSTEM_PATH
  NO_CMAKE_SYSTEM_PACKAGE_REGISTRY
)
```

This can be very annoying, since the `${ZEPHYR_BASE}/../*` is hard-coded and, depending on the project structure, it will just find _some_ random `ZephyrBuildConfig.cmake` file, since there is no way to override the hard-coded path.

I would propose something like this:
```
if (NOT ZephyrBuildConfiguration_ROOT)
  set(ZephyrBuildConfiguration_ROOT "${ZEPHYR_BASE}/../*")
endif()

find_package(ZephyrBuildConfiguration
  QUIET NO_POLICY_SCOPE
  NAMES ZephyrBuild
  NO_CMAKE_PATH
  NO_CMAKE_ENVIRONMENT_PATH
  NO_SYSTEM_ENVIRONMENT_PATH
  NO_CMAKE_PACKAGE_REGISTRY
  NO_CMAKE_SYSTEM_PATH
  NO_CMAKE_SYSTEM_PACKAGE_REGISTRY
)
```

That way it will always respect the given root directory, even if it finds no file at that root directory. In my opinion, this a better default, because when you specify a root directory, you probably don't expect it to also search the default path.

nylnx, 2023-05-28:
@tejlmand Please take a look at my PR. It's only 5 changed lines :P If you would like some additional documentation about this change I can add it as well.

tejlmand, 2023-06-14:
> This can be very annoying, since the ${ZEPHYR_BASE}/../* is hard-coded and, depending on the project structure, it will just find some random ZephyrBuildConfig.cmake file,

not some random `ZephyrBuildConfig.cmake`, but a `ZephyrBuildConfig.cmake` within your Zephyr workspace, ref: https://docs.zephyrproject.org/latest/develop/west/workspaces.html

And the reason for this lookup is to allow users or high-level projects to provide a configuration, similar in nature to a  like how the `.zephyrrc` is provided, ref: https://docs.zephyrproject.org/latest/build/zephyr_cmake_package.html#zephyr-build-configuration-cmake-package

Zephyr itself doesn't provide a default `ZephyrBuildConfig.cmake` file, so either you have created such a file yourself or the project on which you are working on is providing such a file.
If the latter, and the file in question doesn't meet your needs or expectations, then you have the possibility of providing your own file, as you seem to be aware.
If you want to able to completely disable the provided config file, even though it is provided by the downstream project, then you should probably request the project in question to support an opt-out option.

> since there is no way to override the hard-coded path.

As you already noticed, there is the possibility to point `ZephyrBuildConfiguration_ROOT` to a different location which will be searched before the default paths.

CMake v3.27 even allows you to set `<package>_ROOT` in your environment, meaning you won't need to apply it on command-line on each build.

>  you probably don't expect it to also search the default path.

trying the default / hard-coded guess if `<package>_ROOT` fails is the default CMake behavior, and so far I have seen no compelling arguments that we should deviate from what CMake does, ref: https://cmake.org/cmake/help/latest/command/find_package.html#config-mode-search-procedure. 

> And if you insist on the old mechanism; just set `ZephyrBuildConfiguration_ROOT` to `<mycustomfirstguess>;${ZEPHYR_BASE}/../*`.

and that is the case today, the `ZephyrBuildConfiguration_ROOT` is searched **before** any hard-coded guesses.

tejlmand, 2023-06-14:
If there is a need for an enhancement to the current functionality then please provide a good use-case.

Also note that the `ZephyrBuildConfig.cmake` is intended for settings handling, as the name implies.
A project can in principle do anything it likes in such file, like target creation, but that would be a misuse of the intention of the feature.

As the example shows: https://docs.zephyrproject.org/latest/build/zephyr_cmake_package.html#zephyr-build-configuration-cmake-package

such use could be setting of default compiler if none is already defined.
Or adding extra BOARD_ROOT for projects (although doing so in a Zephyr module is a better approach).

If a project is doing something foolish, then it's probably better to make a request towards the project itself for an improved build config file.

nylnx, 2023-06-14:
@tejlmand Something "foolish" is using the recommended [T3: Forest Topology](https://docs.zephyrproject.org/latest/develop/west/workspaces.html#t3-forest-topology). Suppose you have a something like this (basically a copy from the T3 doc): 

```
t3_workspace
├── app1
│   ├── CMakeLists.txt
│   └── ZephyrBuildConfig.cmake
├── app2
│   └── CMakeLists.txt
│
└── zephyr
```
`app1` is happy, since it doesn't even need to set a `ZephyrBuildConfiguration_ROOT`. But `app2` now has to actively deter CMake from finding `ZephyrBuildConfig.cmake` and being applied to `app2`. It will always find it, except I implement measures to prevent that from happening like:

- Create own (dummy) `ZephyrBuildConfig.cmake` and set `ZephyrBuildConfiguration_ROOT` (this is the most secure way currently!)
- Use `CMAKE_DISABLE_FIND_PACKAGE_ZephyrBuildConfiguration` until I need to use such a file myself and then option above applies.
- Tell `app1` to move its ZephyrBuildConfig.cmake to an unintuitive folder which is not part of CMake's find paths and set `ZephyrBuildConfiguration_ROOT` on its own.
- Various even worse hacks... (or I forgot an easy elegant solution; please tell me).

My initial idea was to add this to every application as a way to "namespace" the search:
```
get_filename_component(ZephyrBuildConfiguration_ROOT ${CMAKE_CURRENT_LIST_DIR} ABSOLUTE)
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
```
But then I realized this is not helping since the default path still applies...

So I made a PR with the idea to keep the current, but in my opinion flawed, default path intact, while still allowing this kind of namespacing, to not make breaking changes.
 
But maybe this was the wrong proposition... Maybe we should just delete the default path altogether or replace it with a more sensible default, like instead of a *-search, just search in the application root or just the manifest repository to allow global configuration. I don't see a reason a user/an app should have to fight _against_ the inclusion of a custom build configuration, which should be reserved for advanced users, as you yourself implied above.

tejlmand, 2023-06-20:
@nylnx thanks for your comment, it helps a lot in understanding what your intentions are.

> ```
> t3_workspace
> ├── app1
> │   ├── CMakeLists.txt
> │   └── ZephyrBuildConfig.cmake
> ```

This structure helps a lot in understanding what you try to achieve.
The `ZephyrBuildConfig.cmake` was not intended to be on a per application build configuration, but more high level settings / environment kind of settings.
See it as a possibility to set "environment" settings without users requiring to do a `source zephyr-env.sh` as described here: https://docs.zephyrproject.org/latest/develop/env_vars.html#option-3-using-zephyrrc-files

The zephyr-env.sh would be identical across multiple applications, just like the `ZephyrBuildConfig` is intended to be used across applications.

But I do see that the name "BuildConfiguration" can give the impression that it can be used for individual applications.

The Zephyr build config is a very open-ended feature and allows users a lot of freedom.
Meaning that you are of course free to structure your apps as you which and use the feature as it suits you.

Let me give you a proposal on how to achieve what you want.

Create your structure as this:
```
t3_workspace
├── app1
│   ├── CMakeLists.txt
│   ├── ZephyrBuildConfig.cmake
│   └── ZephyrBuildConfigVersion.cmake
```
 
and inside `ZephyrBuildConfigVersion.cmake` you write:
```
# Only source Build configuration if building app1
if(NOT "${CMAKE_CURRENT_LIST_DIR}" STREQUAL "${CMAKE_SOURCE_DIR}")
  set(PACKAGE_VERSION_UNSUITABLE TRUE)
endif()
```

That will ensure that the corresponding `ZephyrBuildConfig.cmake` is sourced only when building for app1.

And app2 will be happy to, as it don't need to care about `app1/ZephyrBuildConfig.cmake` cause it will not be sourced when building app2.

tejlmand, 2023-06-20:
and as bonus, no sample needs to set `ZephyrBuildConfiguration_ROOT` 😉

nylnx, 2023-06-20:
@tejlmand Thank you for your answer and for the solution you gave!

But is this really the best we can do? A separate dummy version file for every `ZephyrBuildConfig.cmake` there is... 

I still feel there might be a place for a `ZephyrAppConfig.cmake` which is application-spaced. I do see now that `ZephyrBuildConfig.cmake` is (workspace-)global and should apply to all/most applications. But I think there are valid use-cases for both of those and a clear separation between those two options would make it clearer and easier to use in general.

How do you see this? Do you, as a collaborator/maintainer, think adding `ZephyrAppConfig.cmake` would make sense or just add unnecessary/additional complexity? Is the problem even a valid one?

I would be willing to help with the PR, but would certainly be a bit more involved than the current PR.

tejlmand, 2023-06-21:
> I still feel there might be a place for a `ZephyrAppConfig.cmake` which is application-spaced. I do see now that `ZephyrBuildConfig.cmake` is (workspace-)global and should apply to all/most [...]

yes, I believe a dedicated app config can be a valid feature, and I also believe it will be good to keep the two features independent. It gives a clear separation between the high level config, and the app specific.

A dedicated app config also allows us to improve an example like this:
https://github.com/zephyrproject-rtos/zephyr/blob/001795b24c980df3a3dfc0e7b385f93a9392d4f9/samples/application_development/out_of_tree_board/CMakeLists.txt#L4-L6

Having such setting in a app config will make config cleaner as well as solve a problem with sysbuild not knowing about app specific board roots. Sysbuild would be able to source such app config independent of the app's CMakeLists.txt file.

Just as it would allow [Zephyr package helper](https://github.com/zephyrproject-rtos/zephyr/blob/main/cmake/package_helper.cmake) to also be able to source such file and be aware of any extra settings.

> I would be willing to help with the PR, but would certainly be a bit more involved than the current PR.

that would be great, and I think such a solution will mostly be related to updating the docs, as the actual sourcing of such config will be similar to existing ZephyrBuildConfig code.

FYI @57300 @nordicjm

nylnx, 2023-07-12:
@jhedberg It's not a bug as @tejlmand stated above. At this point it's basically an RFC. But I think I will close the current linked PR, open a new one and change the title of this issue to reflect that somewhat. I just haven't had the time recently to create a fitting PR as discussion basis.

jhedberg, 2023-07-12:
> @jhedberg It's not a bug as @tejlmand stated above. At this point it's basically an RFC. But I think I will close the current linked PR, open a new one and change the title of this issue to reflect that somewhat. I just haven't had the time recently to create a fitting PR as discussion basis.

All issues need a category. This was processed in yesterday's release engineering meeting where issues are looked at only quite superficially, which then resulted in the bug categorisation. "RFC" is also a valid category, so I've now added that label and removed the bug label.

commit cd2a0389 (2023-08-17): cmake: add ZephyrAppConfiguration package

Fixes #56200
Add the ZephyrAppConfiguration package and update the corresponding
documentation. This adds flexibility to the CMake build configuration
by providing a workspace configuration package and an application
package, which only applies to the current application. The workspace
package stays the same as before, but the application package is
new and lives, per default, inside the application folder.
closed 2023-08-17 by carlescufi