zephyrproject-rtos/zephyr #56201: Bluetooth: controller: Possibility to have an active CIS and extended advertising at the same time

ladivin, 2023-03-24:
**Is your enhancement proposal related to a problem? Please describe.**
If an acceptor (peripheral) listens to a CIS from an initiator (central) and the acceptor starts extended advertising while the CIS is still active, the extended advertising may cause audio glitches (loss of ISO data). 

**Describe the solution you'd like**
Solution proposed by @cvinayak: currently primary channel ADV_EXT_IND PDU advertising nudges automatically around other radio events, the fix would do the same for auxiliary channel AUX_ADV_IND PDUs to nudge around any new peripheral ACL radio event (and not overlap).

**Describe alternatives you've considered**
None

**Additional context**
The motivation behind this request is to have the possibility for a peripheral to accept an audio stream from one central, then start extended advertising again and then based on certain criteria possibly to switch to a stream from a different central.

cvinayak, 2023-04-25:
ISO implementation is still lacking feature completion and conformance testing, this enhancement will be delayed. The strategy would be to implement a Babblesim CI test for the said usecase as a test driven development.

Test:
1. Develop a testcase with procedures to have two extended advertising sets active, one being connectable.
2. Establish an ACL connection and Create CIS from a central device
3. Verify that there is no lost ISO data reception on the peripheral

Implementation changes:
1. Extended Advertising Primary channel radio event, in a low priority execution context evaluates relative drift between AUX_ADV_IND, any peripheral ACL and CIG radio event
2. If the margin between is lower than a threshold, then use ticker_update interface on the AUX_ADV_IND to move it accordingly to a new position in timeline.

cvinayak, 2024-08-05:
@ladivin I have a draft implementation change in #75760 that can make the Extended Advertising to nudge around ACL connection events. Please give a try and let me know...

The feature gets enabled when using in the PR branch:
```
CONFIG_BT_CTLR_ADVANCED_FEATURES=y
CONFIG_BT_TICKER_EXT=y
CONFIG_BT_TICKER_EXT_SLOT_WINDOW_YIELD=y
CONFIG_BT_CTLR_ADV_AUX_SLOT_WINDOW=y
```