zephyrproject-rtos/zephyr #56212: API to set unix time for Real Time Clocks

vChavezB, 2023-03-25:
**Is your feature request related to a problem? Please describe.**
I want to develop a driver for an RTC. Currently the only API I have found is the one located in <zephyr/drivers/counter.h> named as `counter_driver_api`. The problem with this API is that there is no way to set the RTC unix time. I can only get the unix time by using the "get_value" prototype.

**Describe the solution you'd like**
An API to set the value of a counter or a separate API just for RTCs such as in the linux kernel https://github.com/torvalds/linux/blob/master/include/linux/rtc.h#L59

**Describe alternatives you've considered**
I have used temporarily the API `set_top_value` to set unix time of my RTC driver. It works but I am not sure I need the all the struct members from `counter_top_cfg` for this type of action. If using something similar as in the linux kernel you would have `read_time` and `set_time` API.

pdgendt, 2023-03-27:
There is an API proposal here: https://github.com/zephyrproject-rtos/zephyr/pull/52618

vChavezB, 2023-03-27:
Thanks for the link. I will be glad to wait when the PR is merged. In the mean while I will be using I guess the counter API.

closed 2023-03-27 by vChavezB