zephyrproject-rtos/zephyr #56230: RP2040, i2c : wrong clock frequency

Thisora, 2023-03-27:
**Describe the bug**

SCL clock frequency generated do not correspond to DT clock-frequency definition:
```
&i2c1 {
	status = "okay";
	pinctrl-0 = < &i2c1_default >;
	pinctrl-names = "default";
	clock-frequency = <I2C_BITRATE_STANDARD>;
...

```
I2C_BITRATE_STANDARD is suppose to be 100kHz, but i measured 124kHz :
[image]

Which make a huge difference.
I investigate first using i2c logs and i found that hcnt and lcnt are wrong:
               
```
[00:00:00.005,000] <dbg> i2c_dw: i2c_dw_isr: I2C: interrupt received                   
[00:00:00.005,000] <dbg> i2c_dw: i2c_dw_setup: I2C: host configured as Master Device   
[00:00:00.005,000] <dbg> i2c_dw: i2c_dw_setup: I2C: speed set to STANDARD              
[00:00:00.005,000] <dbg> i2c_dw: i2c_dw_setup: I2C: lcnt = 500                         
[00:00:00.005,000] <dbg> i2c_dw: i2c_dw_setup: I2C: hcnt = 400                         
[00:00:00.005,000] <dbg> i2c_dw: i2c_dw_isr: I2C: interrupt received                   
```

Here is how lcnt and hcnt are defined : 

```
#define I2C_STD_HCNT            (CONFIG_I2C_DW_CLOCK_SPEED * 4)
#define I2C_STD_LCNT            (CONFIG_I2C_DW_CLOCK_SPEED * 5)

```

Which is by default :
```
config I2C_DW_CLOCK_SPEED
    default 100 
```

The clock-frequency defined in device tree is not used by the driver.

Target: RP2040
 

**To Reproduce**
Can be reproduce with any program that use RP2040 as i2c controller and measure i2c frequency.

**Expected behavior**
clock-frequency property is used to compute SCL frequency.

**Impact**
Wrong scl signal frequency. 

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Ubuntu 22
 - Toolchain Zephyr SDK
 - Commit sha : 07c6af3b8c35c1e49186578ca61a25c76e2fb308

**Additional context**
[Link](https://discord.com/channels/720317445772017664/938474761405726800/1088370337118433333) of the conversation with @yonsch in discord for infos.

yonsch, 2023-03-27:
@andrei-edward-popa this should be fixed in/after #52901

nashif, 2023-05-30:
@yonsch can you please take a look again here. It seems the PR mentioned above is stale? Is there any other way this can be fixed?

yonsch, 2023-06-03:
> @yonsch can you please take a look again here. It seems the PR mentioned above is stale? Is there any other way this can be fixed?

@nashif Fixed here: #58814

PR #58814 (2023-06-05): rp2040: Fix I2C frequency for all boards
The DW driver used by the RP2040 boards uses a Kconfig symbol to determine the clock speed of the board. This value was set incorrectly by the pico board to 100MHz (instead of 125). For the other boards, it wasn't set at all.
This commit fixes the kconfig defaults. In the future, this value should be taken directly from the devicetree. This is not currently possible as there is no clock control driver for the RP2040 (see #52901).

Fixes #56230
closed 2023-06-05 by nashif