zephyrproject-rtos/zephyr #56232: samples: bluetooth: unicast_audio_client: LE Create CIS fails at a second stream

mkasenberg, 2023-03-27:
**Describe the bug**
The unicast_audio_client app fails at "Starting streams" step. There are four streams to start. After looking at the btmoon logs we can see that a failure occurred at the second stream, at LE Create CIS command.

Btsnoop on the client side shows:
```
< HCI Command: LE Create Connected Isochronous Stream (0x08|0x0064) plen 5                                               #251 [hci0] 8.384016
        Number of CIS: 1
        CIS Handle: 7
        ACL Handle: 0
> HCI Event: Command Status (0x0f) plen 4                                                                                #252 [hci0] 8.384316
      LE Create Connected Isochronous Stream (0x08|0x0064) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 29                                                                                #253 [hci0] 8.761216
      LE Connected Isochronous Stream Established (0x19)
        Status: Connection Failed to be Established (0x3e)
        Connection Handle: 7
        CIG Synchronization Delay: 4440 us (0x001158)
        CIS Synchronization Delay: 2220 us (0x0008ac)
        Central to Peripheral Latency: 4440 us (0x001158)
        Peripheral to Central Latency: 4440 us (0x001158)
        Central to Peripheral PHY: LE 2M (0x02)
        Peripheral to Central PHY: LE 2M (0x02)
        Number of Subevents: 3
        Central to Peripheral Burst Number: 1
        Peripheral to Central Burst Number: 1
        Central to Peripheral Flush Timeout: 1
        Peripheral to Central Flush Timeout: 1
        Central to Peripheral MTU: 40
        Peripheral to Central MTU: 40
        ISO Interval: 8
< HCI Command: LE Remove Isochronous Data Path (0x08|0x006f) plen 3                                                      #254 [hci0] 8.761316
        Connection Handle: 7
        Data Path Direction: Output (Controller to Host) (0x01)
> HCI Event: Command Complete (0x0e) plen 6                                                                              #255 [hci0] 8.761516
      LE Remove Isochronous Data Path (0x08|0x006f) ncmd 1
        invalid packet size
        0c 07 00                                         ...             
< HCI Command: LE Remove Isochronous Data Path (0x08|0x006f) plen 3                                                      #256 [hci0] 8.761616
        Connection Handle: 7
        Data Path Direction: Input (Host to Controller) (0x00)
> HCI Event: Command Complete (0x0e) plen 6                                                                              #257 [hci0] 8.761816
      LE Remove Isochronous Data Path (0x08|0x006f) ncmd 1
        invalid packet size
        0c 07 00
```

Btsnoop on the server side shows:
```
> HCI Event: LE Meta Event (0x3e) plen 7                                                                                #301 [hci0] 21.794816
      LE Connected Isochronous Stream Request (0x1a)
        ACL Handle: 0
        CIS Handle: 7
        CIG ID: 0x00
        CIS ID: 0x01
< HCI Command: LE Accept Connected Isochronous Stream Request (0x08|0x0066) plen 2                                      #302 [hci0] 21.794916
        CIS Handle: 7
> HCI Event: Command Status (0x0f) plen 4                                                                               #303 [hci0] 21.795216
      LE Accept Connected Isochronous Stream Request (0x08|0x0066) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 29                                                                               #304 [hci0] 22.117416
      LE Connected Isochronous Stream Established (0x19)
        Status: Connection Failed to be Established (0x3e)
        Connection Handle: 7
        CIG Synchronization Delay: 65536 us (0x010000)
        CIS Synchronization Delay: 196626 us (0x030012)
        Central to Peripheral Latency: 263169 us (0x040401)
        Peripheral to Central Latency: 66051 us (0x010203)
        Central to Peripheral PHY: Reserved (0x00)
        Peripheral to Central PHY: Reserved (0x00)
        Number of Subevents: 2
        Central to Peripheral Burst Number: 40
        Peripheral to Central Burst Number: 0
        Central to Peripheral Flush Timeout: 2
        Peripheral to Central Flush Timeout: 10
        Central to Peripheral MTU: 16384
        Peripheral to Central MTU: 156
        ISO Interval: 1
```

**To Reproduce**
1. Have 2 x nrf5340dk boards.
2. Build controller app in samples/bluetooth/hci_rpmsg with
`west build -b nrf5340dk_nrf5340_cpunet . -- -DOVERLAY_CONFIG=./nrf5340_cpunet_iso-bt_ll_sw_split.conf`
3. Flash the controller app into NETCORE of both nrf5340dk boards with command:
`west flash --skip-rebuild`
4. Have nrf5340_hci_rpmsg.conf with:
```
### Host buffers (APPCORE) have to match controller buffers (NETCORE)
CONFIG_BT_MAX_CONN=2
CONFIG_BT_BUF_EVT_RX_COUNT=16
CONFIG_BT_BUF_EVT_RX_SIZE=255
CONFIG_BT_BUF_CMD_TX_SIZE=255
CONFIG_BT_BUF_ACL_TX_SIZE=251
CONFIG_BT_L2CAP_TX_MTU=247
CONFIG_BT_BUF_ACL_RX_SIZE=255

### Fixes: Discovering sinks Discovery failed: 5
CONFIG_BT_TINYCRYPT_ECC=y

###  HCI dump over RTT
CONFIG_USE_SEGGER_RTT=y
CONFIG_LOG_BACKEND_RTT=n
CONFIG_BT_DEBUG_LOG=n
CONFIG_BT_DEBUG_MONITOR_RTT=y
```
5. Build the server app in samples/bluetooth/unicast_audio_server with;
`west build -b nrf5340dk_nrf5340_cpuapp -- -DOVERLAY_CONFIG=nrf5340_hci_rpmsg.conf`
and flash into APPCORE of the first nrf5340dk board with:
`west flash --skip-rebuild`

6. Build the client app in samples/bluetooth/unicast_audio_client with;
`west build -b nrf5340dk_nrf5340_cpuapp -- -DOVERLAY_CONFIG=nrf5340_hci_rpmsg.conf`
and flash into APPCORE of the second nrf5340dk board with:
`west flash --skip-rebuild`
7. Run btmon for both boards with:
`sudo btmon -J nRF5340_xxAA_APP`
8. Restart both boards to see error logs in btmon

The issue is reproducible with nrf52840 too, just skip NETCORE part and adjust config with:
`CONFIG_BT_CTLR_ADV_DATA_LEN_MAX=191`
to fix "Failed to set advertising data (err -5)".

**Expected behavior**
The app successfully starts audio streaming.

**Environment (please complete the following information):**
 - Toolchain: Zephyr SDK 15.2
 - HEAD Commit SHA: 6898094111816cfaab6d491a526685aba52f5e21

**Additional context**
[client_btsnoop.log](https://github.com/zephyrproject-rtos/zephyr/files/11075906/client_btsnoop.log)
[server_btsnoop.log](https://github.com/zephyrproject-rtos/zephyr/files/11075907/server_btsnoop.log)
[hci_rpmsg.config.txt](https://github.com/zephyrproject-rtos/zephyr/files/11075928/hci_rpmsg.config.txt)
[client.config.txt](https://github.com/zephyrproject-rtos/zephyr/files/11075933/client.config.txt)
[server.config.txt](https://github.com/zephyrproject-rtos/zephyr/files/11075935/server.config.txt)

Thalley, 2023-03-27:
It seems like the sample attempts to establish more than one CIS, which currently is not support by the controller. @cvinayak Do you have any open PRs that adds support for multiple CIS, which can close this issue?

cvinayak, 2023-03-28:
https://github.com/zephyrproject-rtos/zephyr/pull/55509 is the draft PR with support for Multiple Central LE Create CIS.

PR #55509 (2023-05-05): Bluetooth: Controller: Central and Peripheral multiple CIS support
Implementation of preliminary Central with multiple CIS
create support in Lower Link Layer.

Updated nRF5 Series LLL implementation to support multiple
peripheral CIS establishment from peer central devices.

Add BabbleSim test to verify creation of Central multiple
CIS in a CIG, connect ACL to nine peripherals and create
CIS to each peripheral.

Add BabbleSim test to verify creation of peripheral multiple
CIS in a CIG, connect ACL to one peripherals and create nine
CIS to that peripheral device.

Fixes #55264.
Fixes #55265.
Fi [... cut]
closed 2023-05-05 by carlescufi