zephyrproject-rtos/zephyr #56254: rd client stop hangs if rd client is suspended

chmielewskiandreas, 2023-03-27:
I have tried to integrate rd_client_pause/resume into our product again. Unfortunately I see another bug in its implementation. Unfortunately lwm2m_rd_client_stop with deregister-param set to true hangs if the rd_client is in suspended state. I already raised a discussion about this https://github.com/zephyrproject-rtos/zephyr/discussions/54478 some weeks ago.

@SeppoTakalo, can you take a look into that, please?

chmielewskiandreas, 2023-03-28:
To fix this issue quickly I have opened a PR https://github.com/zephyrproject-rtos/zephyr/pull/56269

SeppoTakalo, 2023-04-05:
@chmielewskiandreas please review https://github.com/zephyrproject-rtos/zephyr/pull/56564

commit 567b2510 (2023-04-06): net: lwm2m: Allow stopping while paused

When LwM2M engine is paused, stop functionality just
marks the intent and when we resume, it will send the
de-registration if that was requested.
Otherwise it will stay in SUSPENDED state until resumed
and then goes into ENGINE_DEREGISTERED -> IDLE.

This also removes the blocking loop from lwm2m_rd_client_stop().
It does not need to block because lwm2m_rd_client_start()
is already checking if client is in IDLE.

Fixes #56254
closed 2023-04-06 by carlescufi