zephyrproject-rtos/zephyr #56264: Qemu_kvm_arm64 build failure tests/kernel/common

decsny, 2023-03-27:
**Describe the bug**
tests/kernel/common fails to build on qemu_kvm_arm64 board.

**To Reproduce**
west build -b qemu_kvm_arm64 -p always tests/kernel/common

**Impact**
Blocking PR #55422 due to twister failure

**Logs and console output**
[qemu_build_fail.log](https://github.com/zephyrproject-rtos/zephyr/files/11083304/qemu_build_fail.log)

**Environment (please complete the following information):**
 - WSL
 - Zephyr SDK
 - commit  e0c90b93939c392274efcc52d7c0a65c2d84d2bd

carlocaione, 2023-03-28:
```
fbf851cdc4d8f0afe60131aea6154945f126d778 is the first bad commit                                                                                                                                                     
commit fbf851cdc4d8f0afe60131aea6154945f126d778                                                                                                                                                                      
Author: Yuval Peress <peress@google.com>                                                                                                                                                                             
Date:   Mon Mar 20 22:45:41 2023 -0600                                                                                                                                                                               
                                                                                                                                                                                                                     
    util.h: Add dependency on time_units.h                                                                                                                                                                           
                                                                                                                                                                                                                     
    The WAIT_FOR macro uses the 'k_us_to_cyc_ceil32' function declared                                                                                                                                               
    in zephyr/sys/time_units.h. It should have been included.                                                                                                                                                        
                                                                                                                                                                                                                     
    Fixes #56006                                                                                          
                                                                                                                                                                                                                     
    Signed-off-by: Yuval Peress <peress@google.com>                                                                                                                                                                  
                                                                                                                                                                                                                     
 include/zephyr/sys/util.h | 5 +++++                                                                                                                                                                                 
 1 file changed, 5 insertions(+)  
```

carlocaione, 2023-03-28:
@yperess FYI

PR #56276 (2023-03-28): arm64: Break circular dependency on syscall.h
Commit fbf851cdc4d8f0afe60131aea6154945f126d778 is introducing a dependency between util.h and time_units.h.

Now, the problem is that time_units.h is from one side included by arm64/syscall.h (syscall.h -> lib_helpers.h -> cpu.h -> util.h) but on the other side it is including syscall.h (time_units.h -> syscall.h).

Break the dependency at cpu.h level.

Fixes: #56264