zephyrproject-rtos/zephyr #56280: mcp23xxx driver incorrectly assumes all chips in family have push-pull outputs

arbrauns, 2023-03-28:
**Describe the bug**

The `mcp23xxx` driver [only allows push-pull pins](https://github.com/zephyrproject-rtos/zephyr/blob/9c69f29d0d37aa608b575154696aba088fc2a15e/drivers/gpio/gpio_mcp23xxx.c#L182-L185). This is correct for the MCP23008, MCP23016 and MCP23017, but the MCP23009 and MCP23018 have open-drain outputs instead of push-pull outputs.

**Expected behavior**
The driver and device bindings should be adjusted to match the real hardware capabilities.

**Impact**
Currently, open-drain outputs have to be configured as if they were push-pull outputs, which can make the devicetree slightly confusing to read (e.g. `GPIO_ACTIVE_HIGH | GPIO_PULL_UP`, which should be `GPIO_OPEN_DRAIN | GPIO_PULL_UP`).

henrikbrixandersen, 2023-03-28:
Please use our bug template when reporting bugs. You need to edit this issue to include the needed information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md

arbrauns, 2023-05-30:
Hasn't fixed itself yet.

henrikbrixandersen, 2023-05-30:
> Hasn't fixed itself yet.

If you have the hardware to test it, you are most welcome to submit a PR for fixing it.

arbrauns, 2023-05-30:
Sure, I'll probably do that eventually, but it's not the top priority right now - if this issue wasn't here I'd definitely forget it though :)

arbrauns, 2023-07-31:
I'll get to it eventually.

arbrauns, 2023-11-27:
See #65797.

arbrauns, 2024-01-29:
The fix PR hasn't magically merged itself yet.

arbrauns, 2024-04-02:
PR is still open.

arbrauns, 2024-06-03:
PR is *still* open with no relevant review.

arbrauns, 2024-08-05:
Sometimes I dream of making the stale bot close its own miserable self

PR #65797 (2024-08-09): mcp23xxx: make devicetree compatibles unique per chip
Pin definitions should correctly reflect the actual drive mode of the GPIO controller, either push-pull or open drain.

EDIT: see https://github.com/zephyrproject-rtos/zephyr/pull/65797#issuecomment-1993904939 for current description of the changes

Fixes #56280.
closed 2024-08-09 by carlescufi