zephyrproject-rtos/zephyr #56281: High power consumption in power-on sleep for nRF52840 after attempting to mount SD card

MartinSollie, 2023-03-28:
**Describe the bug**
(I originally posted this on Discord, but after more testing I am convinced this must be a bug)

I am working with an nRF52840 on a custom board with an SD card connected on SPI, running Zephyr 3.2.99/NCS 2.2.0. The board only has the nRF52840 with accompanying components (passives and crystals), and an SD card slot. There is no voltage regulator separate from the nRF's regulator on the board, and it is powered directly by a Otii Arc power supply set to 3.3V output. 

By measuring the current to the setup with the Otii Arc, I find that a minimal application example running only `fs_mount` directly followed by `k_sleep(K_FOREVER)` significantly increases the power consumption of the nRF compared to not mounting the card (skipping the `fs_mount`).  This happens even if the SD card is not even connected to the nRF. Merely trying to mount (even if unsuccessful) result in power consumption increasing a lot. Power-on sleep for the nRF alone increases from ~2.4µA to ~490µA by calling fs_mount. Unmounting the card does not reduce the power consumption at all, so mounting and then unmounting does not end with the same behaviour as never mounting in the first place. If however I sleep before mounting initially, the current draw is initially low, only high after mounting. Mounting the card and going to _power-off_ sleep gives a low power consumption as expected.

This seems strange to me. Is something in the SD subsystem / driver preventing the CPU from going to power-on sleep, or keeping the high frequency clock active?

I have found that the increase in current starts after the call to `sdhc_spi_rx` from `sdhc_spi_init_card` in `drivers/sdhc/sdhc_spi.c` ([here](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sdhc/sdhc_spi.c#L144)). I am not sure if it is relevant for this problem, but it appears for me that the `SPI_CS_ACTIVE_HIGH` flag used for this call is not actually checked by the nRF SPIM driver, which uses the device tree configuration for the CS pin.

**To Reproduce**
This is the small application I am running to demonstrate the problem:
The J-Link programmer with RTT was not connected to the board when performing the current measurements.

SD card configuration:
```
&spi3 {
    cs-gpios = <&gpio0 19 GPIO_ACTIVE_LOW>;

    sdhc0: sdhc@0 {
            compatible = "zephyr,sdhc-spi-slot";
            reg = <0>;
            status = "okay";
            spi-max-frequency = <25000000>;
			mmc {
				compatible = "zephyr,sdmmc-disk";
				status = "okay";
			};
    };
};
```

Prj.conf:
```
CONFIG_MAIN_STACK_SIZE=4096
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048

CONFIG_PRINTK=y
CONFIG_STDOUT_CONSOLE=y

##### SD card configuration #####
CONFIG_DISK_ACCESS=y
CONFIG_FILE_SYSTEM=y
CONFIG_DISK_DRIVER_SDMMC=y
CONFIG_FAT_FILESYSTEM_ELM=y
CONFIG_FS_FATFS_EXFAT=y
CONFIG_FS_FATFS_LFN=y

##### Logger configuration #####
CONFIG_LOG=y
CONFIG_LOG_BACKEND_RTT_MODE_BLOCK=y
CONFIG_SEGGER_RTT_BUFFER_SIZE_UP=4096
```

main.c:
```
#include <zephyr/kernel.h>
#include <zephyr/logging/log.h>

#include <zephyr/storage/disk_access.h>
#include <zephyr/fs/fs.h>
#include <ff.h>

LOG_MODULE_REGISTER(main);

static FATFS fat_fs;
static struct fs_mount_t mp = {
	.type = FS_FATFS,
	.fs_data = &fat_fs,
};
static const char *disk_mount_pt = "/SD:";
    
void main(void)
{
	// Delay the mounting to see initial power-consumption
	k_sleep(K_SECONDS(5));

	// Mount
	mp.mnt_point = disk_mount_pt;
	int ret = fs_mount(&mp);
	if (ret) {
		LOG_ERR("Error mounting SD card.");
	} else {
		LOG_INF("SD card mounted.");
	}

	// Optional unmount
	k_sleep(K_SECONDS(5));
	(void)fs_unmount(&mp);

	// Power-on sleep forever
	k_sleep(K_FOREVER);
}
```

**Expected behavior**
I expect that:
1. Successfully mounting the card, and then directly calling `k_sleep(K_FOREVER);` should not significantly increase power consumption compared to not mounting the card.
2. An unsuccessful attempt at mounting, e.g. by physically removing the SD card before booting, should not increase power consumption compared to not attempting to mount at all.
3. Unmounting the SD card after successfully mounting should result in the CPU having the same steady-state power consumption as never mounting at all.

**Impact**
Significantly increased power consumption

**Logs and console output**
The first screenshot shows current for the code above, both with and without the SD card inserted in its slot. The current measurements in the top right corner is for the selected (grey) time segment.
[image]

The second screenshot shows the current for 
```
void main(void)
{
	// Power-on sleep forever
	k_sleep(K_FOREVER);
}
```
with the SD card inserted and not inserted. 
[image]

**Environment:**
nRF52840
Zephyr 3.2.99/NCS 2.2.0
The SD card used above is a Transcend 2GB

MartinSollie, 2023-03-30:
The problem is caused by a product anomaly for the nRF52840, see [[195] SPIM: SPIM3 continues to draw current after disable](https://infocenter.nordicsemi.com/topic/errata_nRF52840_Rev2/ERR/nRF52840/Rev2/latest/anomaly_840_195.html).

Therefore it does not happen if a different SPIM instance is used, although this limits the maximum clock frequency to 8 MHz. I tested this with SPIM1. It is also not specific to SD card, but any device using SPIM3.

A configurable fix for this could be added to the SPIM driver, like it is already done for anomaly 58, with the workaround described on Nordic Infocenter.

jfischer-no, 2023-04-12:
> Hopefully, a configurable fix for this could be added to the SPIM driver, like it is already done for anomaly 58, with the workaround described on Nordic Infocenter.

Maybe @anangl can comment on this.

closed 2024-04-13 by github-actions[bot]