zephyrproject-rtos/zephyr #56303: wifi sample crash on m5stickc_plus

jvermillard, 2023-03-28:
**Describe the bug**
The wifi sample compiles fine with for m5stickc_plus but it crash as soon as I use a simple wifi shell command, like `wifi status` or `wifi scan`

**To Reproduce**
build: `west build -b m5stickc_plus samples/net/wifi --pristine`

flash: `west flash --esp-baud-rate 1500000`

connect using: `west espressif monitor`

type "wifi status" => crash

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
```
❯ west build -b m5stickc_plus samples/net/wifi --pristine                                                                                                                                     zephyr/git/main 
-- west build: making build dir /home/jvermillard/sandbox/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/jvermillard/sandbox/zephyrproject/zephyr/samples/net/wifi
-- CMake version: 3.25.1
-- Found Python3: /usr/bin/python3 (found suitable exact version "3.9.2") found components: Interpreter 
-- Cache files will be written to: /home/jvermillard/.cache/zephyr
-- Zephyr version: 3.3.99 (/home/jvermillard/sandbox/zephyrproject/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: m5stickc_plus
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.2 (/home/jvermillard/sandbox/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (/home/jvermillard/sandbox/zephyr-sdk-0.15.2)
-- Found Dtc: /home/jvermillard/sandbox/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/jvermillard/sandbox/zephyrproject/zephyr/boards/xtensa/m5stickc_plus/m5stickc_plus.dts
-- Generated zephyr.dts: /home/jvermillard/sandbox/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/jvermillard/sandbox/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/jvermillard/sandbox/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/jvermillard/sandbox/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/jvermillard/sandbox/zephyrproject/zephyr/boards/xtensa/m5stickc_plus/m5stickc_plus_defconfig'
Merged configuration '/home/jvermillard/sandbox/zephyrproject/zephyr/samples/net/wifi/prj.conf'
Configuration saved to '/home/jvermillard/sandbox/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/jvermillard/sandbox/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/jvermillard/sandbox/zephyr-sdk-0.15.2/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc
CMake Warning at /home/jvermillard/sandbox/zephyrproject/zephyr/CMakeLists.txt:828 (message):
  No SOURCES given to Zephyr library: lib__libc__common

  Excluding target from build.

CMake Warning at /home/jvermillard/sandbox/zephyrproject/zephyr/CMakeLists.txt:828 (message):
  No SOURCES given to Zephyr library: drivers__wifi

  Excluding target from build.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/jvermillard/sandbox/zephyrproject/zephyr/build
-- west build: building application
[2/342] Preparing syscall dependency handling

[5/342] Generating include/generated/version.h
-- Zephyr version: 3.3.99 (/home/jvermillard/sandbox/zephyrproject/zephyr), build: zephyr-v3.3.0-1776-g6b12ad91d2c1
[147/342] Performing configure step for 'EspIdfBootloader'
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
[... 279 lines cut ...]

0x400d3abf: exec_cmd at /home/jvermillard/sandbox/zephyrproject/zephyr/subsys/shell/shell.c:558
 (inlined by) execute at /home/jvermillard/sandbox/zephyrproject/zephyr/subsys/shell/shell.c:800

0x400d3c12: state_collect at /home/jvermillard/sandbox/zephyrproject/zephyr/subsys/shell/shell.c:1002
 (inlined by) shell_process at /home/jvermillard/sandbox/zephyrproject/zephyr/subsys/shell/shell.c:1472

0x400d3592: shell_signal_handle at /home/jvermillard/sandbox/zephyrproject/zephyr/subsys/shell/shell.c:1289

0x400d42f9: shell_thread at /home/jvermillard/sandbox/zephyrproject/zephyr/subsys/shell/shell.c:1348
 (inlined by) shell_thread at /home/jvermillard/sandbox/zephyrproject/zephyr/subsys/shell/shell.c:1305

0x400d1b8f: z_thread_entry at /home/jvermillard/sandbox/zephyrproject/zephyr/lib/os/thread_entry.c:36

[00:00:13.107,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:13.107,000] <err> os: Current thread: 0x3ffc9428 (shell_uart)
[00:00:13.242,000] <err> os: Halting system
```
**Environment (please complete the following information):**
 - OS: Linux Debian 11
 - Toolchain zephyr-sdk-0.15.2
 - Commit SHA 6b12ad91d2c1977f89aacee02097093dabdf270a

**Additional context**
N/A

closed 2023-05-11 by carlescufi

jvermillard, 2024-02-01:
I tried again with lastest master today and same result with the wifi sample on this board, same crash

jvermillard, 2024-02-01:
it's probably because the overlay for this board is missing, if I add:
m5stickc_plus.conf with:

```
CONFIG_WIFI=y

CONFIG_NETWORKING=y
CONFIG_NET_L2_ETHERNET=y

CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=y
CONFIG_ESP32_WIFI_STA_AUTO_DHCPV4=y

CONFIG_NET_LOG=y
```

then it works

sylvioalves, 2024-02-01:
@jvermillard I guess you need to add a proper overlay enabling esp32 wifi driver: `samples/net/wifi/boards/m5stick_plus.overlay`:
```c
&wifi {
	status = "okay";
};
```

jvermillard, 2024-02-02:
I did it yes, let's me try to send a PR for this

jvermillard, 2024-02-02:
https://github.com/zephyrproject-rtos/zephyr/pull/68451

PR #57627 (2023-05-11): driver: wifi: esp32: add wifi statistics and status
Add Wi-Fi statistics counter and connection status interface.

Fixes #56303