zephyrproject-rtos/zephyr #56317: net: ipv4: fragments: Out of order packet reassembly does not work

nordicjm, 2023-03-29:
**Describe the bug**
Description from https://github.com/zephyrproject-rtos/zephyr/issues/54577

Had a quick look and I think there is an error in both IPv4 and IPv6 defragmentation (common code) here in fragments_are_ready:

```
	/* Fragments can arrive in any order, for example in reverse order:
	 *   1 -> Fragment3(M=0, offset=x2)
	 *   2 -> Fragment2(M=1, offset=x1)
	 *   3 -> Fragment1(M=1, offset=0)
	 * We have to test several requirements before proceeding with the reassembly:
	 * - We received the first fragment (Fragment Offset is 0)
	 * - All intermediate fragments are contiguous
	 * - The More bit of the last fragment is 0
	 */
	for (i = 0; i < CONFIG_NET_IPV4_FRAGMENT_MAX_PKT; i++) {
...
		more = net_pkt_ipv4_fragment_more(pkt);
	}

	if (more) {
		return 0;
	}
```

So the more value is set to the value of the last received packet, but the last received packet doesn't have to be the final packet in the chain, in fact if the packets are received as the comment suggests, more will wrongly be set to 1 when it is processed.

I'm not a great fan of how that function runs, it makes a lot of assumptions and only checks if packets received after the first have a higher offset, so if you receive the first packet, the final (3rd) packet then the second packet, that function is going to discard the one is receives last because the offset of that packet is lower than the new expected offset.

Essentially, out of order packets will not work.

**Expected behavior**
Out of order packets to work

**Impact**
None to showstopper depending upon network that device is operating in

**Environment (please complete the following information):**
 - Commit SHA or Version used: 9011a067f21d2cd7dc01a179503c981d34834c59

nordicjm, 2023-03-29:
*To be confirmed, missed the `shift_packets` function call, issue might be invalid

closed 2023-03-29 by nordicjm

ssharks, 2023-03-29:
If you want a test to reproduce the isssue, you might want to take a look at the test code I made for the IPv4 issue: https://github.com/zephyrproject-rtos/zephyr/compare/main...ssharks:zephyr:net/ip4v_frag_test