zephyrproject-rtos/zephyr #56326: Some peripheral tests are failing if CONFIG_BT_EXT_ADV=n

sjanc, 2023-03-29:
**Describe the bug**
Some BLE qualification tests are failing if tester application is build without CONFIG_BT_EXT_ADV=y.
This seems to affect peripheral role privacy tests.

Affected tests:
GAP/BROB/BCST/BV-03-C
GAP/CONN/UCON/BV-06-C
GAP/PRIV/CONN/BV-10-C

There seems to be some discrepancy in host on how privacy and peripheral role works with and without ExtAdv enabled.
tester application doesn't have any CONFIG_BT_EXT_ADV dependent code and simply calls bt_le_adv_start().
Note that tester is always advertising with legacy PDUs 

**To Reproduce**

Build tester application with CONFIG_BT_EXT_ADV=n and execute any of mentioned qualification tests.

**Expected behavior**
Tests should pass regardless of CONFIG_BT_EXT_ADV being enabled or not.

jori-nordic, 2023-04-11:
Hi @sjanc, PTS is not fast to install and setup, would you have any logs that you could attach to this issue?

sjanc, 2023-04-20:
Hi,

Attaching logs from pass and failed runs for GAP/BROB/BCST/BV-03-C.
Only difference between those two is tester configuration prj.conf:
CONFIG_BT_EXT_ADV=y for pass
CONFIG_BT_EXT_ADV=n for fail

[GAP_BROB_BCST_BV_03_C_FAIL.zip](https://github.com/zephyrproject-rtos/zephyr/files/11286742/GAP_BROB_BCST_BV_03_C_FAIL.zip)
[GAP_BROB_BCST_BV_03_C_PASS.zip](https://github.com/zephyrproject-rtos/zephyr/files/11286743/GAP_BROB_BCST_BV_03_C_PASS.zip)

PR #59199 (2023-06-17): Bluetooth: host: invalidate the RPA when starting legacy adv
This fixes the failure to use a resolvable private address in this scenario.

1. call `bt_le_oob_get_local`, will generate and mark RPA as valid
2. start connectable adv w/ IDENTITY bit
3. start connectable adv w/o IDENTITY
4. RPA is not set (in `bt_id_set_private_addr`) because RPA is still marked as valid

When EXT_ADV is enabled and the controller supports it, a different code path is taken that doesn't have this issue.

Unconditionally invalidating the RPA when starting advertising works around this issue.

Fixes #56326
commit 931a089e (2023-06-17): Bluetooth: host: invalidate the RPA when starting legacy adv

This fixes the failure to use a resolvable private address in this
scenario.

1. call `bt_le_oob_get_local`, will generate and mark RPA as valid
2. start connectable adv w/ IDENTITY bit
3. start connectable adv w/o IDENTITY
4. RPA is not set (in `bt_id_set_private_addr`) because RPA
   is still marked as valid

When EXT_ADV is enabled and the controller supports it, a different code
path is taken that doesn't have this issue.

Unconditionally invalidating the RPA when starting advertising works around
this issue.

Fixes #56326
closed 2023-06-17 by nashif