zephyrproject-rtos/zephyr #56332: mec172xmodular_assy6930 standalone fails to boot hello_world on zephy3.3 or mainline

galak, 2023-03-29:
Zephyr 3.3 or mainline don't boot hello_world on mec172xmodular_assy6930 when used in standalone.

Changing some of the boot params in spi_cfg seem to make it work:

```
diff --git a/boards/arm/mec172xmodular_assy6930/support/spi_cfg_4MBit.txt b/boards/arm/mec172xmodular_assy6930/support/spi_cfg_4MBit.txt
index 809e67b356..43debfe40a 100644
--- a/boards/arm/mec172xmodular_assy6930/support/spi_cfg_4MBit.txt
+++ b/boards/arm/mec172xmodular_assy6930/support/spi_cfg_4MBit.txt
@@ -16,8 +16,8 @@ BoardID = 0
 [IMAGE "0"]
 ImageLocation = 0x1000
 SpiFreqMHz = 24
-SpiReadCommand = Dual
-SpiDriveStrength = 4
+SpiReadCommand = Quad
+SpiDriveStrength = 8
```

galak, 2023-03-29:
@scottwcpg, @albertofloyd fyi.

scottwcpg, 2023-03-29:
@galak  If your 6930 board has the 1.8V Gigadevice GD25LR256 that had an issue with dual.  I will try to find the details

galak, 2023-03-29:
@scottwcpg looks like MEC1723, REV B board, with Gigadevices GD25LR256.

galak, 2023-03-29:
I have JP26 set to 1-2 for 1.8V since its a gigadevices chip.

galak, 2023-04-04:
@scottwcpg any updates on this?

scottwcpg, 2023-04-05:
Hi @galak 
The GigaDevice SPI flash on the MEC172x MECC card does not support dual mode. This issue was also seen by @albertofloyd at Intel.  I was informed a request was made of GigaDevice to release a public application note.  The work-around is to use Quad mode.  Application level code will need to implement the application note to switch the part to another mode.

albertofloyd, 2023-04-11:
@galak only Dual to Quad change should be  required, the spi strength is only needed when using with RVP not in standalone mode. Do confirm

closed 2023-06-07 by nashif