zephyrproject-rtos/zephyr #56336: k_forever is causing a problem with fifo

ChrisKurz, 2023-03-29:
There are already some discussions about K_FOREVER usage and possibility that this fails. It looks like using k_fifo_get(k_forever) can also fail. First, I saw it with high data throughput during UART communication.  In this case the k_fifo_put() was called in an ISR. It looks like that multiple interrupts cause at least to execute the ISR three times, before the k_fifo_get(k_forever) instruction that was placed in another thread was executed. When the k_fifo_put() was done 3-times, the k_fifo_get(k_forever) does not stop and wait any longer. 

An easy way to reproduce the issue is to use the zephyr/samples/basic/button example. Do a k_fifo_put() in the button event handler (e.g. putting a button press counter in the fifo, and do a printk() in main and ouptut the button press counter there). I main add k_fifo_get(k_forever). If you press the button fast enough, the main will not stop any longer at k_fifo_get() and wait for another k_fifo_put(). 
A easier way to reproduce the issue is to call k_fifo_put() 3 times in the button event handler... 

**Expected behavior**
k_fifo_get(k_forever) should wait in case there is no further data in the fifo... 

**Impact**
showstopper...  In case the k_fifo is used in ISRs an asynchronous interrupt signal may cause the k_fifo_get(forever) function will not stop any longer (and wait for new data)

**Environment (please complete the following information):**
 - OS: Windows
 - Zephyr v3.2.0

DariusBabrauskas, 2023-03-30:
Possible problem is that, if is thread  which waiting data in k_fifo_get(), then  k_fifo_put  funtion not place data into list, but direct into this thread variable swap_data.
thread->base.swap_data= data;
So only 1 data from 3, will get k_fifo_get.

jgl-meta, 2023-04-04:
@peter-mitsis

peter-mitsis, 2023-04-19:
@ChrisKurz - Can you please provide a code sample? I see that you provided a fairly detailed description, but a code sample would also be helpful to ...
1. eliminate potential ambiguities
2. ensure that we are all looking at the same thing
3.  and to be able to verify the correctness of the test scenario itself

ChrisKurz, 2023-04-20:
Hi, yes, no problem... here is the zipped project:
[project.zip](https://github.com/zephyrproject-rtos/zephyr/files/11280168/project.zip)

I have tested this with nRF52840DK and nRF Connect SDK v2.3.0. 

The attached project is done in a way that you immediately see the problem. Connect the DK to a PC and start a terminal program. (115200bd, 8 data bits, 1 stop bit, no parity, no flow control)
When "Press the button" is shown in the terminal, press button 1 on the DK once. Then you will see something like this:
[image]
Note that the output "Counter: 3" is continuously received... 
And if you press button 1 again, then the counter is increased by 3 and the new value is shown in the terminal. but still the line 
            my_getData = k_fifo_get(&my_fifo, K_FOREVER);
is not waiting for new data and immediately jumps to next instruction (so in terminal constantly the counter value is received... 

You can also check an remove one k_fifo_put() function call in the button_pressed function. Like this:
[image]
If you start code execution, then it looks like the program execution is correctly working. Terminal output looks then like this:
[image]

Best regards

nylnx, 2023-04-20:
@ChrisKurz You are using `k_fifo` wrong. You need distinct elements and cannot use the same `my_data` multiple times. This is because `k_fifo` is a linked list in the background and the reserved pointer you have to have in your `my_data` struct is actually the pointer to the next element. In your example the same physical memory gets overwritten multiple times and it will never work correctly. Currently you did only allocate memory for exactly _one_ FIFO element. Everything above that will just behave unpredictably.

Solutions are to use `k_fifo_alloc_put` or manage the memory yourself with memory slabs or something like that. 

It is described in the explanation of the FIFO documentation, but it really takes a moment to correctly understand. Also the [Writing to a FIFO](https://docs.zephyrproject.org/latest/kernel/services/data_passing/fifos.html#writing-to-a-fifo) is wrong in the sense that they used `k_fifo_put` in the code example and only mentioned `k_fifo_alloc_put` below. The zephyr documentation has quite a few such gotchas, where the crucial points of a particular structure/api are not adequately described.

ChrisKurz, 2023-04-20:
@nylnx Thanks for your feedback... Please note that the previously attached project is just a way to show you in a simple example code what may happen if the k_fifo_put/k_fifo_get API is used in an asynchronous interrupt service routine. 
The real problem was seen in a project that uses the fifo to pass data from an ISR to a thread. This project used the k_fifo_put once in its ISR. Unfortunately there were too many interrupt triggers so that the the k_fifo_put was executed several times before the k_fifo_get could be executed. 
Even if the single-linked list pointer is overwritten, I would not expect that the k_fifo_get(..., k_forever) function in the main would not stop any longer and wait for new data... 

You can also remove the marked code section as stated in the pressed_button function. Then k_fifo_put is executed only once. But if you connect a pulse generated at the button input and if you slowly increase the frequency, you will find out that at a certain frequency the code execution will fail.....

Since the FIFO documentation only talks about that it supports to be called in ISRs, the assumption was that using k_fifo_put would also work. In this case the expectation is that the point in the single-linked list is only updated once.... 
So in my previous post I stated that when doing k_fifo_put() twice in the ISR the output seems to be correct. Well, seeing the counter value twice is not correct. If a variable is already on the fifo, there should be only one read via k_fifo_get()...

ChrisKurz, 2023-04-20:
@peter-mitsis Understood... but the webpage you mention also states:
"A data item may be added to a FIFO by a thread or an ISR."
And there is no restriction here regarding certain fifo functions. 

As @DariusBabrauskas stated above: "It seems to me, that k_fifo_put is not isr_ok."

DariusBabrauskas, 2023-04-21:
@ChrisKurz 
I make test k_fifo_put test:
1. k_fifo_put  not locked 
2. k_fifo_put locked 

$ test k_fifo
Test kfifo
m: do not lock
0x2000beb8 0x2000beb8
0x2000bec0 0x2000bec0
0x2000bec8 0x2000bec8
0x2000bed0 0x2000bed0
m: end test
$ test k_fifo 1
Test kfifo
m: dolock
0x2000beb8 0x2000beb8
0x2000bec0 0x2000bec0
0x2000bec8 0x2000bec8
0x2000bed0 0x2000bed0
m: end test

Result all ok. 
At first put  - data is put on swap data and waiting   thread  became unwaiting.
Next  k_fifo_put  then placed into fifo list.

Maybe possible problem is uart OVR  error, then you lost bytes.
BR

```
/* Modified test code:*/
#define STACK_SIZE	(1024 + CONFIG_TEST_EXTRA_STACKSIZE)
#define LIST_LEN	4
struct fdata_t {
	sys_snode_t snode;
	uint32_t data;
};

static K_FIFO_DEFINE(fifo1);
/* Data to put into FIFO */
static struct fdata_t data1[LIST_LEN];
static struct fdata_t data2[LIST_LEN];

static K_THREAD_STACK_DEFINE(tstack, STACK_SIZE);

static struct k_thread tdata;
static struct k_sem end_sema;
`
static void thread_entry_fn_single(void *p1, void *p2, void *p3)
{
	void *rx_data;
	uint32_t i;

	/* Give control back to Test thread */
	k_sem_give(&end_sema);

	/* Get items from fifo */
	for (i = 0U; i < LIST_LEN; i++) {
		rx_data = k_fifo_get((struct k_fifo *)p1, K_MSEC(1000));//K_NO_WAIT
		//zassert_equal(rx_data, (void *)&data1[i], NULL);
		printk("%p %p\r\n",rx_data, (void *)&data1[i]);
	}

	while(1)
		k_sleep(K_MSEC(10));

}

static void test_single_fifo_play(u32 dolock)
{
	void *rx_data;
	uint32_t i;

	/* Init kernel objects */
	k_sem_init(&end_sema, 0, 1);

	k_tid_t tid = k_thread_create(&tdata, tstack, STACK_SIZE,
				thread_entry_fn_single, &fifo1, NULL, NULL,
				K_PRIO_PREEMPT(0), K_INHERIT_PERMS, K_NO_WAIT);

	/* Let the child thread run */
	k_sem_take(&end_sema, K_FOREVER);

	k_sleep(K_MSEC(15));//wait while k_fifo_get enter, to get waiting thread
	/* Put items into fifo */

	if(dolock){
		printk("m: dolock\r\n");
		unsigned int gIrqKey = irq_lock()  ;
		for (i = 0U; i < LIST_LEN; i++) {
			k_fifo_put(&fifo1, (void *)&data1[i]);
		}
		irq_unlock(gIrqKey) ;
	}
	else{
		printk("m: do not lock\r\n");
		for (i = 0U; i < LIST_LEN; i++) {
			k_fifo_put(&fifo1, (void *)&data1[i]);
		}
	}

	k_sleep(K_SECONDS(6));
	printk("m: end test\r\n");

	/* Clear the spawn thread to avoid side effect */
	k_thread_abort(tid);
}

static int cmd_kfifo(const struct shell* shell, size_t argc, char** argv)
{
	shell_print(shell, "Test kfifo");

	u32 dolock=0;
	if( argc >= 2 )
	{
		dolock = 1;
	}

	 test_single_fifo_play(dolock);

	return 0;
}

	//SHELL_CMD(k_fifo,	NULL, "Test k_fifo",cmd_kfifo),

peter-mitsis, 2023-04-21:
I am going to try writing a summary of the situation thus far.

1. It has been reported (which includes a project to demo it) has been made that the FIFOs do not work as expected from an ISR.
2. Responders have noted that the supplied project is using FIFOs incorrectly.
3. Both reporters and responders have pointed to documentation to back up their claims.

It seems to me that the crux of the matter is that the behaviour is not what is expected, and that this is based on a different interpretations of the same documentation. My suggestion is to update the documentation to make it absolutely clear that items must not be re-added to a FIFO if they have not yet been removed.

ChrisKurz, 2023-04-23:
@peter-mitsis Updating the docu is a solution - adding something like: 
The user code must ensure that data is removed before new data is added to the FIFO. The user code must also take this into account when handling in interrupt service routines.

closed 2023-05-02 by nashif

GeorgeCGV, 2023-10-11:
Just got into the problem today with slip. The implementation makes use of `net_recv_data` (_that makes use of `submit_to_queue`_) in an ISR. That results in a fault.

peter-mitsis, 2023-10-11:
@GeorgeCGV - I may be missing something, but if the FIFO entry is being re-used, would that not suggest that the network code is attempting to re-use a packet that has not yet been consumed, or returned to the pool of packets?

GeorgeCGV, 2023-10-11:
@peter-mitsis what is observed is not yet fully known.

Slip adds pkt into the rx queue from ISR, while the net rx thread gets them from the queue.

For whatever reason during the first `submit_to_queue` call the queue is no longer empty and contains some rubbish entries (_was empty upon init_). However, I don't see any other calls that put anything into the queue. That leads to an invalid pkt being read and eventually leads to a fault within net rx processing. It happens only when optimizations are enabled which complicates the debugging.

Didn't perform further investigation.

GeorgeCGV, 2023-10-11:
@peter-mitsis what is observed is not yet fully known.

Slip adds pkt into the rx queue from ISR, while the net rx thread gets them from the queue.

For unknown reasons, the queue is not empty upon the first `submit_to_queue` call and contains some rubbish entries (_was empty upon init_). However, I don't see any other calls that put anything into the queue. That results in an invalid pkt and a fault during network rx pkt processing. It happens only when optimizations are enabled.

I didn't perform further investigation and would be glad to know if someone has an idea what it might be.

If it relates to

> It has been reported (which includes a project to demo it) has been made that the FIFOs do not work as expected from an ISR.

then the SLIP re-implementation to offload pkt handling would resolve the issue (_that is the "hope" at the moment_). The offloading helps overcome serial overruns as well.

GeorgeCGV, 2023-10-12:
@peter-mitsis in the slip the usart pipe and iface (_queue `put` parts_) are initialized before the networking rx queue processing (_queue `get` parts_).

commit 2adfa3e1 (2023-05-02): doc: Update FIFO and LIFO documentation

Updates the FIFO and LIFO documentation to clarify behavior
surrounding re-adding data items to queues.

Fixes #56336