zephyrproject-rtos/zephyr #56352: ESP32 Bootloader issue

hf81-ble, 2023-03-30:
Hi,

# Toolchain:
i installed the sdk-16 as full package.  Also initialized my fresh workspace with west init and update to MR 3.3.0

# Build NRF OK
Also a demo with nrf works.

# Build ESP32
Using the target plattform ESP32 fails. May hints are from the time the espressif stuff was not complete integrated in the zephyr toolchain.

for ESP32 i fetched the binary libs with

`west blobs fetch hal_espressif`

i try to build the ESP32 hello world.

But the configuration with the bootloader fails. Any hints?

See Log:

```$ west build -p always -b esp32 samples/hello_world/
-- west build: making build dir G:\Projekte\nrf\zephyrproject\zephyr\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: G:/Projekte/nrf/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.23.1
-- Found Python3: C:/Program Files/Python310/python.exe (found suitable exact version "3.10.4") found components: Interpreter
-- Cache files will be written to: G:/Projekte/nrf/zephyrproject/zephyr/.cache
-- Zephyr version: 3.3.0 (G:/Projekte/nrf/zephyrproject/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.7.1")
-- Board: esp32
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.0 (C:/tools/zephyr-sdk-0.16.0)
-- Found toolchain: zephyr 0.16.0 (C:/tools/zephyr-sdk-0.16.0)
-- Could NOT find Dtc (missing: DTC) (Required is at least version "1.4.6")
-- Found BOARD.dts: G:/Projekte/nrf/zephyrproject/zephyr/boards/xtensa/esp32/esp32.dts
-- Generated zephyr.dts: G:/Projekte/nrf/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: G:/Projekte/nrf/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: G:/Projekte/nrf/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing G:/Projekte/nrf/zephyrproject/zephyr/Kconfig
Loaded configuration 'G:/Projekte/nrf/zephyrproject/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration 'G:/Projekte/nrf/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to 'G:/Projekte/nrf/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to 'G:/Projekte/nrf/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/tools/zephyr-sdk-0.16.0/xtensa-espressif_esp32_zephyr-elf/bin/xtensa-espressif_esp32_zephyr-elf-gcc.exe
-- Configuring done
-- Generating done
-- Build files have been written to: G:/Projekte/nrf/zephyrproject/zephyr/build
-- west build: building application
[1/183] Generating ../../../../include/generated/core-isa-dM.h
[2/183] Creating directories for 'EspIdfBootloader'
[3/183] Creating directories for 'EspPartitionTable'
[4/183] No download step for 'EspIdfBootloader'
[5/183] No download step for 'EspPartitionTable'
[6/183] Generating ../../../../include/generated/zsr.h
[7/183] Generating include/generated/version.h
-- Zephyr version: 3.3.0 (G:/Projekte/nrf/zephyrproject/zephyr), build: zephyr-v3.3.0
[8/183] No update step for 'EspPartitionTable'
[9/183] No update step for 'EspIdfBootloader'
[10/183] No patch step for 'EspIdfBootloader'
[11/183] No patch step for 'EspPartitionTable'
[12/183] No configure step for 'EspPartitionTable'
[13/183] Performing build step for 'EspPartitionTable'
[14/183] No install step for 'EspPartitionTable'
[15/183] Completed 'EspPartitionTable'
[16/183] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[... 57 lines cut ...]
[61/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/log/log.c.obj
[62/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/esp_hw_support/port/esp32/rtc_init.c.obj
[63/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/esp_timer/src/esp_timer_impl_lac.c.obj
[64/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/esp_hw_support/port/esp32/rtc_clk_init.c.obj
[65/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/esp_hw_support/mac_addr.c.obj
[66/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/hal/esp32/interrupt_descriptor_table.c.obj
[67/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/esp_system/port/soc/esp32/clk.c.obj
[68/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/esp_timer/src/ets_timer_legacy.c.obj
[69/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/hal/interrupt_controller_hal.c.obj
[70/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/efuse/esp32/esp_efuse_table.c.obj
[71/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/efuse/esp32/esp_efuse_fields.c.obj
[72/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/esp_timer/src/esp_timer.c.obj
[73/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/efuse/esp32/esp_efuse_utility.c.obj
[74/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/esp_hw_support/regi2c_ctrl.c.obj
[75/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/esp_hw_support/esp_clk.c.obj
[76/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/driver/periph_ctrl.c.obj
[77/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/zephyr/esp32/src/stubs.c.obj
[78/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/efuse/src/esp_efuse_api_key_esp32.c.obj
[79/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/efuse/src/esp_efuse_utility.c.obj
[80/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/zephyr/esp32/src/common/dport_access.c.obj
[81/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/zephyr/esp_shared/src/common/esp_system_api.c.obj
[82/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/components/efuse/src/esp_efuse_api.c.obj
[83/183] Building C object zephyr/CMakeFiles/zephyr.dir/G_/Projekte/nrf/zephyrproject/modules/hal/espressif/zephyr/esp32/src/heap_caps.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'G:\Projekte\nrf\zephyrproject\zephyr\build'```

henrikbrixandersen, 2023-03-30:
Please use our bug template when reporting bugs. You need to edit this issue to include the needed information from https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/bug_report.md

sylvioalves, 2023-03-30:
@hf81-ble, I've spotted an environment issue in there:
```c
CMake Error at G:/Projekte/nrf/zephyrproject/modules/hal/espressif/tools/cmake/kconfig.cmake:30 (message):
  Windows requires an MSYS2 version of gcc on the PATH to build mconf-idf.
  Consult the setup docs for ESP-IDF on Windows.
  ```
  This is related to espressif dependencies on Windows, which we haven't yet handled issues like this. Would you take a look at the link below to manage this?  

https://gitdemo.readthedocs.io/en/latest/windows-setup.html

hf81-ble, 2023-04-26:
i fixed it, by setting the IDF_PATH .... but this should normally happen if i use the export environment script or setup

closed 2023-07-11 by github-actions[bot]