zephyrproject-rtos/zephyr #56411: Add new Cryptography module, trezor crypto library, alternative to tinycrypt

JiapengLi, 2023-03-31:
## Origin

trezor crypto

https://github.com/trezor/trezor-firmware/tree/master/crypto

## Purpose

## Mode of integration

integrate this external component as a module

repository name: trezor-crypto

## Maintainership

@carlescufi  (maintainer of tinycrypto)

## Pull Request

N/A

## Description

Heavily optimized cryptography algorithms for embedded devices.

These include:

- AES/Rijndael encryption/decryption
- Big Number (256 bit) Arithmetics
- BIP32 Hierarchical Deterministic Wallets
- BIP39 Mnemonic code
- ECDSA signing/verifying (supports secp256k1 and nist256p1 curves, uses RFC6979 for deterministic signatures)
- ECDSA public key derivation
- BIP340 Schnorr signature signing/verifying
- Base32 (RFC4648 and custom alphabets)
- Base58 address representation
- Ed25519 signing/verifying (also SHA3 and Keccak variants)
- ECDH using secp256k1, nist256p1 and Curve25519
- HMAC-SHA256 and HMAC-SHA512
- PBKDF2
- RIPEMD-160
- SHA1
- SHA2-256/SHA2-512
- SHA3/Keccak
- BLAKE2s/BLAKE2b
- Chacha20-Poly1305
- unit tests (using Check - check.sf.net; in test_check.c)
- tests against OpenSSL (in test_openssl.c)
- integrated Wycheproof tests

What is its primary functionality (e.g., SQLLite is a lightweight
database)?

**do crypto**

What problem are you trying to solve? (e.g., a state store is
required to maintain ...)

**tinycrypto (or uECC) is not blockchain or web3 friendly, lacks of main feature (ECDSA recovery id)**

## Dependencies

What other components does this package depend on?

**libc / random**

Will the Zephyr project have a direct dependency on the component, or
will it be included via an abstraction layer with this component as a
replaceable implementation?

**it is a tinycrypt alternative**

## Revision

4d0381055c5ff5ef750599fe177b17395dc9b8e8

## License

**Distibuted under MIT License.**

ceolin, 2023-04-05:
I was looking about releasing cadence and couldn't find anything and how vulnerabilities are disclosed. Though there is information about how to reporting them.

dleach02, 2023-04-10:
I think the project should first align and implement the PSA crypto support before we start pulling in additional cryptographic libraries.

ceolin, 2023-04-21:
> I think the project should first align and implement the PSA crypto support before we start pulling in additional cryptographic libraries.

Completely agree, specially due concerns raised here. Closing this issue now, feel free to re-open it if something new happens.

closed 2023-04-21 by ceolin