zephyrproject-rtos/zephyr #56414: tests: kernel: device: test failing in CI for qemu_x86_tiny

cfriedt, 2023-03-31:
**Describe the bug**

The following test is currently failing in `main`:
```
west build -p auto -b qemu_x86_tiny@768 -t run -T tests/kernel/device/kernel.device.pm
```

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? `qemu_x86_tiny`
 - What have you tried to diagnose or workaround this issue? just a git bisect
 - Is this a regression? Yes. The bad commit is 06c9bf47b37c0722559f3333c2e85240d4ecbddb

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -p auto -b qemu_x86_tiny@768 -t run -T tests/kernel/device/kernel.device.pm`
2. See error

**Expected behavior**
Tests should pass

**Impact**
Breaking CI in `main`

**Logs and console output**
```
SeaBIOS (version rel-1.16.1-0-g3208b098f51a-prebuilt.qemu.org)
Booting from ROM..ASSERTION FAIL [page_frames_initialized] @ WEST_TOPDIR/zephyr/kernel/mmu.c:1325
        page fault at 0x40196266 happened too early
E: EAX: 0x00000001, EBX: 0x00009500, ECX: 0x00000000, EDX: 0x00000000
E: ESI: 0x00000000, EDI: 0x00001000, EBP: 0x401905ec, ESP: 0x401905e8
E: EFLAGS: 0x00000046 CS: 0x0008 CR3: 0x0016f000
E: call trace:
E: EIP: 0x4013d1db
E:      corrupted? (bp=0x401905ec)
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x40190728 (unknown)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux (CI)
 - Toolchain: SDK v0.16.0
 - Commit SHA or Version used: bf46f7351066c78cf5eb2d056fdfdfb182dab985

**Additional context**
git bisect steps
```
git bisect start HEAD v3.3.0 --
git bisect run twister -p qemu_x86_tiny@768 -s tests/kernel/device/kernel.device.pm
git bisect reset
```

cfriedt, 2023-03-31:
This test isn't failing across the board, just on a single platform, so I've set it to medium, but if anyone thinks it should by high priority, please escalate.

nordicjm, 2023-03-31:
Pinging @dcpleung as this issue seems to vanish when demand paging support is disabled.

PR #56436 (2023-03-31): pm: mark pm_device_runtime_auto_enable as boot function
When demand paging is enabled, only a minimal set of functions are available at boot. Anything not marked as boot functions would not be loading in memory at boot which would result in page faults when jumping to those functions. However, at early boot, demand paging has not been enabled yet. So we need to mark necessary functions as boot functions so they are placed in the correct linker section where they are loaded at boot.

Fixes #56414
commit 8dda1fe1 (2023-03-31): pm: mark pm_device_runtime_auto_enable as boot function

When demand paging is enabled, only a minimal set of functions
are available at boot. Anything not marked as boot functions
would not be loading in memory at boot which would result in
page faults when jumping to those functions. However, at early
boot, demand paging has not been enabled yet. So we need to
mark necessary functions as boot functions so they are placed
in the correct linker section where they are loaded at boot.

Fixes #56414
closed 2023-03-31 by nashif