zephyrproject-rtos/zephyr #56415: Issues with power management introduction into Intel CAVS25

nashif, 2023-03-31:
**Describe the bug**

Few Kernels tests started failing after 95168e6776d6c8803a5e1f3ed10e1a8b6bc98311 was merged:

```
INFO     - 1) tests/boards/intel_adsp/ssp/boards.intel_adsp.ssp on intel_adsp_cavs25 failed (Timeout)
INFO    - 2) tests/kernel/tickless/tickless_concept/kernel.tickless.concept on intel_adsp_cavs25 failed (Timeout)
INFO    - 3) tests/kernel/timer/timer_api/kernel.timer.tickless on intel_adsp_cavs25 failed (Timeout)
```

**To Reproduce**
Steps to reproduce the behavior:
1. west twister -p intel_adsp_cavs25 -s tests/boards/intel_adsp/ssp/boards.intel_adsp.ssp

**Expected behavior**
Tests shall pass

**Impact**

Multiple kernel tests failing.

**Logs and console output**

In the case of SSP, no output.  With tickless tests, the test just times out.

nashif, 2023-03-31:
@kv2019i  FYI

ceolin, 2023-04-05:
There are a bunch of hw specific flow in this commit, looking Zephyr's pm, the way core 0 is handled in `pm_state_set` raises some suspicious.

```
+       if (state == PM_STATE_SOFT_OFF) {
+               core_desc[cpu].intenable = XTENSA_RSR("INTENABLE");
+               z_xt_ints_off(0xffffffff);
+               soc_cpus_active[cpu] = false;
+               z_xtensa_cache_flush_inv_all();
+               if (cpu == 0) {
+                       uint32_t ebb = EBB_BANKS_IN_SEGMENT;
+                       /* turn off all HPSRAM banks - get a full bitmap */
+                       uint32_t hpsram_mask = (1 << ebb) - 1;
+                       /* do power down - this function won't return */
+                       power_down_cavs(true, uncache_to_cache(&hpsram_mask));
+               } else {
+                       z_xt_ints_on(core_desc[cpu].intenable);
+                       k_cpu_idle();
+               }
```

how core 0 is supposed to wake up if all interruptions are masked ?

One more thing, if is not necessary to do anything in cores !=0 other than `k_cpu_idle` you should just remove `cpu-power-states = <&d3>;` in the device tree for these cores, then when these cores go idle (idle thread runs) the function `pm_system_suspend` will return `false` and the idle thread will invoke `k_cpu_idle`. This way you simplify a lot this code path.

juimonen, 2023-04-13:
@ceolin you know is the test framework setting the PM_STATE_SOFT_OFF? In sof we are forcing it in driver power off message... I thought it was not set automatically.

juimonen, 2023-04-13:
@ceolin hmm I guess the "new" power-off handling should be surrounded with #ifdef CONFIG_PM_POLICY_CUSTOM in power.c then the tests seem to work. it is kind of assumed here that this should be only called by custom handler in the app... this is my blind copy from ace side which I think we are not testing with zephyr tests...

So cavstools can't handle currently the ipc response sent from assembly, also for example the ssp test would need next_state handler for custom policy.

juimonen, 2023-04-14:
fix here: https://github.com/zephyrproject-rtos/zephyr/pull/56843

nashif, 2023-05-18:
this is fixed now