zephyrproject-rtos/zephyr #56416: AT45 flash memory driver doesn't handle sector erase of 0a/0b correctly

rossj2, 2023-03-31:
**Describe the bug**

Many (most/all?) Atmel/Renesas AT45 DataFlash chips such as the AT45DB641E (64Mbit) split sector 0 into 0a and 0b. Although the standard sector size of this chip is 256KB, sector 0a is only 2KB, with 0b making up the balance (254KB). This has not be catered for by the Zephyr driver's (spi_flash_at45.c) erase operation. Instead of erasing the full 256KB sector, only the first 2KB is erased.

If it turns out that all AT45 DataFlash parts have a 2KB 0a sector, then this could be handled transparently by the driver. Otherwise, an additional parameter in the DT entry may be required to qualify the sector-size.

**To Reproduce**

Devicetree entry:

```
	spi_flash: at45db641e@0 {
		compatible = "atmel,at45";
		reg = <0>;
		spi-max-frequency = <DT_FREQ_M(15)>;
		jedec-id = [1f 28 00];
		size = <DT_SIZE_M(64)>;
		sector-size = <DT_SIZE_K(256)>;
		block-size = <DT_SIZE_K(2)>;
		page-size = <256>;
		enter-dpd-delay = <2000>;
		exit-dpd-delay = <35000>;
		wp-gpios = <&gpio0 10 GPIO_ACTIVE_LOW>;
	};

```

Then:

```
static const struct device *spi_flash = DEVICE_DT_GET(DT_NODELABEL(spi_flash));
flash_erase(spi_flash , 0, 0x40000);

```

**Expected behavior**

Erase the flash range requested (i.e. the whole 256KB in this case).

**Impact**

Workarounds are either...

1. Don't use sector 0.
2. Make two separate flash_erase() calls to force the erase into two parts, so as to prevent a sector erase (resulting in 128 x 2KB block erases). This is obviously slower.

I'll use option 1 as I don't require all the chip.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain Nordic NCS 2.3.0

Custom board with a AT45DB641E on board.

de-nordic, 2023-04-21:
I think Atmel/Microchip devices didn't divide sector. This seems later addition by next owners of DataFlash series.

rossj2, 2023-04-21:
I just checked an Atmel datasheet I have for the AT45DB081D (8Mbit) dated 2010 (3596M–DFLASH–5/10). That had the split 0a/0b sectors. So this has been around for quite a while.

de-nordic, 2023-04-21:
> I just checked an Atmel datasheet I have for the AT45DB081D (8Mbit) dated 2010 (3596M–DFLASH–5/10). That had the split 0a/0b sectors. So this has been around for quite a while.

Seems that the last letter matters. D, E and F seem to have it. B does not https://ww1.microchip.com/downloads/en/DeviceDoc/doc2224.pdf, but I do not think it is manufactured.

PR #57292 (2023-05-09): drivers: spi_flash_at45: Fix erasing of first two sectors
Most available AT45 flash chips have their first two sectors shorter than the consecutive ones. Usually, the first sector is marked as 0a and has its size equal to eight pages (one block) and the second one (usually 0b) is the complement to the size of a regular sector. This commits modifies the driver so that erasing of these first two sectors is performed correctly. This modified behavior is configurable with a new DT property so that it is still possible to also use legacy AT45 chips that do not feature such sector split. Such legacy [... cut]
closed 2023-05-09 by carlescufi