zephyrproject-rtos/zephyr #56442: flash_stm32_ospi chip erase returns immediately, but chip remains busy

bdunlay, 2023-03-31:
**Describe the bug**

Flash memory chip erase (which occurs when erase size >= flash size) returns immediately. Immediately subsequent calls to operate on flash fail because flash is busy.

Should the chip erase call return immediately, or block until the operation is complete? If it should return immediately despite the ongoing operation, what is the recommended way to check the status of the device for when the operation has completed?

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? B-L4S5I-IOT01A
 - What have you tried to diagnose or workaround this issue? Waiting a significant amount of time between erase / format. Reducing partition size to trigger sector erase instead of chip erase.
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit? No

**To Reproduce**
Steps to reproduce the behavior: modified littlefs sample

1. Modify littlefs sample to reference the [ospi partition](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/b_l4s5i_iot01a/b_l4s5i_iot01a.dts#L267-L270)) 'store/store_partition' 
2. Modify littlefs sample to enable [storage wipe](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/fs/littlefs/prj.conf#L8) `CONFIG_APP_WIPE_STORAGE=y`
3. `west build samples/subsys/fs/littlefs --board b_l4s5i_iot01a

**Expected behavior**
Erase followed by format successful

**Impact**
Unable to reliably use full flash memory without resorting to some kind of tricks to sleep or test write in a loop.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.3.0 ***
Sample program to r/w files on littlefs
Area 5 at 0x0 on ospi-nor-flash@0 for 8388608 bytes
E: Erasing flash area ... 0
I: LittleFS version 2.5, disk version 2.0
I: FS at ospi-nor-flash@0:0x0 is 2048 0x1000-byte blocks with 512 cycle
I: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
E: WEST_TOPDIR/modules/fs/littlefs/lfs.c:1234: Corrupted dir pair at {0x0, 0x1}
W: can't mount (LFS -84); formatting
E: OSPI AutoPoll failed
E: Erase failed : flash busy
E: format failed (LFS -5)
E: fs mount error (-5)
FAIL: mount id 5 at /lfs: -5
```

**Environment (please complete the following information):**
 - OS: OS X
 - Toolchain (e.g Zephyr SDK, ...): SDK 0.15.2 
 - Commit SHA or Version used: Zephyr 3.3.0

FRASTM, 2023-04-03:
Correct.
Looking at the stm32 ospi flash driver, the full chip erase (BULK) command could include a simple control on the Write Enable Latch (WEL) : wait until the WEL bit is cleared to return from the flash_stm32_ospi_erase function. 
Note that, as you mentionned, the operation is quite long and might block for several seconds (  (45s).

FRASTM, 2023-04-04:
@bdunlay can you please have a look at the PR https://github.com/zephyrproject-rtos/zephyr/pull/56472
if its helps

closed 2023-05-10 by fabiobaltieri