zephyrproject-rtos/zephyr #56448: USB CDC ACM sample not working with STM32F302RC

deyro, 2023-04-02:
**Describe the bug**
I am trying to run the `cdc_acm` sample on a stm32f302rct6 custom board. The problem I am facing is, when I attached the USB with my computer the USB does not get detected. I am sure this is not a hardware issues as the built-in DFU bootloader works fine, also I checked the board with a STM32Cube generated code and the USB gets detected as a CDC device.

Below is the device tree of my custom board:
```
/dts-v1/;
#include <st/f3/stm32f302Xc.dtsi>
#include <st/f3/stm32f302r(b-c)tx-pinctrl.dtsi>

/ {
	model = "Custom STM32F302RC board";
	compatible = "rdt,tscb-f302rc";
	chosen {
		zephyr,console = &usart1;
		zephyr,shell-uart = &usart1;
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
	};
};

&clk_lsi {
	status = "okay";
};

&clk_hse {
	clock-frequency = <DT_FREQ_M(12)>;
	status = "okay";
};
&pll {
	prediv = <1>;
	mul = <4>;
	clocks = <&clk_hse>;
	status = "okay";
};
&rcc {
	clocks = <&pll>;
	clock-frequency = <DT_FREQ_M(48)>;
	ahb-prescaler = <1>;
	apb1-prescaler = <2>;
	apb2-prescaler = <1>;
};

&i2c1 {
	pinctrl-0 = <&i2c1_scl_pb6 &i2c1_sda_pb7>;
	pinctrl-names = "default";
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;
};

&usart1 {
	pinctrl-0 = <&usart1_tx_pc4 &usart1_rx_pc5>;
	pinctrl-names = "default";
	current-speed = <115200>;
	status = "okay";
};

&rtc {
	clocks = <&rcc STM32_CLOCK_BUS_APB1 0x10000000>,
		 <&rcc STM32_SRC_LSI RTC_SEL(2)>;
	status = "okay";
};

zephyr_udc0: &usb {
	pinctrl-0 = <&usb_dm_pa11 &usb_dp_pa12>;
	pinctrl-names = "default";
	status = "okay";
};
```

**Logs and console output**
Logs after booting, without USB attached
```
*** Booting Zephyr OS build v3.3.0-rc2-44-g461a13c085de ***
[00:00:00.000,000] <inf> cdc_acm_echo: Wait for DTR
[00:00:00.004,000] <inf> usb_cdc_acm: Device suspended
```
the below logs are printed when I attached the USB
```
[00:00:53.397,000] <inf> usb_cdc_acm: Device resumed
[00:00:53.397,000] <inf> usb_cdc_acm: from suspend
[00:00:54.312,000] <inf> usb_cdc_acm: Device suspended
[00:00:54.633,000] <inf> usb_cdc_acm: Device resumed
[00:00:54.633,000] <inf> usb_cdc_acm: from suspend
[00:00:55.840,000] <inf> usb_cdc_acm: Device suspended
```

**Expected behavior**
The USB should get detected as a CDC device and data can be sent or received.

**Impact**
Showstopper. My actual application will need USB with a shell interface.

**Environment**
 - OS: ubuntu 18.04
 - Toolchain: zephyr-sdk-0.15.2
 - Zephyr Version used: v3.3.0-rc2-44-g461a13c085

Can someone tell me if I am doing something incorrectly or this is a bug.

Thanks

deyro, 2023-04-09:
I tried updated zephyr version to `zephyr-v3.3.0-2372-gda633c6148` but it still doesn't work. I get the same result.
Can someone please help me out.

Thanks

deyro, 2023-04-09:
I enabled debug logs for the USB driver and stack
```
CONFIG_USB_DRIVER_LOG_LEVEL_DBG=y
CONFIG_USB_DEVICE_LOG_LEVEL_DBG=y
```

Below are the logs which are printed
Logs after booting, without USB attached
```
[00:00:00.000,000] <dbg> usb_descriptor: usb_get_device_descriptor: __usb_descriptor_start 0x200001dc
[00:00:00.000,000] <dbg> usb_descriptor: usb_get_device_descriptor: __usb_descriptor_end 0x2000029b
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Configuration descriptor 0x200001ee
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Association descriptor 0x200001f7
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Interface descriptor 0x200001ff
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Endpoint descriptor 0x2000021b
[00:00:00.000,000] <dbg> usb_dc_stm32: usb_dc_ep_check_cap: ep 81, mps 16, type 3
[00:00:00.000,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: Fixing EP address 80 -> 81
[00:00:00.000,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: endpoint 0x81
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Interface descriptor 0x20000222
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Endpoint descriptor 0x2000022b
[00:00:00.000,000] <dbg> usb_dc_stm32: usb_dc_ep_check_cap: ep 82, mps 64, type 2
[00:00:00.000,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: Fixing EP address 80 -> 82
[00:00:00.000,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: endpoint 0x82
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Endpoint descriptor 0x20000232
[00:00:00.000,000] <dbg> usb_dc_stm32: usb_dc_ep_check_cap: ep 1, mps 64, type 2
[00:00:00.000,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: Fixing EP address 0 -> 1
[00:00:00.000,000] <dbg> usb_descriptor: usb_validate_ep_cfg_data: endpoint 0x1
[00:00:00.000,000] <dbg> usb_descriptor: usb_fix_descriptor: Now the wTotalLength is 75
[00:00:00.000,000] <wrn> cbprintf_package: (unsigned) char * used for %p argument. It's recommended to cast it to void * because it may cause misbehavior in certain configurations. String:"%s: idx_max %d, ascii_idx_max %d, buf %p" argument:3
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: idx_max 11, ascii_idx_max 5, buf 0x2000023f
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char R : 52, idx 5 -> 11
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char Y : 59, idx 4 -> 9
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char H : 48, idx 3 -> 7
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char P : 50, idx 2 -> 5
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char E : 45, idx 1 -> 3
[00:00:00.000,000] <dbg> usb_descriptor: ascii7_to_utf16le: char Z : 5a, idx 0 -> 1
[00:00:00.001,000] <wrn> cbprintf_package: (unsigned) char * used for %p argument. It's recommended to cast it to void * because it may cause misbehavior in certain configurations. String:"%s: idx_max %d, ascii_idx_max %d, buf %p" argument:3
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: idx_max 41, ascii_idx_max 20, buf 0x2000024d
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char e : 65, idx 20 -> 41
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char l : 6c, idx 19 -> 39
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char p : 70, idx 18 -> 37
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char m : 6d, idx 17 -> 35
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char a : 61, idx 16 -> 33
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char s : 73, idx 15 -> 31
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char   : 20, idx 14 -> 29
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char M : 4d, idx 13 -> 27
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char C : 43, idx 12 -> 25
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char A : 41, idx 11 -> 23
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char   : 20, idx 10 -> 21
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char C : 43, idx 9 -> 19
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char D : 44, idx 8 -> 17
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char C : 43, idx 7 -> 15
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char   : 20, idx 6 -> 13
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char r : 72, idx 5 -> 11
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char y : 79, idx 4 -> 9
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char h : 68, idx 3 -> 7
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char p : 70, idx 2 -> 5
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char e : 65, idx 1 -> 3
[00:00:00.001,000] <dbg> usb_descriptor: ascii7_to_utf16le: char Z : 5a, idx 0 -> 1
[00:00:00.001,000] <dbg> usb_descriptor: usb_update_sn_string_descriptor: Serial Number
                                         53 46 57 17 00 21 00 37                          |SFW..!.7
[00:00:00.001,000] <wrn> cbprintf_package: (unsigned) char * used for %p argument. It's recommended to cast it to void * because it may cause misbehavior in certain configurations. String:"%s: idx_max %d, ascii_idx_max %d, buf %p" argument:3
[00:00:00.002,000] <dbg> usb_descriptor: ascii7_to_utf16le: idx_max 31, ascii_idx_max 15, buf 0x20000279
[... cut ...]
[00:02:52.986,000] <dbg> usb_dc_stm32: HAL_PCD_ResumeCallback:
[00:02:52.986,000] <inf> usb_cdc_acm: Device resumed
[00:02:52.986,000] <inf> usb_cdc_acm: from suspend
[00:02:53.106,000] <dbg> usb_dc_stm32: HAL_PCD_ResetCallback:
[00:02:53.656,000] <dbg> usb_dc_stm32: HAL_PCD_ResetCallback:
[00:02:54.275,000] <dbg> usb_dc_stm32: HAL_PCD_ResetCallback:
[00:02:54.892,000] <dbg> usb_dc_stm32: HAL_PCD_ResetCallback:
[00:02:55.001,000] <dbg> usb_dc_stm32: HAL_PCD_SuspendCallback:
[00:02:55.001,000] <inf> usb_cdc_acm: Device suspended
```

jfischer-no, 2023-04-12:
> **Describe the bug** I am trying to run the `cdc_acm` sample on a stm32f302rct6 custom board. The problem I am facing is, when I attached the USB with my computer the USB does not get detected. I am [...]

It can be anything and no one can reproduce it because it is your custom board. Maybe you can add logs from the host side or reproduce it with a board supported in the tree. IMO it should be discussion not a bug report.

deyro, 2023-04-13:
@jfischer-no Thanks for your message. I am pretty sure that my custom hardware is not the problem because, as I mentioned before, it works with the CubeMX code. I am happy to try and reproduce the issue with a supported board. Which board would you recommend?

erwango, 2023-05-04:
> One from https://docs.zephyrproject.org/latest/boards/arm/index.html that is most similar to yours.

@deyro Have you been able to perform the test ?

carlescufi, 2023-06-07:
Closing due to inactivity, please reopen if required

deyro, 2023-06-25:
Hi @erwango and @jfischer-no,

I tested the `zephyr/samples/subsys/usb/cdc_acm` sample with 2 boards supported by Zephyr
1. [stm32f3_disco](https://docs.zephyrproject.org/latest/boards/arm/stm32f3_disco/doc/index.html)
2. [b_g474e_dpow1](https://docs.zephyrproject.org/latest/boards/arm/b_g474e_dpow1/doc/index.html)

The `b_g474e_dpow1` works fine. The USB gets detected and I am getting echo from the minicom terminal.

For `stm32f3_disco` I am facing a different issue than what is mentioned in the earlier messages. Now the USB gets detected as a `ttyACMx port` on Ubuntu but I am **not getting any serial messages echoed on the minicom terminal**. Below are the logs printed on the terminal of the STLink. By default the `stm32f3_disco` does not come with an LSE installed so I soldered an 8MHz LSE and capacitors separately.

```
*** Booting Zephyr OS build v3.3.0-rc2-44-g461a13c085de ***
[00:00:00.000,000] <inf> cdc_acm_echo: Wait for DTR
[00:00:00.003,000] <inf> usb_cdc_acm: Device suspended
[00:00:07.551,000] <inf> usb_cdc_acm: Device resumed
[00:00:07.551,000] <inf> usb_cdc_acm: from suspend
[00:00:07.567,000] <inf> usb_cdc_acm: Device suspended
[00:00:07.569,000] <inf> usb_cdc_acm: Device resumed
[00:00:07.569,000] <inf> usb_cdc_acm: from suspend
[00:00:07.734,000] <inf> usb_cdc_acm: Device suspended
[00:00:07.804,000] <inf> usb_cdc_acm: Device resumed
[00:00:07.804,000] <inf> usb_cdc_acm: from suspend
[00:00:08.082,000] <inf> usb_cdc_acm: Device configured
[00:00:15.415,000] <inf> cdc_acm_echo: DTR set
[00:00:15.524,000] <inf> cdc_acm_echo: Baudrate detected: 115200
```

Please let me know what might be the issue.
Please let me know if more info/logs are required.

My Zephyr version is: 3.3.0-rc2

Thanks

deyro, 2023-07-02:
Hi @carlescufi, can this issue be reopened?

jfischer-no, 2023-07-03:
@deyro Seems stm32f3_disco hardware specific to me, I cannot help you with that, sorry.

FRASTM, 2023-07-20:
testing on a stm32f3_disco, characters on the user USB are echoed ( like on another stm32 board)
[image]

deyro, 2023-08-27:
thanks @FRASTM for confirming and sorry for the late reply. Now I think there must be some problem with my hardware.

Can you tell me the following:
- to the default stm32f3_disco what hardware changes you made (if any) to make the USB work?
- which version of Zephyr you have tested this with?

closed 2023-10-04 by deyro