zephyrproject-rtos/zephyr #56449: LSE enable problem on STM32F302RC

deyro, 2023-04-02:
**Describe the bug**
I am looking to use the internal RTC of STM32F302RC and run the `alarm` sample inside `drivers/counter`. I have a custom board with STM32F302RC with an external Low Speed lock (LSE) of 32.768 KHz. When I run the sample the main function does not get called. I put the program into debug mode and found that it hangs in the following line of `set_up_fixed_clock_sources` function in file `clock_stm32_ll_common.c`
```
		while (!LL_PWR_IsEnabledBkUpAccess()) {
			/* Wait for Backup domain access */
		}
```

Below is the device tree of my custom board with LSE and RTC enabled:
```
/dts-v1/;
#include <st/f3/stm32f302Xc.dtsi>
#include <st/f3/stm32f302r(b-c)tx-pinctrl.dtsi>

/ {
	model = "TSCB STM32F302RC board";
	compatible = "rdt,tscb-f302rc";
	chosen {
		zephyr,console = &usart1;
		zephyr,shell-uart = &usart1;
		zephyr,sram = &sram0;
		zephyr,flash = &flash0;
	};
};

&clk_lse {
	status = "okay";
};

&clk_hse {
	clock-frequency = <DT_FREQ_M(12)>;
	status = "okay";
};
&pll {
	prediv = <1>;
	mul = <4>;
	clocks = <&clk_hse>;
	status = "okay";
};
&rcc {
	clocks = <&pll>;
	clock-frequency = <DT_FREQ_M(48)>;
	ahb-prescaler = <1>;
	apb1-prescaler = <2>;
	apb2-prescaler = <1>;
};

&i2c1 {
	pinctrl-0 = <&i2c1_scl_pb6 &i2c1_sda_pb7>;
	pinctrl-names = "default";
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;
};

&usart1 {
	pinctrl-0 = <&usart1_tx_pc4 &usart1_rx_pc5>;
	pinctrl-names = "default";
	current-speed = <115200>;
	status = "okay";
};

&rtc {
	clocks = <&rcc STM32_CLOCK_BUS_APB1 0x10000000>,
		 <&rcc STM32_SRC_LSE RTC_SEL(1)>;
	status = "okay";
};

zephyr_udc0: &usb {
	pinctrl-0 = <&usb_dm_pa11 &usb_dp_pa12>;
	pinctrl-names = "default";
	status = "okay";
};
```

I have tested the board by running a STM32CubeMX generated code with the same clock settings. The code runs properly with RTC set and get date/time functions. So it is not an hardware issue.

I have also tested the `alarm` sample with LSI clock. It runs properly.

**Expected behavior**
The LSE should get initialized and the alarm sample should run properly.

**Impact**
Showstopper. My end application requires an RTC with backup.

**Logs and console output**
None

**Environment (please complete the following information):**
 - OS: ubuntu 18.04
 - ToolchainL zephyr-sdk-0.15.2
 - Zephyr Version used: v3.3.0-rc2-44-g461a13c085

What might be the problem here?

Thanks

erwango, 2023-04-03:
We're likely missing PWR clk activation.
Would you have a try to call the following function before the while loop ?
```
      __HAL_RCC_PWR_CLK_ENABLE();
```

deyro, 2023-04-04:
Thank you. I added the line as shown below and it worked. Would you add it in the main tree?

```
		__HAL_RCC_PWR_CLK_ENABLE();
		/* Set the DBP bit in the Power control register 1 (PWR_CR1) */
		LL_PWR_EnableBkUpAccess();
		while (!LL_PWR_IsEnabledBkUpAccess()) {
			/* Wait for Backup domain access */
		}
```

erwango, 2023-04-04:
Thanks for the test. I'll push it upstream, no problem.

closed 2023-04-11 by carlescufi

katyo, 2023-05-12:
@erwango Seems stm32f0 needs same fix.

bdesterBE, 2023-06-11:
Hi,
F1 series (specifically, using an F105RCT6) appears to be giving me this same problem. The fix was not ultimately to add the `__HAL_RCC_PWR_CLK_ENABLE();` call in `clock_stm32_ll_common.c`?

erwango, 2023-06-12:
> Hi, F1 series (specifically, using an F105RCT6) appears to be giving me this same problem. 

Thanks for raising this. Could you open a dedicated issue? Otherwise it is likely that this point will likely be forgotten.

> The fix was not ultimately to add the `__HAL_RCC_PWR_CLK_ENABLE();` call in `clock_stm32_ll_common.c`?

`clock_stm32_ll_common` is used by a wide range of STM32 series, so I prefer to fix in `drivers/clock_control/clock_stm32x.c` files, unless a complete analysis shows that it is useful on all impacted series.

PR #56505 (2023-04-11): drivers: clock_control:  stm32f3: Enable PWR clock to access BDCR
BDCR could be required for LSE or RTC for instance. Enable it here as for now, no sophisticated PM handling is available on F3 series.

Fixes #56449
PR #57855 (2023-05-15): drivers: clock_control: stm32f3: Enable PWR clock to access BDCR
BDCR and PWR_CR could be required for LSE or RTC for instance. Enable it here as for now, no sophisticated PM handling is available on F0 and F3 series.

Fixes #56449
Fixup for #56505
commit c48422f5 (2023-05-15): drivers: clock_control: stm32f3: Enable PWR clock to access BDCR and PWR_CR

BDCR and PWR_CR could be required for LSE or RTC for instance.
Enable it here as for now, no sophisticated PM handling is available
on F0 and F3 series.

Fixes #56449
Fixup for #56505