zephyrproject-rtos/zephyr #56461: Zephyr timers are not accurate

fabio2602, 2023-04-03:
Hi all,
we are here to reporting a strange (for us) behaviour of zephyr timers:

We are working with decawave 1001 (Ultrawide band module from Qorvo) on Nordic micro, using:

Zephyr 3.1.0.
NFR connect SDK 2.1.1
Toolchain 2.1.1
We would like to manage the timers with microseconds precisions:

k_timer_start(&test_timer, K_USEC(451), K_USEC(2543));

what we are getting from serial log (this could be the problem?) is that timer triggers every 19.9 Milliseconds:

[00:00:15.264,160] <inf> main: timer 4657
[00:00:15.284,088] <inf> main: timer 4657
[00:00:15.303,955] <inf> main: timer 4657
[00:00:15.323,883] <inf> main: timer 4657
[00:00:15.343,811] <inf> main: timer 4657
[00:00:15.363,677] <inf> main: timer 4657
[00:00:15.383,605] <inf> main: timer 4657

What is wrong?

Kind Regards

nordicjm, 2023-04-03:
nRF connect bugs should be reported downstream, pinging @anangl as this may be related to recent fixed bugs in zephyr in the nordic timer driver

fabio2602, 2023-04-04:
Hi all,
so what I have to forward the same to @anangl ?

Kind Regards

nordicjm, 2023-04-04:
> so what I have to forward the same to @anangl ?

Awaiting for their reply/input, if it is a zephyr bug/issue then that's OK, if it is an ncs bug and not a zephyr but then this will need to be moved to the dev zone.

anangl, 2023-04-04:
@fabio2602 How can this behavior be reproduced? Could you provide a piece of code and `prj.conf` with which it is possible to see this?

fabio2602, 2023-04-05:
Hi Andrzej,
here attached the very simple sample project used to test.

Thanks in Advance

Kind
[timer_test.zip](https://github.com/zephyrproject-rtos/zephyr/files/11158085/timer_test.zip)
 Regards

fabio2602, 2023-04-12:
Hi all,
any news about?

Kind Regards

anangl, 2023-04-12:
This is not a problem in Zephyr timers but in the used application. Since logging is done there in the timer handler and it is done in the immediate mode, the CPU has no chance to execute any other application code than the handler (that's why the counter that is incremented in the main loop does not change at all). The handler is supposed to be called every ~2.5 ms but its execution takes much longer - just the message transmission over UART takes more than 4 ms.
Instead of using a log message in the handler, try to toggle a pin state there and check with e.g. a logic analyzer that the pin toggles with the required period, with accuracy of the system timer tick (which is 1/32768 s for nRF platforms, so ~30.5 us). That's what you can expect from the kernel timers. If you need better accuracy, use the TIMER peripheral instead (through the counter API or by using the nrfx_timer driver directly).

closed 2023-04-12 by anangl

fabio2602, 2023-04-13:
Hi Andrzej,
your comment is the confirmation of some tests that I performed, waiting for your  response.

Thank you for your time.
Kind Regards