zephyrproject-rtos/zephyr #56470: CONFIG_ADC_ASYNC=y,ADC_STM32_Die_Temp_Read, reading Raw data after calculation getting -ve temparature value on nucleo_f429zi

Niranjan-eaton, 2023-04-03:
**Describe the bug**
After migration from 3.1.0 to 3.2.0, when CONFIG_ADC_ASYNC=y build and flashed on nucleo_f429zi.
Path : \examples\drivers\ADC\ADC_STM32_Die_Temp_Read
Command : 
west build -b nucleo_f429zi ./ --pristine
west flash
[image]

**To Reproduce**
Steps to reproduce the behavior:
Path : \examples\drivers\ADC\ADC_STM32_Die_Temp_Read
Command : 
west build -b nucleo_f429zi ./ --pristine
west flash

**Expected behavior**
Able to read Internal ADC Die temperature +ve values

**Impact**
Major reading tmparatues as -ve values

**Logs and console output**

**Environment (please complete the following information):**
 - OS: Zephyr 3.2.0 Windows
 - Toolchain zephyr-sdk-0.15.2 zephyr-v3.2.0
 - Commit SHA or Version used

heinwessels, 2023-04-03:
Was this issue maybe fixed with https://github.com/zephyrproject-rtos/zephyr/pull/52474?

erwango, 2023-04-03:
@Niranjan-eaton Note that `examples\drivers\ADC\ADC_STM32_Die_Temp_Read` is not a zephyr upstream sample. So it's not possible for us to reproduce the issue or understand what this sample is supposed to do.

erwango, 2023-04-05:
@Niranjan-eaton Any update ?
Please note that the information you provided is not sufficient to allow analysis as the sample you mentioned is not known to publicly available.
Also please provide your feedback on  #52474

Niranjan-eaton, 2023-04-05:
zephyr-v3.2.0\zephyr\drivers\adc\adc_stm32.c

static int start_read(const struct device *dev const struct adc_sequence *)
{
commented below line of code
//adc_stm32_teardown_channels(dev,index)
}
Why are we using adc_stm32_teardown_channels(dev,index)

Niranjan-eaton, 2023-04-05:
> @Niranjan-eaton Any update ? Please note that the information you provided is not sufficient to allow analysis as the sample you mentioned is not known to publicly available. Also please provide your feedback on #52474

verified, but not working 52474

Niranjan-eaton, 2023-04-05:
> zephyr-v3.2.0\zephyr\drivers\adc\adc_stm32.c
> 
> static int start_read(const struct device *dev const struct adc_sequence *) { commented below line of code, working fine. //adc_stm32_teardown_channels(dev,index) } Want to know why we are using adc_stm32_teardown_channels(dev,index)

Want to understand more on adc_stm32_teardown_channels(dev,index)

erwango, 2023-04-05:
Converting to discussion, as there is no actual issue described so far.

Niranjan-eaton, 2023-04-06:
> @Niranjan-eaton Any update ? Please note that the information you provided is not sufficient to allow analysis as the sample you mentioned is not known to publicly available. Also please provide your feedback on #52474

Taken #52474 and tried, getting -ve temperature values instead of +ve
Enabled CON_ADC_ASYN = y
board:Nucleo_F429zi

    overlay file
 /{
    zephyr,user{
    io-channels = <&adc1 18>
       };
};

In debug, we found that adc_read_async(), reading as -ve values for Internal Die temperature.
after commented below line of code in start_read, working fine, reading Internal Die temparature as +ve values //adc_stm32_teardown_channels(dev,index)

static int start_read(const struct device *dev const struct adc_sequence *) { commented below line of code, working fine. //adc_stm32_teardown_channels(dev,index) }

 Want to know why we are using adc_stm32_teardown_channels(dev,index)

Niranjan-eaton, 2023-04-10:
Fixed.

closed 2023-04-10 by Niranjan-eaton