zephyrproject-rtos/zephyr #56482: Canot flash on WeAct Studio Black Pill V2.0

wangyiwu, 2023-04-03:
**Describe the bug**

I installed zephyr to windows 10 following the instructions of getting_started, without any errors.

then I use the command

`west build -p always -b blackpill_f411ce samples\basic\blinky`

to build the project.

and use the command to flash:

`west flash`

the problem you're facing:

```
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\vuong\zephyrproject\.venv\Scripts\west.exe\__main__.py", line 7, in <module>
  File "C:\Users\vuong\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 866, in main
    app.run(argv or sys.argv[1:])
  File "C:\Users\vuong\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 111, in run
    self.run_command(argv)
  File "C:\Users\vuong\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 385, in run_command
    self.run_extension(args.command, argv)
  File "C:\Users\vuong\zephyrproject\.venv\Lib\site-packages\west\app\main.py", line 467, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "C:\Users\vuong\zephyrproject\.venv\Lib\site-packages\west\commands.py", line 194, in run
    self.do_run(args, unknown)
  File "C:\Users\vuong\zephyrproject\zephyr\scripts\west_commands\flash.py", line 32, in do_run
    do_run_common(self, my_args, runner_args, domains=domains)
  File "C:\Users\vuong\zephyrproject\zephyr\scripts/west_commands\run_common.py", line 174, in do_run_common
    do_run_common_image(command, user_args, user_runner_args, d.build_dir)
  File "C:\Users\vuong\zephyrproject\zephyr\scripts/west_commands\run_common.py", line 249, in do_run_common_image
    runner = runner_cls.create(runner_config, args)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vuong\zephyrproject\zephyr\scripts/west_commands\runners\core.py", line 564, in create
    ret = cls.do_create(cfg, args)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vuong\zephyrproject\zephyr\scripts/west_commands\runners\dfu.py", line 96, in do_create
    ret.ensure_device()
  File "C:\Users\vuong\zephyrproject\zephyr\scripts/west_commands\runners\dfu.py", line 100, in ensure_device
    if not self.find_device():
           ^^^^^^^^^^^^^^^^^^
  File "C:\Users\vuong\zephyrproject\zephyr\scripts/west_commands\runners\dfu.py", line 108, in find_device
    output = self.check_output(cmd)
             ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\vuong\zephyrproject\zephyr\scripts/west_commands\runners\core.py", line 734, in check_output
    return subprocess.check_output(cmd, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\subprocess.py", line 548, in run
    with Popen(*popenargs, **kwargs) as process:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\subprocess.py", line 1024, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Python311\Lib\subprocess.py", line 1493, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [WinError 2] The system cannot find the file specified
```

**Expected behavior**

I expect that I can flash blinking code to my board.

**Environment (please complete the following information):**

 - OS: Windows 10
 - Toolchain: Zephyr SDK

jgl-meta, 2023-04-04:
This is not a bug. Feel free to ask on discord.

closed 2023-04-04 by jgl-meta

vilmire, 2023-10-09:
https://github.com/redbear/Duo/blob/master/docs/dfu-util_installation_guide.md