zephyrproject-rtos/zephyr #56486: BLE logging backend not working if it's the only backend enabled

fabiobaltieri, 2023-04-03:
**Describe the bug**
https://github.com/zephyrproject-rtos/zephyr/pull/56307 was recently introduced to support logging with BLE as a backend. Unfortunately it looks like that logging does not work if BLE is the only backend enabled in the project.

**To Reproduce**
`$ west build -t flash -p -b nrf52dk_nrf52832 samples/subsys/logging/ble_backend`

Connect (I'm using nRF Connect on a Pixel 6), enable notifications, get the logs.

`$ west build -t flash -p -b nrf52dk_nrf52832 samples/subsys/logging/ble_backend -DCONFIG_LOG_BACKEND_RTT=n -DCONFIG_LOG_BACKEND_UART=n`

Connect, enable notifications, nothing received.

**Expected behavior**
The notification endpoint sends log messages.

**Impact**
It's experimental. :-)

**Logs and console output**
n/a

**Environment (please complete the following information):**
 - Commit SHA or Version used: a18aa915ce

fabiobaltieri, 2023-04-03:
cc @vChavezB

vChavezB, 2023-04-03:
Hi @fabiobaltieri, I found the problem. I was using the incorrect API to enable the logger for the first time.

Check if this branch I made with the fix works.

https://github.com/vChavezB/zephyr/tree/ble_logger_fix

I will probably also update the sample to include MTU request update and in case the MTU size is not adequate enough the BLE backend will not send the notification.

fabiobaltieri, 2023-04-03:
Yeah, that did it, thanks for looking into it!

fabiobaltieri, 2023-04-03:
> I will probably also update the sample to include MTU request update and in case the MTU size is not adequate enough the BLE backend will not send the notification.

That make sense to me, may still be worth looking into that failure mode though and figure if it's possible to avoid locking the whole system up when the condition happens.

closed 2023-04-04 by fabiobaltieri