zephyrproject-rtos/zephyr #56494: RNG fails on STM32L42xxx STM32L43xxx

papadkostas, 2023-04-03:
When enabling RNG in L42xxx & L43xxx chips, the os is not starting, at least I can't see the boot banner appearing or anything else on zephyr console. What I noticed is that the RNG status registers indicate that the clock error was detected and the isr is triggering all the time.

AHB runs on 80MHz and CLK48SEL bit is 0b00, but HSI48ON is 0b00 so internal 48MHz clock is not working.
Is there a way to enable it or I should hack inside rng driver?

The following are just in case.
```
&pll {
	div-m = <1>;
	mul-n = <20>;
	div-p = <7>;
	div-q = <2>;
	div-r = <4>;
	clocks = <&clk_hsi>;
	status = "okay";
};

&rcc {
	clocks = <&pll>;
	clock-frequency = <DT_FREQ_M(80)>;
	ahb-prescaler = <1>;
	apb1-prescaler = <1>;
	apb2-prescaler = <1>;
};
```
I verified this on nucleo-l432kc and custom l422kb
I think this is not appearing on l47xxx since gsm modem sample is running without any issues.

On dts enable RNG
```
&rng {
	status = "okay";
};
```

On prj.conf enable
```
CONFIG_ENTROPY_GENERATOR=y
```

And a sample code
```
#include <zephyr/kernel.h>
#include <zephyr/random/rand32.h>

void main(void)
{
	while(true){
		k_sleep(K_SECONDS(1));
		printk("I'm alive!");
		printk("RNG data : %d \r\n", sys_rand32_get());
	}
}
```

FRASTM, 2023-04-04:
In the dts/arm/st/l4/stm32l4.dtsi, the clock for the rng node is defined as MSI 48
```
			clocks = <&rcc STM32_CLOCK_BUS_AHB2 0x00040000>,
				 <&rcc STM32_SRC_MSI CLK48_SEL(3)>;
```
This is a common 48MHz clock source for MMC,USB and RNG.

Can you please check that 
1. the MSI should be enabled at range 11 in your platform
```
 &clk-msi {
	msi-range = <11>; /* 48MHz (reset value) */
	status = "okay";
};
```
2. no other node SDMMC, USB should change the CLK48SEL for its own purpose

papadkostas, 2023-04-04:
Thanks, this worked. Boards boot now.

closed 2023-04-04 by papadkostas

erwango, 2023-04-05:
> Shouldn't this be added on nucleo_l432kc dts?

By default this board doesn't enable RNG, so there is nothing to be fixed on that board description.
What could be enhanced is the documentation to help users to get through it more easily on their own.

I'm thinking of: 
- `dts/arm/st/l4/stm32l4.dtsi`
 Add a comment about clock setting requirements:
```
			clocks = <&rcc STM32_CLOCK_BUS_AHB2 0x00040000>,
+				/* Following domain clock setting requires MSI
+				 * clock to be enabled with msi-range = <11>;
+				 */
				 <&rcc STM32_SRC_MSI CLK48_SEL(3)>;
```
Also, in `dts/bindings/rng/st,stm32-rng.yaml`:
```
  clocks:
    required: true
+    description: |
+        Specifies the clock line to be enabled in clock controller as well as
+        the clock domain used, for instance:
+                   <&rcc STM32_SRC_MSI CLK48_SEL(3)> /* RNG clock domain set to MSI */
+         A correctly configured domain clock is required to allow the integrated low
+         sampling clock detection mecanism to behave properly.
+         In provided example, MSI should be configured to provide 48Mhz clock.
```
Would have this helped you ?

Otherwise, I'm thinking how to make the error more visible, maybe by triggering the assert only when kernel initialization has been done. I'll push a PR to see if working.

wojciechslenska, 2023-04-05:
The same issue I observed on `b_u585i_iot02a` board. 

Looks like here there is a problem with the default configuration.
```
&clk_msis {
	status = "okay";
	msi-range = <4>;
	msi-pll-mode;
};
...
&rcc {
	clocks = <&pll1>;
	clock-frequency = <DT_FREQ_M(160)>;
	ahb-prescaler = <1>;
	apb1-prescaler = <1>;
	apb2-prescaler = <1>;
	apb3-prescaler = <1>;
};
...
&rng {
	status = "okay";
};
```
So here by default MSI is 4 MHZ and AHB is 160 MHz

erwango, 2023-04-05:
> The same issue I observed on b_u585i_iot02a board. Looks like here there is a problem with the default configuration.

By default RNG domain clock source is HSI48 (reflected in u5.dtsi as no domain clock is specified).
You need to enable matching domain clock (HSI48) on target board if you want to use this device, which is actually the case on current latest upstream. Just tested and this is working.

wojciechslenska, 2023-04-05:
@erwango 
Thanks a lot, that was the case, in my setup HSI48 was disabled. After enabling everything works properly. 
Checked also on stm32h5 and it also works well.

papadkostas, 2023-04-19:
I don't know what kind of magic happened and worked that day by adding msi instead of hsi48.
I couldn't get it to run right now and I realised its hsi48 that needs on L43x & L42x
```
&clk_hsi48 {
	status = "okay";
};
```

mtahirbutt, 2024-05-17:
> I don't know what kind of magic happened and worked that day by adding msi instead of hsi48. I couldn't get it to run right now and I realised its hsi48 that needs on L43x & L42x
> 
> ```
> &clk_hsi48 {
> 	status = "okay";
> };
> ```

thanks, that also helped to resolve the problem with l496 with gsm modem issue.