zephyrproject-rtos/zephyr #56498: logging: BLE Backend logger sample stops working if MTU Size not correct

vChavezB, 2023-04-03:
**Describe the bug**

As mentioned by user @fabiobaltieri  in this comment https://github.com/zephyrproject-rtos/zephyr/pull/56307#issuecomment-1494424595,  if the MTU Size is not set correctly and there is not a MTU negotiation the program spins somewhere https://github.com/zephyrproject-rtos/zephyr/pull/56307#issuecomment-1494550410.

**To Reproduce**

Set the MTU Size to maximum by uncommenting the following lines
https://github.com/zephyrproject-rtos/zephyr/blob/06f98a5371c978a4977792d9c6d8fb266bb55d6f/samples/subsys/logging/ble_backend/prj.conf#L9-L11

Use a BLE Central that does not increase MTU negotation size and subscribe to the BLE Backend TX characteristic

**Expected behavior**

The program should continue execution even if the MTU size is set incorrectly.

**Suggestions**

I checked what is happening and the problem is if the BLE notification fails it will get stuck trying to send the message:

https://github.com/zephyrproject-rtos/zephyr/blob/06f98a5371c978a4977792d9c6d8fb266bb55d6f/subsys/logging/backends/log_backend_ble.c#L82-L88

since the function buffer_write will not exist until the backend has processed the message

https://github.com/zephyrproject-rtos/zephyr/blob/2a5de5fc30018befab612faff124ffcef9329d07/subsys/logging/log_output.c#L154-L158

I have two suggestions, 

1. Ignore the notify result and return the length so the program doesnt get stuck. This in turn makes the Bluetooth stack report the warning about the MTU size (e.g.  `bt_att: No ATT channel for MTU 57`)

2.  Do not attempt to send the BLE notification at all if the MTU size is not at least the size of the buffer

https://github.com/zephyrproject-rtos/zephyr/blob/06f98a5371c978a4977792d9c6d8fb266bb55d6f/subsys/logging/backends/Kconfig.ble#L19

To achieve this something as follows could be done: 
```c
const uint16_t mtu_size= bt_gatt_get_mtu(logger_conn) -3;
if (mtu_size  < LOG_BACKEND_BLE_BUF_SIZE ) {
    // optionally report to the user or assert.
}
else {
    // notify
}
```

Feedback is suggested so I can update the ble backend so it doesnt get stuck in the write output loop independently if the user does not set correctly the MTU size.

fabiobaltieri, 2023-04-04:
Thanks for the troubleshooting!

cc @nordic-krch: do you think it would make sense to add some defensiveness in `buffer_write` as well? Maybe bail out if `outf` is pushing data out.

carlescufi, 2023-04-13:
@jhedberg @jori-nordic @alwa-nordic @theob-pro please provide input for @vChavezB

jori-nordic, 2023-04-14:
@vChavezB what about a combination of both, to be more resilient:
- do the MTU check, that way a helpful warning can be logged out over the other available backends
- also check the return value, and print the gatt return code as an error

In both cases, sending the message will fail, I think it's pragmatic to just drop it. The alternative is letting the log system queue the messages until we have a channel w/ the correct MTU.

alwa-nordic, 2023-04-14:
I think we don't want errors in the output unless the intention is to signal that we don't support a case where the MTU and this is something the user has to deal with. I imagine it must be annoying for the user to get errors originating from the logging system while they are trying to read logs. So it's better to drop the log entry without error and maybe instead send "<logs dropped>"

There is a also the possibility of chunking the log data into multiple notifications. The simplest solution is to split the data into chunks of at most 20 bytes. That's what fits into notifications with the minimum MTU (23).

If you have `logger_conn` as you allude to in the suggested change, then you should direct the notification like `bt_gatt_notify(logger_conn, ...)`. Then you can also split the data according to the MTU.

Unrelated issue: We should consider using `BT_ATT_CHAN_OPT_UNENHANCED_ONLY`. Otherwise, we are technically not guaranteed in-order delivery of the notifications. Btw, there is currently no API for getting the ATT MTU of the UATT channel. We should consider adding API to expose this information to the application.

jori-nordic, 2023-04-14:
Yeah the original was 20 bytes but it looks like crap on the nRF mobile apps, pretty unreadable with every log line being split over 4 messages/lines in the UI

vChavezB, 2023-04-14:
> So it's better to drop the log entry without error and maybe instead send "1 log entry dropped".

Is this the API function that should be used for this?

https://github.com/zephyrproject-rtos/zephyr/blob/d6990ff8d929e82aa68b8084e4347f79d05dcbbd/include/zephyr/logging/log_output.h#L182

> I think we don't want errors in the output unless the intention is to signal that we don't support the case where the MTU small 

Would it make sense to send a LOG_ERR when initializing the BLE backend if the MTU Size is not enough for the option `CONFIG_LOG_BACKEND_BLE_BUF_SIZE` by comparing it with `CONFIG_BT_L2CAP_TX_MTU` and in addition add the service at run-time depending on this condition ?

alwa-nordic, 2023-04-14:
> Would it make sense to send a LOG_ERR when initializing the BLE backend if the MTU Size is not enough for the option `CONFIG_LOG_BACKEND_BLE_BUF_SIZE` by comparing it with `CONFIG_BT_L2CAP_TX_MTU` and in addition add the service at run-time depending on this condition ?

Yes, that would make sense. I assume there is no reason to have `CONFIG_LOG_BACKEND_BLE_BUF_SIZE` be larger than what fits in a single notification with best-case MTU?

PR #56878 (2023-06-19): logging: Fixed BLE backend buffer size independent of MTU
The BLE backend has a bug that when the MTU size is less than the buffer the notification always fails and does not allow to flush the buffer.

This fix checks for the MTU size of the current connection and adjusts it. In addition, the buffer size is no longer settable by the user and depends on the transmission size of MTU set with `CONFIG_BT_L2CAP_TX_MTU`.

Fixes #56498

Based on the last comment [from @alwa-nordic ](https://github.com/zephyrproject-rtos/zephyr/issues/56498#issuecomment-1508438132) I have made the buffer size depend d [... cut]
closed 2023-06-19 by carlescufi