zephyrproject-rtos/zephyr #56502: LWM2M RD Client stop fn locks up if already suspended

GrantRolls, 2023-04-04:
When using LWM2M Client - the sample code flow makes a call to lwm2m_rd_client_stop in error handling cases
This can occur when the engine has already been suspended, and since some recent (~1yr change) this will block indefinitely

The rd client service needs to be looked at so that a call to stop can be made at any time and does not block on failure (or has a maximum timeout)

See discussion.
https://github.com/zephyrproject-rtos/zephyr/discussions/54478#discussion-4827052

**To Reproduce**
Using lwm2m-client sample code
* Load the lwm2m-client sample
* Add code to force close LTE client, outside of the lwm2m client loop (by calling lte_lc_offline)
* Observe lem2m client enter LTE_NETWORK_ERROR state entered
* Observe block on lwm2m_rd_client_stop call
* Observe block occurring on ln 1360 of lwm2m_rd_client.c

**Expected behavior**
The stop call should not block, at least not indefinitily

**Impact**
No work around for handling error cases, other than using non-threadsafe checks on the state of the client - by the calling code - before calling stop

rlubos, 2023-04-04:
This looks like a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/56254

carlescufi, 2023-04-05:
Closing in favor of https://github.com/zephyrproject-rtos/zephyr/issues/56254, please reopen if this is not a duplicate.

closed 2023-04-05 by carlescufi