zephyrproject-rtos/zephyr #56504: Switch return type of main to 'int'

keith-packard, 2023-04-04:
## Introduction

The application entry point used by zephyr, `main`, doesn't currently have any useful return value. As a result, it has been declared to return `void`. I propose to change that to `int` as required by the C and C++ standards for "hosted" applications.

### Problem description

Zephyr avoids compiler issues when using a `void` return type of `main` by a combination of three methods:

 1. Compiling with `-ffreestanding`
 2. Special casing C++ applications and having those use a return type of `int`.
 3. Adding compiler flags, `-Wno-main`  and `-Wno-main-return-type`

Switching to a return type of  `int` removes the need for using any of these.

Removing the `-ffreestanding` compiler parameter, as proposed in #54638, will enable significantly better error detection in code using standard C library functions as well as improve generated code quality as the compiler will be able to use knowledge about C library semantics during compilation. However, clang emits an error (even with `-Wno-main-return-type`) when building without the `-ffreestanding` flag. 

### Proposed change

Applications will need to switch the return type of their `main` entry point from `void` to `int`. Zephyr will continue to ignore the return value meaning that there will be no change in how the system operates.

## Detailed RFC

PR #54628 shows the sequence of tasks needed to accomplish this goal:

 1. Updating documentation describing the declaration of  `main`
 2. Declaring main in kernel/init.c to return `int` and then explicitly ignoring the value
 3. Switching the definition of main in all in-tree samples and tests to return `int`.

### Proposed change (Detailed)

Most of the changes for function definitions are performed with a coccinelle script, with a handful of manual fixups.

### Dependencies

This should have limited effect on other parts of the system, except for pending additions of new tests and samples. As no behavior change is proposed, this should not affect test results anywhere.

### Concerns and Unresolved Questions

Should the return value of  `main` actually do something? If so, what should that be? The current PR uniformly returns `0`, so we could, in the future, define non-zero return values to do something different.

## Alternatives

The obvious alternative is to do nothing. However, leaving the return value as `void` means that we can never remove the `-ffreestanding` compiler flag as clang will generate an error on a `void` valued `main` definition.

carlescufi, 2023-04-04:
Architecture WG:

- @carlescufi for applications that do not change their `void` to `int`, would that trigger a compiler warning or error? @keith-packard do you know?
- @cfriedt asks about deprecation. But it's unclear how to deprecate this
- @nashif asks for more details: what will happen to out-of-tree apps, what will they need to change, what will happen  if they don't
- No objections in general but we ask @keith-packard to provide info about what happens to existing out-of-tree apps
- @galak would like this to be tried with GCC, clang  and others

keith-packard, 2023-04-04:
> Architecture WG:
> 
>     * @carlescufi for applications that do not change their `void` to `int`, would that trigger a compiler warning or error? @keith-packard do you know?

If they are building without -ffreestanding, then yes. Otherwise, no.

>     * @cfriedt asks about deprecation. But it's unclear how to deprecate this

There's nothing to deprecate at this point; existing code will continue to work correctly unless they disable the -ffreestanding compiler flag, or until we define semantics for the return value.

>     * @nashif asks for more details: what will happen to out-of-tree apps, what will they need to change, what will happen  if they don't

All they'll need to do is change the return type and add appropriate return values. If they don't change, then nothing happens until they remove the `-ffreestanding` compiler flag.

>     * No objections in general but we ask @keith-packard to provide info about what happens to existing out-of-tree apps
> 
>     * @galak would like this to be tried with GCC, clang  and others

The PR has been through CI, which involves testing with both GCC and clang.

carlescufi, 2023-04-04:
@keith-packard thanks for the reply, but there's something I still don't quite get:

>If they are building without -ffreestanding, then yes. Otherwise, no.

Assuming #54638 gets merged, and then assuming that #54628 gets merged too, if I'm a happy out-of-tree user with my `void main()` app and I update Zephyr, what will happen exactly? I am using the Zephyr build system as-is.

carlescufi, 2023-04-11:
@keith-packard thanks for the [insight](https://github.com/zephyrproject-rtos/zephyr/issues/56504#issuecomment-1496397005). But I am still unclear about something: if we merge #54628 but not #54638, then there's no impact to users whatsoever? i.e. no need for them to change their `void main()` to `int main()` at all, regardless of C library used?

keith-packard, 2023-04-11:
> @keith-packard thanks for the [insight](https://github.com/zephyrproject-rtos/zephyr/issues/56504#issuecomment-1496397005). But I am still unclear about something: if we merge #54628 but not #54638, [...]

Good question -- picolibc already disables `-ffreestanding`, and as this patch re-enables the compiler warnings related to main return type, applications using picolibc will need to change their main return type. Once we merge #54638, then all applications not using the minimal C library would need to switch. We could leave the compiler warning flags disabled, which would limit the effect for applications not using clang.

carlescufi, 2023-04-12:
> Good question -- picolibc already disables `-ffreestanding`, and as this patch re-enables the compiler warnings related to main return type, applications using picolibc will need to change their [...]

OK, thanks for the info. Could you please add all of this info in the release notes? Basically I am looking at this from the point of view of a user that maintains an out-of-tree app and reads the release notes, and does not want to be surprised by a strange warning about the `main` return type

keith-packard, 2023-04-12:
Sounds good. I guess I'm still hoping to see the `-ffreestanding` flag disabled more of the time, in which case the release notes for this section will have to be re-written again. But, having them correct as of this patch seems like a good practice.

```
* The declaration of :c:func:`main` has been changed from ``void
  main(void)`` to ``int main(void)``. The main function is required to
  return the value zero. All other return values are reserved. This aligns
  Zephyr with the C and C++ language specification requirements for
  "hosted" environments, avoiding compiler warnings and errors. These
  compiler messages are generated when applications are built in "hosted"
  mode (which means without the ``-ffreestanding`` compiler flag). As the
  ``-ffreestanding`` flag is currently enabled unless the application is
  using picolibc, only applications using picolibc will be affected by this
  change at this time.
```

closed 2023-04-13 by stephanosio