zephyrproject-rtos/zephyr #56506: fpga: allow bitstream to be supplied chunk-wise using callback

arbrauns, 2023-04-04:
**Is your enhancement proposal related to a problem? Please describe.**
Currently, `fpga_load()` expects the entire bitstream to be stored in a contiguous memory region. Because bitstream files tend to be quite large and sparse, it is tempting to 1) store them on external, non-memory-mapped storage and/or 2) compress them. With the current FPGA driver, the entire bitstream would have to be loaded into memory first, which would defeat the purpose and might not be possible at all.

**Describe the solution you'd like**
If possible, FPGA drivers should allow the bitstream to be loaded in chunks by calling a supplied callback. This callback can then load data from external storage and/or uncompress the bitstream. The API would look something like the following:

```
// @param offset The current byte position within the bitstream (sum of all previous chunk lengths)
// @param buf Destination for buffer address containing bitstream data
//
// @retval 0 if the end of bitstream has been reached
// @retval >0 the number of bytes stored in the buffer pointed to by buf
// @retval negative errno code on failure
typedef int (*fpga_load_callback)(size_t offset, uint8_t **buf);

static inline int fpga_load_chunked(const struct device *dev, fpga_load_callback cb);
``` 

For this to be supported, the hardware has to support pauses during the programming process (while the callback produces more data).

**Describe alternatives you've considered**
None.

cc @tgorochowik @kgugala as code owners

tgorochowik, 2023-04-04:
Sounds good! It wasn't needed for our use-case but I can see how that could be useful.

Feel free to propose a PR with the changes! Make sure the new callback is optional so that current drivers continue to work without any modifications

cfriedt, 2024-09-09:
It's possible that some FPGAs might have timing requirements on bitstream loading that might make this somewhat tricky, but that might just be during the handshake phase. 

It seems useful though.

I don't have an immediate need for this feature though. @arbrauns - would you care to make a PR?

arbrauns, 2024-09-09:
We've since added external flash to the device where this would have been relevant, so no current need from our side either.

cfriedt, 2024-09-09:
Ok - we can leave it open until someone needs this feature