zephyrproject-rtos/zephyr #56507: lib\os\timeutil.c: warning: comparing floating-point is unsafe

wniva, 2023-04-04:
I added the compiler option CONFIG_COMPILER_OPT="-Wall -Werror-implicit-function-declaration -Wfloat-equal" to my projects. This should help to write a "safe" code.
I noticed that the following warnings come from the Zephyr code:

lib/os/timeutil.c: In function 'timeutil_sync_ref_from_local':
lib\os\timeutil.c:140:17: warning: comparing floating-point with '==' or '!=' is unsafe [-Wfloat-equal]
lib\os\timeutil.c:150:25: warning: comparing floating-point with '==' or '!=' is unsafe [-Wfloat-equal]
lib/os/timeutil.c: In function 'timeutil_sync_local_from_ref':
lib\os\timeutil.c:170:17: warning: comparing floating-point with '==' or '!=' is unsafe [-Wfloat-equal]
lib\os\timeutil.c:177:24: warning: comparing floating-point with '==' or '!=' is unsafe [-Wfloat-equal]

My environment:
Hardware: Nordic nRF52832
Nordic SDK: 2.1.3 (Zephyr OS build v3.1.99-ncs1-2) and 2.2.0

It would be good if this would be corrected in one of the next versions.

Thanks! :-)

cfriedt, 2023-04-04:
@wniva - are you able to reproduce this on `main` or another supported release branch? E.g. `v2.7-branch`, `v3.3-branch`, `v3.2-branch`.

Also, as I'm sure you're aware, this issue is a direct result of user-specified compiler options, which is not really recognized as a bug.

It might be better categorized as a feature request. E.g.
* provide a macro to abstract floating point comparison
* provide a Kconfig macro to perform "safe" floating-point comparisons
* if disabled, use ==, else perform additional checks and comparison via some other means

Can you also suggest what might be better for comparison implementation? I can think of
* __ASSERT() on NaNs
* __ASSERT() when comparing a non-infinite value with an infinite value
* 1st macro option: use a range comparison (i.e. "within" and some "delta")
* 2nd macro option: use an upper / lower inclusive comparison
* 3rd macro option: binary comparison (i.e. use dereference `uint32_t *` and compare as ints)

andyross, 2024-02-09:
timeutil.c is sort of an odd bird (not the least that it relies on floating point in an RTOS where even today many of our platforms lack FPU hardware!), and more or less lacks a maintainer currently.

I'm going to close this, though.   We aren't useing -Wfloat-equal, and almost certainly never will.   Floating point equality is part of the language standard and does have uses (e.g. testing for zero/unset/initial-state[1]) even if it's an easy footgun.   If there are specific precision problems with the code in timeutil.c let's open specific bugs and fix them.

[1] Pretty sure the two issues flagged above fall into this category.

closed 2024-02-09 by andyross