zephyrproject-rtos/zephyr #56531: snippets: conflict with -DDTC_OVERLAY_FILE and boards/<board>.overlay

mniestroj, 2023-04-04:
**Describe the bug**
Applying devicetree overlay files doesn't work nicely with using `-DDTC_OVERLAY_FILE` on the command line or when `boards/<board>.overlay` is part of application directory.

**To Reproduce**
Default behavior (for reference when `boards/<boards>.overlay` applied correctly) without snippet use:
```
$ west build -b particle_xenon samples/subsys/shell/fs
...
-- Found BOARD.dts: zephyr/boards/arm/particle_xenon/particle_xenon.dts
-- Found devicetree overlay: zephyr/samples/subsys/shell/fs/boards/particle_xenon.overlay
-- Generated zephyr.dts: build/zephyr/zephyr.dts
...
```

Snippet devicetree overlay applied, but `boards/<overlay>.overlay` is ignored:
```
$ west build -b particle_xenon -S cdc-acm-console samples/subsys/shell/fs
...
-- Found BOARD.dts: zephyr/boards/arm/particle_xenon/particle_xenon.dts
-- Found devicetree overlay: zephyr/snippets/cdc-acm-console/cdc-acm-console.overlay
-- Generated zephyr.dts: build/zephyr/zephyr.dts
...
```

`boards/<overlay>.overlay` applied, but snippet overlay ignored (related to #29236):
```
$ west build -b particle_xenon -S cdc-acm-console samples/subsys/shell/fs -- -DDTC_OVERLAY_FILE=boards/particle_xenon.overlay
-- Found BOARD.dts: zephyr/boards/arm/particle_xenon/particle_xenon.dts
-- Found devicetree overlay: boards/particle_xenon.overlay
-- Generated zephyr.dts: build/zephyr/zephyr.dts
```

**Expected behavior**
1. `boards/<board>.overlay` should be applied automatically, regardless of used snippets.
2. Snippets overlays applied even when `-DDTC_OVERLAY_FILE=` is used (related to #29236)

**Impact**
Annoyance, as snippets feature can be used for applications that do not use board specific overlays. Same issue is for application specific overlays (`app.overlay`).

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain: Zephyr SDK 0.15.1
 - Commit SHA: c6b3f009ff7d892a94ee44a31cd07c7968072077

mbolivar-nordic, 2023-04-04:
@mniestroj makes sense, will fix. Thanks for filing.

mbolivar-nordic, 2023-05-15:
Reassigned to @tejlmand for now since he seems to have a zephyr_get() based solution for this? Not exactly sure though. CC @stephanosio since he is the other potential assignee here

tejlmand, 2023-06-06:
Identical to: https://github.com/zephyrproject-rtos/zephyr/issues/57139
Fixed by: #57661

closed 2023-06-06 by tejlmand