zephyrproject-rtos/zephyr #56535: Incorrect clock configuration on nucleo_f303k8

polyakovkrylo, 2023-04-04:
**Describe the bug**
A clear and concise description of what the bug is.
(Note that [Github Discussions](https://github.com/zephyrproject-rtos/zephyr/discussions) are available to first verify that the issue
is a genuine Zephyr bug and not a consequence of Zephyr services misuse.)

The clock on nucleo_f303k8 seems to run twice slower than it is configured.

Please also mention any information which could help others to understand
the problem you're facing:

I have a NULCEO F303K8 board that I would like to use with Zephyr. I tried to flash it with the hello_world example, but for some reason uart2 works at baud rate that is twice slower than configured (57600 while set to 115200 in the dts file). I also tried printing in a while loop with one-second sleeping between iterations, but it seems that the pause is also twice longer than it should be. 

**To Reproduce**
Steps to reproduce the behavior:
1. cd ~/zephyrproject/zephyr
2. west build -b nucleo_f303k8 samples/hello_world
3. west flash
5. serial console works at 57600 instead of configured 115200

**Expected behavior**
A clear and concise description of what you expected to happen.
usart2 baud rate should be 115200

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
annoyance, frustration

**Environment (please complete the following information):**
 - OS: Pop OS  Linux pop-os 6.2.6-76060206-generic #202303130630~1679424972~22.04~4a8cde1 SMP PREEMPT_DYNAMIC Tue M x86_64 x86_64 x86_64 GNU/Linux
 - Toolchain Zephyr SDK 0.16.0
 - zephyr commit 9b36e723f49a61727cac2c1f637a0abe438bb0a5

**Additional context**
I am using default dts configuration. I am a little bit confused by the clock configuration as I am a newbie, but comparing with cubemx configuration I saw that using HSI as the PLL source can only be done with  a /2 divisor for STM32F303K8, thus reaching a maximum of 64 MHz (in the dts configuration it is set to 72MHz, however). 
Nonetheless, setting rcc frequency to 64MHz did not seem to make any difference. 
As for HSE, AFAIK, external oscillator is not available in the default hardware configuration of the board.

polyakovkrylo, 2023-04-05:
Hi @erwango ,

I am afraid the changes you proposed do not fix the problem. As far as I understand, there is no external oscillator on the board by default (according [this manual](https://www.st.com/resource/en/user_manual/um1956-stm32-nucleo32-boards-mb1180-stmicroelectronics.pdf), X1 is reserved for it, but not populated on STMF3XX series). As a result, when I run zephyr with HSE as the clock source, the program simply does not start. Is it possible that I missing something in the configuration?

erwango, 2023-04-06:
> Hi @erwango ,
> 
> I am afraid the changes you proposed do not fix the problem. As far as I understand, there is no external oscillator on the board by default (according [this manual](https://www.st.com/resource/en/user_manual/um1956-stm32-nucleo32-boards-mb1180-stmicroelectronics.pdf), X1 is reserved for it, but not populated in the STMF3XX series). As a result, when I run zephyr with HSE as the clock source, the program simply does not start.

Oops, forgot about that possibility, thanks for heads up, I'l fix this.

PR #56555 (2023-04-07): boards: nucleo_f303k8: Fix clock configuration
Using HSI as PLL source, the max sys clock reachable is 36MHz. Use HSE instead.

Fixes #56535
closed 2023-04-07 by fabiobaltieri