zephyrproject-rtos/zephyr #56550: networking: clarify expected interface states of WiFi drivers

JordanYates, 2023-04-05:
**Is your enhancement proposal related to a problem? Please describe.**

It is currently not clear what the expected sequencing of events for WiFi drivers are supposed to be. Without clear guidelines on the expected behaviour, users are left debating definitions and are unable to identify whether bugs are present or whether it is expected behaviour.

Specifically, the relationship between the Carrier operational state, dormant states, co-processor state and wifi events is not clear.
The existing documentation is rather abstract.
https://docs.zephyrproject.org/latest/connectivity/networking/api/net_if.html#network-interface-state-management

Without a concrete expectation of how the drivers should behave, determining which drivers are compliant with the behaviour is a waste of time, and writing generic management code is impossible.

**Describe the solution you'd like**

An official source of truth of the expected behaviour. As an example (not necessarily correct):

1. Communication with the co-processor is established, flag `X` is set, event `Y` is generated.
2. Application requests the interface to be active, flag `NET_IF_UP` is set, event `NET_EVENT_IF_ADMIN_UP` is generated.
3. Application requests association to a network
4. Co-processor connects to the network, event `NET_EVENT_WIFI_CONNECT_RESULT` is generated.
5. DHCP completes on the network interface, flag `NET_IF_LOWER_UP` is set, event `NET_EVENT_IF_UP` is generated.

**Additional context**
From discussions on Discord with @krish2718 and @rlubos it is clear there is plenty of uncertain definitions in the correct documentation.

rlubos, 2023-04-05:
> The existing documentation is rather abstract.

The documentaton is abstract because it covers the abstract concept of the network interface. IMO it should not be bloated with detailed instructions on how to deal with specific interface types.

> An official source of truth of the expected behaviour. As an example (not necessarily correct):
Communication with the co-processor is established, flag X is set, event Y is generated.
Application requests the interface to be active, flag NET_IF_UP is set, event NET_EVENT_IF_ADMIN_UP is generated.
Application requests association to a network through NET_REQUEST_WIFI_CONNECT
Co-processor connects to the network, event NET_EVENT_WIFI_CONNECT_RESULT is generated.
DHCP completes on the network interface, flag NET_IF_LOWER_UP is set, event NET_EVENT_IF_UP is generated.

All the drivers need to do is to manipulate two flags on the network interface:
* Update the PHY status, represented with `NET_IF_LOWER_UP`
* Update the network association status, represented with `NET_IF_DORMANT`

All of the remaining interface state management logic is implemented in the net_if code itself (
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/net_if.c#L4124).

And this is all covered by the existing docs. More details can be found in the RFC.

> DHCP completes on the network interface

There is not DHCP in this picture - DHCP runs after the interface is operational up.

JordanYates, 2023-04-05:
Ok, then the two parts which are unclear from the docs to me are:
1. What is the definition of the PHY (In particular for the case where there is no "cable" as such)
2. If the PHY is defined as the RF medium (as has been implied):
  i. What is the distinction (if any) between `NET_IF_LOWER_UP` and `NET_IF_DORMANT`?
  ii. What is the distinction (if any) between `NET_EVENT_IF_UP` and `NET_EVENT_WIFI_CONNECT_RESULT`?
  iii. How do we detect whether the interface is in a state where we can send commands to it? (Completed initialisation)

rlubos, 2024-02-09:
Apologies for late reply, I somehow missed the question in my GH notifications.

> 1. What is the definition of the PHY (In particular for the case where there is no "cable" as such [Bluetooth IPSP, WiFi, etc])

I guess that'd be radio peripheral ready to transmit packets (so PHY could be considered ready as soon as driver is initialized).

> 2. If the PHY is defined as the RF medium connection (as has been implied):
>    i. What is the distinction (if any) between `NET_IF_LOWER_UP` and `NET_IF_DORMANT`?

As above, LOWER_UP (carrier) would  represent whether radio is ready to transmit any packet at all (radio peripheral). DORMANT represents whether interface is able to transmit application data (IP packets). Interface remains dormant until it connects with an actual Wi-Fi network (but in order to connect, you need to already be able to send Wi-Fi packets, hence LOWER_UP must be set).

>    ii. What is the distinction (if any) between `NET_EVENT_IF_UP` and `NET_EVENT_WIFI_CONNECT_RESULT`?

For the particular case of Wi-Fi, none I think. Interface is operational UP as soon as it connects to the Wi-Fi network (so we should get both events).

>    iii. How do we detect whether the interface is in a state where we can send commands to it? (Completed initialisation)

`net_if_is_carrier_ok()`, provided that the particular driver controls carrier (LOWER_UP) state correctly.

JordanYates, 2024-02-09:
With the addition of the connectivity library the core problem is more or less solved.
The correct ordering and states are the ones that work with that connectivity library.

rlubos, 2024-02-09:
Ok, let's close the issue then, thanks @JordanYates

closed 2024-02-09 by rlubos