zephyrproject-rtos/zephyr #56557: STM32H747I-DISCO LEDs marked GPIO_ACTIVE_HIGH while being GPIO_ACTIVE_LOW

SkryptX, 2023-04-05:
The STM32H747I-DISCO has four user LEDs and according to the official documentation [UM2411 5.11](https://www.st.com/resource/en/user_manual/dm00504240-discovery-kit-with-stm32h747xi-mcu-stmicroelectronics.pdf) they are active low. In the device tree definition in `zephyr/boards/arm/stm32h747i_disco/stm32h747i_disco.dtsi` they are marked as `GPIO_ACTIVE_HIGH`.

```
leds {
    compatible = "gpio-leds";
    green_led_1:led_1 {
        gpios = <&gpioi 12 GPIO_ACTIVE_HIGH>;
        label = "User LD1";
        status = "disabled";
    };
    orange_led_2:led_2 {
        gpios = <&gpioi 13 GPIO_ACTIVE_HIGH>;
        label = "User LD2";
        status = "disabled";
    };
    red_led_3:led_3 {
        gpios = <&gpioi 14 GPIO_ACTIVE_HIGH>;
        label = "User LD3";
        status = "disabled";
    };
    blue_led_4:led_4 {
        gpios = <&gpioi 15 GPIO_ACTIVE_HIGH>;
        label = "User LD4";
        status = "disabled";
    };
};
```

erwango, 2023-04-05:
@SkryptX Thanks for reporting this. Would you be able to create a PR and contribute this change directly ?

SkryptX, 2023-04-05:
@erwango I'm not currently setup to contribute on open-source repos, but with the amount of tickets I open here, I should probably start with that... It's a nice small change. I'll take it as a test later this week :)

PR #56696 (2023-04-11): boards: stm32h747i_disco: fix gpio_leds definition
Fixes #56557
Fix gpio_leds devicetree definition for the STM32H747I-DISCO board. The LEDs were incorrectly marked as GPIO_ACTIVE_HIGH instead of GPIO_ACTIVE_LOW.
closed 2023-04-11 by carlescufi