zephyrproject-rtos/zephyr #56566: net: per-interface pools and settings

krish2718, 2023-04-05:
**Is your feature request related to a problem? Please describe.**
In a multi-interface system having single networking pools (and settings) is undesired to avoid starvation. 

**Describe the solution you'd like**
Each interface should have its own pools and networking settings (e.g., `CONFIG_NET_BUF_DATA_SIZE`, `CONFIG_NET_BUF_TX_COUNT` etc).

**Describe alternatives you've considered**
Trying to tweak the common settings but still one of the interface is starved.

**Additional context**

A few usecases would be 

* `netusb` + `Wi-Fi` (control + data)
* `thread` + `Wi-Fi` (Thread border router)

jukkar, 2023-04-05:
I think this has been discussed in the past (several year ago). It would indeed prevent starvation for other network interfaces but would increase the resource consumption in the whole system. Each network interface would need to allocate things separately which imho would lead to waste of resources.
It is certainly possible to have net_buf / pkt pools per network interface, but it would not solve the problem of starvation of the resources for the one network interface. And this would also complicate the handling of het network packets in the system.

Perhaps we could come up with a alternate solution that would prevent this starvation of network buffers in the first place. For example, we could throttle the network driver if it starts to receive packets too fast by just dropping the packets we cannot handle. We do this already but in an unpredictable way when we run out of net bufs and we are forced to drop the network packet. Imho this dropping is happening a bit too late to be useful to avoid resource starvation.

The planned throttling should happen early enough when we still have resources for all the network interfaces, either by some automatic way when we cross some utilization limit, or by telling application / some mgmt api that we have crossed the limit and asking permission to drop the packet or not, or something like that.

rlubos, 2023-04-05:
> Each interface should have its own pools and **networking settings (e.g., CONFIG_NET_BUF_DATA_SIZE, CONFIG_NET_BUF_TX_COUNT etc)**.

It's not clear to me how could this be achieved with KConfig, i. e. how to provide per-instance settings.

Generally, for TCP we can limit the net_buf/net_pkt usage by tweaking the TX/RX window sizes (either statically or with socket options). We miss such functionality for UDP though, but probably we could come up with something similar around the `SO_SNDBUF`/`SO_RVCBUF` options.