zephyrproject-rtos/zephyr #56567: Error BLE connection after move to 3.2.0 version

rubenwat, 2023-04-05:
Hello, I was working in 2.7.4 version and my BLE worked perfectly but after trying to move to 3.2.0 it stopped working and I have no idea about the reason but seems bt_conn_ref/ bt_conn_unref is not properly performed. You can see in the picture what I am getting when I try to stablish a connection between nrf connect and my peripheral (nordic nrf52840). The error is not HW because this FW and this HW works fine with 2.7.4. I can provide you all information you need or I can send you the code. Thank you.

[image]

[image]
[image]

cvinayak, 2023-04-07:
Please use the bug report template and provide details as requested, commit revision used, detailed steps to build your application and steps to reproduce the issue.

jgl-meta, 2023-04-11:
Can you give 3.3.0 a try? It may have been fixed.

rubenwat, 2023-04-12:
I tried with version 3.3.0 and unfortunately I get the same error, so I have no idea what this bug could be. I could send you a simplified code privately if you wish.

carlescufi, 2023-04-13:
This looks a bit like a stack overflow, please use the [thread analyzer](https://docs.zephyrproject.org/latest/services/debugging/thread-analyzer.html) to find out if any stacks are overflowing.

rubenwat, 2023-04-25:
I already solved this bug.

closed 2023-04-25 by rubenwat

brandonrc, 2025-04-26:
@rubenwat  can you please share you solution when closing a bug.