zephyrproject-rtos/zephyr #56579: Twister and k_work_submit_to_queue not working

nono313, 2023-04-05:
**Describe the bug**
When pushing a work item to a work queue from within a ztest function, the work handler is not called.

Target used for twister: native_posix.

**To Reproduce**
- Initialize a work queue
- Start the work queue
- Initialize a work item
- Submit the work item to the queue

Test file content I have used: 

```
#include <zephyr/ztest.h>

ZTEST_SUITE(sample_workq, NULL, NULL, NULL, NULL, NULL);

struct k_work_q work_q;

K_THREAD_STACK_DEFINE(stack_rx, 512);

void work_handler(struct k_work *work)
{
	printf("HANDLE\r\n");
}

K_WORK_DEFINE(work, work_handler);

ZTEST(sample_workq, test_workq)
{
	int ret;
	k_work_queue_init(&work_q);
	k_work_queue_start(&work_q, stack_rx,
					K_THREAD_STACK_SIZEOF(stack_rx), 
					5,
					NULL);

	k_work_init(&work, work_handler);

	ret = k_work_submit_to_queue(&work_q, &work);
	zassert(ret >= 0, "error submit");
	k_msleep(100);
}
```

**Expected behavior**
Work item should be executed and HANDLER should be printed in the log output

**Impact**
I am using work queues in my system and am trying to implement unit tests for a module of my system that uses work queues

**Logs and console output**

```
Running TESTSUITE sample_workq
===================================================================
START - test_workq

   Assertion succeeded at WEST_TOPDIR/app/tests_u/src/sample_workq.c:29 (sample_workq_test_workq)
ASSERTION FAIL [handler != ((void *)0)] @ WEST_TOPDIR/zephyr/kernel/work.c:667
@ WEST_TOPDIR/zephyr/lib/os/assert.c:43
[00:00:00.000,000] [1;31m<err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0[0m
[00:00:00.000,000] [1;31m<err> os: Current thread: 0x805aee0 (unknown)[0m
[00:00:00.000,000] [1;31m<err> os: Halting system[0m
```

**Environment (please complete the following information):**
 - OS: WSL (Ubuntu under Windows 10)
 - Toolchain Zephyr SDK  0.15.1
 - fcaa60a99fa9d5256078ed28557856ec3709cfa9

**Additional context**

golowanow, 2023-04-05:
@nono313
 I've got no issue with your test branching from the current main/HEAD 5422a317ad
```
DEBUG   - OUTPUT: *** Booting Zephyr OS build zephyr-v3.3.0-2131-g5422a317ad09 ***
DEBUG   - Could not find a matching testcase for _try.56579
DEBUG   - OUTPUT: Running TESTSUITE sample_workq
DEBUG   - OUTPUT: ===================================================================
DEBUG   - OUTPUT: START - test_workq
DEBUG   - OUTPUT: HANDLE
DEBUG   - OUTPUT:  PASS - test_workq in 0.110 seconds
```

nono313, 2023-04-11:
You are right, I am not able to reproduce it either on the latest main. Thanks for takign the time to test it. I'm closing the issue.