zephyrproject-rtos/zephyr #56583: Missing *-pinctrl.dtsi

elmorejd, 2023-04-05:
I am currently trying to build Zephyr for the Nucleo U575ZI-Q board, but I am encountering an error that is preventing me from proceeding. Specifically, the build process is failing with the following error message:

zephyr/boards/arm/nucleo_u575zi_q/nucleo_u575zi_q-common.dtsi:8:45: fatal error: st/u5/stm32u575zitxq-pinctrl.dtsi: No such file or directory
#include <st/u5/stm32u575zitxq-pinctrl.dtsi>

It appears that the build process is unable to locate the "stm32u575zitxq-pinctrl.dtsi" file in the "st/u5" directory, which is causing the build to fail. I am also not finding the stm32u575zitxq-pinctrl.dtsi within the project. 

I have already double-checked that I have the latest version of Zephyr and the necessary dependencies installed. Can you please advise on how to resolve this issue and successfully build Zephyr for the Nucleo U575ZI-Q board?

Thank you in advance for your help.

nordicjm, 2023-04-06:
I can see that exact file here: https://github.com/zephyrproject-rtos/hal_stm32/blob/main/dts/st/u5/stm32u575zitxq-pinctrl.dtsi
Did you do `west update`?

mtg1603, 2023-04-06:
I had a similar problem compiling my code on Windows. I solved removing the system environment variable of Cygwin from PATH (bin and lib folders). Apparently, my Cygwin installation (which includes a CMake) causes a conflict with the other CMake installed (that is used by the Zephyr), causing references to Zephyr modules to not be found.

gmarull, 2023-04-11:
Closing, as file exists in module.

closed 2023-04-11 by gmarull