zephyrproject-rtos/zephyr #56589: ci:kernel.timernxp-rt1xxx: jitter_drift test failure

hakehuang, 2023-04-06:
**Describe the bug**
kernel.timer.timer.jitter_drift test failure

Please also mention any information which could help others to understand
the problem you're facing:
CI testing on NXP RT11xx platform found below issue with kernel.timer.timer.jitter_drift

**To Reproduce**
Steps to reproduce the behavior:
1. cd tests/kernel/timer/timer_behavior
2. west build -b mimxrt1060_evk
3. west flash --runner=pyocd

**Expected behavior**
Test PASS

**Impact**
timer behavior

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

```
START - test_jitter_drift
periodic timer behavior test (approx 10 seconds)
periodic timer samples gathered, calculating statistics
FAIL - periodic_diff@53. Caught a timer wrap around which isn't handled!
FAIL - periodic_diff@53. Caught a timer wrap around which isn't handled!
timer clock rate 600000000, kernel tick rate 10000
period duration statistics for 9999 samples (1 rollovers):
expected: 1000 us,       	600000.000000 cycles
min:      956.911667 us, 	574147 cycles
max:      1025.383333 us, 	615230 cycles
mean:     999.995693 us, 	599997.415542 cycles
variance: 0.542482 us, 	195293.601203 cycles
stddev:   0.736534 us, 	441.920349 cycles
timer start cycle 13706861, end cycle 1718713719,
total time 2841678.096667 us, expected time 10000000.000000 us,
expected time drift 0.000000 us, difference -7158321.903333 us

Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:210: timer_jitter_drift_test_jitter_drift: (abs(time_diff_us) &lt; CONFIG_TIMER_TEST_MAX_DRIFT is false)
Drift (in microseconds) outside expected bound
FAIL - test_jitter_drift in 2.951 seconds
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v3.3.0-2005-g2a5de5fc3001

hakehuang, 2023-04-06:
@dleach02 @danieldegrasse @mmahadevan108

dleach02, 2023-06-01:
appears to also impact rt1015 (#58729)

danieldegrasse, 2023-06-01:
@hakehuang it seems that a1d21ca69bf471b06ebaad6ce2ec74a9b32724ef may have fixed this issue, as the SYSTICK wrapping around is not necessarily a reason for test failure. When testing on the RT1060 using the following commands:

```
$ west build -p always -b mimxrt1060_evk tests/kernel/timer/timer_behavior
$ west flash
```

I am unable to see a test failure. I do not see any specific test configurations in the `testcase.yaml`, is there a setting I am missing?

hakehuang, 2023-06-02:
> I am unable to see a test failure. I do not see any specific test configurations in the `testcase.yaml`, is there a setting I am missing?

@danieldegrasse , the same steps. it has probilities to reproduce this issue. and its related low on latest main branch code. so I can close it

closed 2023-06-02 by hakehuang